Azure/Industrial-IoT #2019: Certificate Issues with v2.9 (BadCertificateRevocationUnknown)

delema98, 2023-07-24:
After upgrading to Version 2.9.0 we encounter some certificate issues (see stack trace below). In the past (while using v2.8.4) we were using the command line options:  
"--pf=/mount/opcpublishermodule.settings.json",     
"--aa",                                             
"--tm",                                             
"--RejectUnknownRevocationStatus=false",           
"--mq=100",                                         
"--batchsize=2",                                    
"--heartbeatinterval=15",                           
"--iothubprotocol=AmqpOverTcp",                     
"PkiRootPath=/mount/pki"         

Is it possible that the CLI command ""--RejectUnknownRevocationStatus=false" is not supported anymore?
Additionally to the stack trace shown below we get the following error message:
```
Option --RejectUnknownRevocationStatus=false wrong or not supported, please use -h option to get all the supported options.
```

The certificates are structured like this: https://reference.opcfoundation.org/v104/GDS/docs/F.1/

Stack trace:
```
[09:50:44.5132] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[512]
      Imported the PFX private key for [***].
[09:50:44.5530] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Own certificate Subject '***, CN=***, OU=**, O=**, L=**, C=**' (Thumbprint: ***) loaded.
[09:50:44.5569] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Checking application instance certificate.
[09:50:44.6482] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[512]
      Imported the PFX private key for [***].
[09:50:44.6495] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Check certificate: [***, CN=***, OU=**, O=**, L=**, C=**] [***]
[09:50:44.6522] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Check application instance certificate. [***, CN=***, OU=**, O=**, L=**, C=**] [***]
[09:50:44.9421] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Certificate Validation failed. Reason=BadCertificateRevocationUnknown. [***, CN=***, OU=**, O=**, L=**, C=**] [***]
[09:50:44.9456] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Error validating certificate. Exception: Object reference not set to an instance of an object.. Use certificate anyway?
[09:50:44.9494] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Error The certificate with subject ***, CN=***, OU=**, O=**, L=**, C=** in the configuration is invalid.
       Please update or delete the certificate from this location:
       /mount/pki/own
       while configuring OPC UA stack - retry...

[09:50:44.9497] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Detailed error while configuring OPC UA stack.
      Opc.Ua.ServiceResultException: The certificate with subject ***, CN=***, OU=**, O=**, L=**, C=** in the configuration is invalid.
       Please update or delete the certificate from this location:
       /mount/pki/own

         at Opc.Ua.Configuration.ApplicationInstance.CheckApplicationInstanceCertificate(Boolean silent, UInt16 minimumKeySize, UInt16 lifeTimeInMonths)
         at Azure.IIoT.OpcUa.Publisher.Stack.OpcUaClientOptionsEx.BuildApplicationConfigurationAsync(OpcUaClientOptions options, String identity, CertificateValidationEventHandler handler, ILogger logger) in /home/vsts/work/1/s/src/Azure
.IIoT.OpcUa.Publisher/src/Stack/Extensions/OpcUaClientOptionsEx.cs:line 103
```

marcschier, 2023-07-24:
Thanks for reporting this bug, it is a regression of #1777 and caused by the fact that this environment variable was never exposed in the command line, and we only tested the command line options.  Is end of August ok for a fix?

delema98, 2023-07-25:
Thanks for your quick answer and your support! Would a pull request from my side significantly speed up this process? If I find some time to look into this, I might be able to support.

marcschier, 2023-07-25:
Thank you for the offer @delema98, the fix has been integrated as part of the latest snapshot and will be part of 2.9.1, likely mid to end of August.  If you want to test you can publish a docker image from main, and run that one instead of the official one that will be released.

closed 2023-07-25 by marcschier