Azure/Industrial-IoT #2020: Error when trying to connect to OPC server with UsernamePassword authentication

zkaiser-ff, 2023-07-25:
**Describe the bug**
I'm trying to publish from an OPC server that has UsernamePassword authentication. The logs are showing the following error.
`Failed to create session 'opc.tcp://172.xxx.xxx.xxx:4840_7BA50A69' due to One or more errors occurred. (Object reference not set to an instance of an object.)`

I am able to connect to the OPC server successfully using Prosys OPC UA Browser.

**To Reproduce**
Steps to reproduce the behavior:
1. Set published nodes file with endpoint, UsernamePassword as the authentication mode and the OpcAuthenticationUsername and OpcAuthenticationPassword values set.

**Expected behavior**
The nodes in the published nodes file are published to the IoT hub

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Ubuntu 20.04
 - OpcPublisher: 2.8.7

**Additional context**
Add any other context about the problem here.

publishednodes.json file

```
[
  {
    "EndpointUrl": "opc.tcp://172.xxx.xxx.xxx:4840",
    "EndpointSecurityMode": "None",
    "EndpointSecurityPolicy": "Basic256Sha256",
    "OpcAuthenticationMode": "UsernamePassword",
    "OpcAuthenticationUsername": "**********",
    "OpcAuthenticationPassword": "**********",
    "OpcNodes": [
      {
        "Id": "ns=4;s=machine_status"
      },
      {
        "Id": "ns=4;s=actual_speed_cph"
      }
    ]
  }
]
```

Startup command
```
"Cmd": [ "--pf=/app/opcpublisher/resources/publishednodes.json", "--aa", "--rf"],
```

Logs
```
[2023-07-25 16:25:29.983 INF OpcUa] ChannelId 0: in Connecting state.
[2023-07-25 16:25:30.047 INF OpcUa] ChannelId 0: in Opening state.
[2023-07-25 16:25:30.048 INF OpcUa] ChannelId 0: Token #0 created. CreatedAt=16:25:30.048. Lifetime=3600000.
[2023-07-25 16:25:30.086 INF OpcUa] Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
[2023-07-25 16:25:30.086 INF OpcUa] Sender Certificate: (none)
[2023-07-25 16:25:30.086 INF OpcUa] ChannelId 3581383580: Token #1 activated. CreatedAt=16:25:30.048. Lifetime=3600000.
[2023-07-25 16:25:30.086 INF OpcUa] ChannelId 3581383580: in Open state.
[2023-07-25 16:25:30.087 INF OpcUa] ChannelId 3581383580: Token Expiry 07/25/2023 17:25:30, renewal scheduled in 2699970 ms.
[2023-07-25 16:25:30.246 WRN OpcUa] Discovery endpoint contains invalid Url: opc.tcp://{machine name}:4840
[2023-07-25 16:25:30.246 WRN OpcUa] Discovery endpoint contains invalid Url: opc.tcp://{machine name}:4840
[2023-07-25 16:25:30.246 WRN OpcUa] Discovery endpoint contains invalid Url: opc.tcp://{machine name}:4840
[2023-07-25 16:25:30.247 INF OpcUa] ChannelId 3581383580: in Closing state.
[2023-07-25 16:25:30.294 WRN OpcUa] ChannelId 3581383580: Force reconnect reason=BadConnectionClosed 'Remote side closed connection'
[2023-07-25 16:25:30.295 INF OpcUa] ChannelId 3581383580: in Closed state.
[2023-07-25 16:25:30.295 INF OpcUa] ChannelId 3581383580: CLIENTCHANNEL SOCKET CLOSED: 01223386
[2023-07-25 16:25:30.295 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Creating session 'opc.tcp://172.xxx.xxx.xxx:4840_7BA50A69' for endpoint 'opc.tcp://172.xxx.xxx.xxx:4840'...
[2023-07-25 16:25:30.296 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Azure IIoT opc.tcp://172.xxx.xxx.xxx:4840_7BA50A69 took 00:00:00.0009999.
[2023-07-25 16:25:30.297 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to create session 'opc.tcp://172.xxx.xxx.xxx:4840_7BA50A69' due to One or more errors occurred. (Object reference not set to an instance of an object.)
```

I did notice in the logs that the {machine name} listed has a few spaces in it. I'm not sure if that's the cause of the issue. Prosys doesn't have any issues with that, but our Ignition SCADA system does say that url isn't reachable, but allows me to substitute the host name for the IP address. Is there a way to do that with the OPCPublisher?

marcschier, 2023-07-25:
Would you mind trying this with OPC Publisher 2.9.0 (latest) and using the --sl and --ll=Debug switches to get more logs from the OPC UA stack underneath?  It is possible that it has to do with username/password auth, but the configuration looks ok, so I think this is rather that the server closed the connection and then we typically see these issues (Transport channel inside the OPC UA stack is accessed even though it is null as result of it being closed).

zkaiser-ff, 2023-07-25:
Here's the logs from that:

```
[16:47:20.0075] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Processing event ConnectRetry in State Connecting...
[16:47:20.0076] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Connecting Client opc.tcp://172.xxx.xxx.xxx:4840_0CD6582F_<<UnknownWriterGroup>> [state:Connecting|refs:1] to opc.tcp://172.xxx.xxx.xxx:4840...
[16:47:20.0078] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 0: in Connecting state.
[16:47:20.1153] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 0: in Opening state.
[16:47:20.1154] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 0: Token #0 created. CreatedAt=16:47:20.115. Lifetime=3600000.
[16:47:20.1647] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
[16:47:20.1648] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Sender Certificate: (none)
[16:47:20.1648] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383799: Token #1 activated. CreatedAt=16:47:20.115. Lifetime=3600000.
[16:47:20.1649] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383799: in Open state.
[16:47:20.1649] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383799: Token Expiry 07/25/2023 17:47:20, renewal scheduled in 2699962 ms.
[16:47:20.2333] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Discovery endpoint contains invalid Url: opc.tcp://ON4306 Forsman :4840
[16:47:20.2335] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Discovery endpoint contains invalid Url: opc.tcp://ON4306 Forsman :4840
[16:47:20.2335] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Discovery endpoint contains invalid Url: opc.tcp://ON4306 Forsman :4840
[16:47:20.2336] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Selecting endpoint opc.tcp://172.xxx.xxx.xxx:4840/ with SecurityMode None and Basic256Sha256 SecurityPolicyUri from:
            #000: opc.tcp://ON4306 Forsman :4840|None [http://opcfoundation.org/UA/SecurityPolicy#None]
            #065: opc.tcp://ON4306 Forsman :4840|Sign [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
            #115: opc.tcp://ON4306 Forsman :4840|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
[16:47:20.2337] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383799: in Closing state.
[16:47:20.2821] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383799: Force reconnect reason=BadConnectionClosed 'Remote side closed connection'
[16:47:20.2822] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383799: in Closed state.
[16:47:20.2822] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383799: CLIENTCHANNEL SOCKET CLOSED: 02E008E7
[16:47:20.2828] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      No endpoint found that matches connection of session opc.tcp://172.xxx.xxx.xxx:4840_0CD6582F_<<UnknownWriterGroup>>.
[16:47:20.2828] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Event ConnectRetry in State Connecting processed.
[16:47:21.6793] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Processing event ConnectRetry in State Connecting...
[16:47:21.6793] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Connecting Client opc.tcp://172.xxx.xxx.xxx:4840_0CD6582F_<<UnknownWriterGroup>> [state:Connecting|refs:1] to opc.tcp://172.xxx.xxx.xxx:4840...
[16:47:21.6795] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 0: in Connecting state.
[16:47:21.7561] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 0: in Opening state.
[16:47:21.7562] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 0: Token #0 created. CreatedAt=16:47:21.756. Lifetime=3600000.
[16:47:21.7837] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
[16:47:21.7837] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Sender Certificate: (none)
[16:47:21.7838] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383800: Token #1 activated. CreatedAt=16:47:21.756. Lifetime=3600000.
[16:47:21.7839] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383800: in Open state.
[16:47:21.7839] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383800: Token Expiry 07/25/2023 17:47:21, renewal scheduled in 2699979 ms.
[16:47:21.8361] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Discovery endpoint contains invalid Url: opc.tcp://ON4306 Forsman :4840
[16:47:21.8362] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Discovery endpoint contains invalid Url: opc.tcp://ON4306 Forsman :4840
[... 167 lines cut ...]
      ChannelId 3581383804: Token Expiry 07/25/2023 17:47:28, renewal scheduled in 2699953 ms.
[16:47:28.5024] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Discovery endpoint contains invalid Url: opc.tcp://ON4306 Forsman :4840
[... cut ...]
      ChannelId 3581383804: in Closed state.
[16:47:28.5502] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      ChannelId 3581383804: CLIENTCHANNEL SOCKET CLOSED: 01BCF2F7
[16:47:28.5504] warn: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      No endpoint found that matches connection of session opc.tcp://172.xxx.xxx.xxx:4840_0CD6582F_<<UnknownWriterGroup>>.
[16:47:28.5504] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Event ConnectRetry in State Connecting processed.
```

mregen, 2023-07-29:
Hi @zkaiser-ff, neither DNS hostnames nor Url are allowed to have space characters in it. For the hostname its forbidden, in a url the space char is escaped. So I'm surprised that Prosys accept such a Url as a hostname. 

Even when we can connect to discover the endpoints with the IP address the response contains an invalid Url. The code which tries to replace the hostname in the returned discovery Url can not work because the returned Url is invalid with space characters. 

Some SDK are more strict about the validation of the returned Url than others, thats why it works with some tools. But in any case the returned discovery Url with spaces is invalid and may lead to all sort of strange behavior. A server configuration tool should prevent such invalid configuration.

I can only suggest to fix the hostname/discovery configuration of the OPC server to make it work.

closed 2023-08-02 by marcschier