Azure/Industrial-IoT #2026: OpcPublisher container has constantly increasing memory consumption

Haeavar, 2023-08-09:
**Describe the bug**
We are using the opc-publisher v2.9.0 as container in our kubernetes environment and observe an increasing memory consumption. 

See the screenshot for a sample. The peeks down to 200M is the time when the container is restartet.
[image]

**Expected behavior**
The memory consumption should not constantly increasing.

**Additional context**

This is our OpcPublisher configuration
    - PublishedNodesFile=/appdata/publisher/pn_v2-direct.json
    - DefaultTransport=Mqtt
    - MqttClientConnectionString=HostName=mosquitto;Port=1883;Protocol='v5'
    - MessagingMode=PubSub
    - MessageEncoding=Json
    - LogLevel=Information
    - UseStandardsCompliantEncoding=True
    - DefaultQueueSize=10000
    - DefaultSamplingInterval=50
    - PublisherId=opcbatch
    - SiteId=Z0001Z0001
    - BatchTriggerInterval=0
    - BatchSize=1
    - TelemetryTopicTemplate={RootTopic}/messages/{DataSetWriterGroup}
    - DefaultMetaDataUpdateTime=0
    - DisableDataSetMetaData=true

See the diagnostics information for a overview of our current publisher configuration (via pn.json).

2023-08-09T05:39:01.479Z   DIAGNOSTICS INFORMATION for          : Z0008A0011 (OPC Publisher 2.9.0.1)
2023-08-09T05:39:01.479Z   # Time                               : 2023-08-09T05:39:01.4780094Z
2023-08-09T05:39:01.479Z   # Ingestion duration                 :    00:00:25:44 (dd:hh:mm:ss)
2023-08-09T05:39:01.479Z   # Ingress DataChanges (from OPC)     :            124 (All time ~0.08/s; 03 in last 60s ~0.05/s)
2023-08-09T05:39:01.479Z   # Ingress EventData (from OPC)       :              0
2023-08-09T05:39:01.479Z   # Ingress ValueChanges (from OPC)    :            321 (All time ~0.21/s; 03 in last 60s ~0.12/s)
2023-08-09T05:39:01.479Z   # Ingress Events (from OPC)          :             31
2023-08-09T05:39:01.479Z   # Ingress BatchBlock buffer size     :              0
2023-08-09T05:39:01.479Z   # Encoding Block input/output size   :              0 | 0
2023-08-09T05:39:01.479Z   # Encoder Notifications processed    :            124
2023-08-09T05:39:01.479Z   # Encoder Notifications dropped      :              0
2023-08-09T05:39:01.479Z   # Encoder IoT Messages processed     :             90
2023-08-09T05:39:01.479Z   # Encoder avg Notifications/Message  :              1
2023-08-09T05:39:01.479Z   # Encoder worst Message split ratio  :              0
2023-08-09T05:39:01.479Z   # Encoder avg IoT Message body size  :            677
2023-08-09T05:39:01.479Z   # Encoder avg IoT Chunk (4 KB) usage :            0.2
2023-08-09T05:39:01.479Z   # Estimated IoT Chunks (4 KB) per day:              0
2023-08-09T05:39:01.479Z   # Outgress input buffer count        :              0
2023-08-09T05:39:01.479Z   # Outgress input buffer dropped      :              0
2023-08-09T05:39:01.479Z   # Outgress IoT message count         :             90 (0.06/s)
2023-08-09T05:39:01.479Z   # Connection retries                 :              1
2023-08-09T05:39:01.479Z   # Opc endpoint connected?            :           True
2023-08-09T05:39:01.479Z   # Monitored Opc nodes succeeded count:             26
2023-08-09T05:39:01.479Z   # Monitored Opc nodes failed count   :              2
2023-08-09T05:39:01.479Z   DIAGNOSTICS INFORMATION for          : A0280E0024 (OPC Publisher 2.9.0.1)
2023-08-09T05:39:01.479Z   # Time                               : 2023-08-09T05:39:01.4780094Z
2023-08-09T05:39:01.479Z   # Ingestion duration                 :    00:00:25:44 (dd:hh:mm:ss)
2023-08-09T05:39:01.479Z   # Ingress DataChanges (from OPC)     :             27 (All time ~0.02/s; 01 in last 60s ~0.02/s)
2023-08-09T05:39:01.479Z   # Ingress EventData (from OPC)       :              0
2023-08-09T05:39:01.479Z   # Ingress ValueChanges (from OPC)    :             52 (All time ~0.03/s; 01 in last 60s ~0.02/s)
2023-08-09T05:39:01.479Z   # Ingress Events (from OPC)          :              9
2023-08-09T05:39:01.479Z   # Ingress BatchBlock buffer size     :              0
2023-08-09T05:39:01.479Z   # Encoding Block input/output size   :              0 | 0
2023-08-09T05:39:01.479Z   # Encoder Notifications processed    :             27
2023-08-09T05:39:01.479Z   # Encoder Notifications dropped      :              0
2023-08-09T05:39:01.479Z   # Encoder IoT Messages processed     :             27
2023-08-09T05:39:01.479Z   # Encoder avg Notifications/Message  :              1
2023-08-09T05:39:01.479Z   # Encoder worst Message split ratio  :              0
2023-08-09T05:39:01.479Z   # Encoder avg IoT Message body size  :            455
[... 81 lines cut ...]
2023-08-09T05:39:01.481Z   # Monitored Opc nodes failed count   :             30
2023-08-09T05:39:01.481Z   DIAGNOSTICS INFORMATION for          : X0000X0000 (OPC Publisher 2.9.0.1)
2023-08-09T05:39:01.481Z   # Time                               : 2023-08-09T05:39:01.4780094Z
2023-08-09T05:39:01.481Z   # Ingestion duration                 :    00:00:25:44 (dd:hh:mm:ss)
2023-08-09T05:39:01.481Z   # Ingress DataChanges (from OPC)     :              1 (All time ~0/s; 00 in last 60s ~0/s)
2023-08-09T05:39:01.481Z   # Ingress EventData (from OPC)       :              0
2023-08-09T05:39:01.481Z   # Ingress ValueChanges (from OPC)    :              1 (All time ~0/s; 00 in last 60s ~0/s)
2023-08-09T05:39:01.481Z   # Ingress Events (from OPC)          :              0
2023-08-09T05:39:01.481Z   # Ingress BatchBlock buffer size     :              0
2023-08-09T05:39:01.481Z   # Encoding Block input/output size   :              0 | 0
2023-08-09T05:39:01.481Z   # Encoder Notifications processed    :              1
2023-08-09T05:39:01.481Z   # Encoder Notifications dropped      :              0
2023-08-09T05:39:01.481Z   # Encoder IoT Messages processed     :              1
2023-08-09T05:39:01.481Z   # Encoder avg Notifications/Message  :              1
2023-08-09T05:39:01.481Z   # Encoder worst Message split ratio  :              0
2023-08-09T05:39:01.481Z   # Encoder avg IoT Message body size  :          3,240
2023-08-09T05:39:01.481Z   # Encoder avg IoT Chunk (4 KB) usage :            0.8
2023-08-09T05:39:01.481Z   # Estimated IoT Chunks (4 KB) per day:              0
2023-08-09T05:39:01.481Z   # Outgress input buffer count        :              0
2023-08-09T05:39:01.481Z   # Outgress input buffer dropped      :              0
2023-08-09T05:39:01.481Z   # Outgress IoT message count         :              1 (0/s)
2023-08-09T05:39:01.481Z   # Connection retries                 :              1
2023-08-09T05:39:01.481Z   # Opc endpoint connected?            :           True
2023-08-09T05:39:01.481Z   # Monitored Opc nodes succeeded count:            -29
2023-08-09T05:39:01.481Z   # Monitored Opc nodes failed count   :             30

marcschier, 2023-08-11:
Could you give me an idea as to the content of the publishednodes.json file?  Also, the log when the publisher starts up and creates the monitored items?  

Finally, is it possible to run the publisher with dotnet-monitor and do 2 memory snapshots then compare where the memory leak is materializing?  

The negative succeeded also looks like another bug.

marcschier, 2023-08-13:
Also, could you please not set a memory limit and then see whether the memory consumption stabalizes at a level?

marcschier, 2023-09-11:
Trying this as part of the docker compose setup using `docker compose -f docker-compose.yaml -f with-mosquitto.yaml -f with-limits.yaml up` (changing limits to 100 mb) and monitoring memory use I could run at 80 MB without seeing any increase in memory.  See my comments for things to try.

closed 2023-09-11 by marcschier