Azure/Industrial-IoT #2032: Writing to the edge using publisher 2.9, direct method? Is cloud service necessary?

mwiot, 2023-08-18:
Hi, 
we have been using iiot with cloud orchestration using version 2.8, and now see that the orchestration approach is abandoned. 

"Customers told us they want to leverage the capabilities of OPC Twin. However, deploying and operating a micro service architecture is often times too costly and too difficult for them. Many times customers were already using an Azure IoT Hub instance to manage other devices and wanted to leverage this instance also for OPC UA enabled assets. **What they were looking for was the standalone version of OPC Publisher but with its capabilities extended to provide read/write/call capabilities also**."

1. Does that mean we can now do everything using iot hub direct methods? Or are we still dependant of using the cloud service for 2.9 functionality? It seems to be optional in the docs, but does it have some functionality that really in practice does not make it optional? I.e If you want to write back to the edge.
2. How do we do write operations to a ua server using the new OPC publisher 2.9? I dont see any write operations in publisher direct methods. Are we dependant on the cloud service to do that?

Direct methods listed in the docs:
[image]

Best regards

marcschier, 2023-08-19:
The configuration API is the same as in previous versions. The full API for OPC Publisher is described here though (which includes the configuration API, but also discovery and twin API): https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/api.md.   The API you are looking for is: https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/api.md#valuewrite.  You can call directly through the IoT Hub, but if you need chunking (because your requests or responses are larger than 256k), then you can only do so using the companion web app, and its REST API.

I realize we still need sample applications that demonstrate this, but I could not get around to doing that.  I will keep this open to track it.

marcschier, 2023-08-31:
Added IoT hub samples into the samples folder. Additional sample contributions are appreciated.