Azure/Industrial-IoT #2033: changed from Connecting to NotReachable

Ilheu, 2023-08-21:
**The configured opc endpoints were working until a week ago, and now they are giving me the following logs (nothing has changed):** 

[10:04:05.2048] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Now monitoring 33 nodes in subscription opc.tcp://192.168.131.20:4840_7DA496B6_<<UnknownWriterGroup>>:<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b):2817639364.
[10:04:05.2050] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Applying changes to 1 subscription(s) took 00:00:00.0154144.
[10:04:05.2233] info: Azure.IIoT.OpcUa.Publisher.Services.WriterGroupDataSource[0]
      Writer with subscription opc.tcp://192.168.131.23:4840_E007F86C_<<UnknownWriterGroup>>:<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b) in writer group <<UnknownWriterGroup>> closed.
[10:04:05.2301] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Now monitoring 12 nodes in subscription opc.tcp://192.168.131.200:48400/UA/ComServerWrapper_A3E695DE_<<UnknownWriterGroup>>:<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b):3.
[10:04:05.2304] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Applying changes to 1 subscription(s) took 00:00:00.0024898.
[10:04:05.2307] info: Azure.IIoT.OpcUa.Publisher.Services.WriterGroupDataSource[0]
      Updated subscription for writer opc.tcp://192.168.131.200:48400/UA/ComServerWrapper_A3E695DE_<<UnknownWriterGroup>>:<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b) in writer group <<UnknownWriterGroup>>.
[10:04:05.2308] info: Azure.IIoT.OpcUa.Publisher.Services.WriterGroupDataSource[0]
      Successfully updated all subscriptions inside the writer group <<UnknownWriterGroup>>.
[10:04:11.6998] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp://192.168.131.23:4840_E007F86C_<<UnknownWriterGroup>> with opc.tcp://192.168.131.23:4840 changed from Connecting to NotReachable
[10:04:11.7000] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #1: Failed to connect opc.tcp://192.168.131.23:4840_E007F86C_<<UnknownWriterGroup>> [state:NotReachable|refs:0] to opc.tcp://192.168.131.23:4840/: Error establishing a connection: BadRequestTimeout...
[10:04:11.7009] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Applying changes to 0 subscription(s) took 00:00:00.0004005.
[10:04:11.7010] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp://192.168.131.23:4840_E007F86C_<<UnknownWriterGroup>> with opc.tcp://192.168.131.23:4840 changed from NotReachable to Disconnected

  DIAGNOSTICS INFORMATION for          : <<UnknownWriterGroup>> (OPC Publisher 2.9.0.1)
   Time                               : 2023-08-21T10:04:57.9243675Z
   Ingestion duration                 :    00:00:02:49 (dd:hh:mm:ss)
   Ingress DataChanges (from OPC)     :            230 (All time ~1.36/s; 87 in last 60s ~1.45/s)
   Ingress EventData (from OPC)       :              0
   Ingress ValueChanges (from OPC)    :            928 (All time ~5.47/s; 87 in last 60s ~6.5/s)
   Ingress Events (from OPC)          :              0
   Ingress BatchBlock buffer size     :              0
   Encoding Block input/output size   :              0 | 0
   Encoder Notifications processed    :            221
   Encoder Notifications dropped      :              0
   Encoder IoT Messages processed     :             20
   Encoder avg Notifications/Message  :             11
   Encoder worst Message split ratio  :              0
   Encoder avg IoT Message body size  :         11,035
   Encoder avg IoT Chunk (4 KB) usage :            2.7
   Estimated IoT Chunks (4 KB) per day:              0
   Outgress input buffer count        :              0
   Outgress input buffer dropped      :              0
   Outgress IoT message count         :             20 (0.11/s)
   Connection retries                 :              1
   Opc endpoint connected?            :           True
   Monitored Opc nodes succeeded count:             12
   Monitored Opc nodes failed count   :              0

[10:05:26.1806] info: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
[10:05:26.1821] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherModule[0]
      Stopped module OpcPublisher.
[10:05:26.2157] info: Azure.IIoT.OpcUa.Publisher.Discovery.NetworkDiscovery[0]
      Cancelling all pending requests...
[10:05:26.2160] info: Azure.IIoT.OpcUa.Publisher.Discovery.NetworkDiscovery[0]
      Pending requests cancelled...
[10:05:26.2161] info: Azure.IIoT.OpcUa.Publisher.Discovery.NetworkDiscovery[0]
      Stopped discovery processor.
[10:05:26.2448] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
      Stopping all clients...
[10:05:26.2495] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Successfully closed client opc.tcp://192.168.131.23:4840_E007F86C_<<UnknownWriterGroup>> [state:Disconnected|refs:0].
[10:05:26.3122] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #81, Reconnecting=False, Error: BadSessionClosed
[10:05:26.3210] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Successfully closed client opc.tcp://192.168.131.200:48400/UA/ComServerWrapper_A3E695DE_<<UnknownWriterGroup>> [state:Disconnected|refs:1].
[10:05:26.3345] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #81, Reconnecting=False, Error: BadSessionClosed
[... 688 lines cut ...]
      Server has revised ns=11;s=::AsGlobalPV:HTdat.Zone[3].rTempControl ('tempZone3') in subscription #3249013590
      The item's actual/desired states:
      SamplingInterval 1000/2000,
      QueueSize 1/1
[10:05:49.7020] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaMonitoredItem.DataItemWithHeartbeat[0]
      Server has revised ns=11;s=::AsGlobalPV:HTdat.Zone[4].rTempControl ('tempZone4') in subscription #3249013590
      The item's actual/desired states:
      SamplingInterval 1000/2000,
      QueueSize 1/1
[10:05:49.7021] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaMonitoredItem.DataItemWithHeartbeat[0]
      Server has revised ns=11;s=::AsGlobalPV:AXdat[11].rActVelAbs ('rActVelAbs') in subscription #3249013590
      The item's actual/desired states:
      SamplingInterval 1000/2000,
      QueueSize 1/1
[10:05:49.7023] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaMonitoredItem.DataItemWithHeartbeat[0]
      Server has revised ns=11;s=::AsGlobalPV:AUTdat.uInProgress ('autoRunning') in subscription #3249013590
      The item's actual/desired states:
      SamplingInterval 1000/2000,
      QueueSize 1/1
[10:05:49.7025] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaMonitoredItem.DataItemWithHeartbeat[0]
      Server has revised ns=11;s=::AsGlobalPV:IMM.ActiveJobType1.CyclicJobInformationType1.InjectionUnitCycleParamType1_1.DosingTime ('dosingTime') in subscription #3249013590
      The item's actual/desired states:
      SamplingInterval 1000/2000,
      QueueSize 1/1
[10:05:49.7026] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaMonitoredItem.DataItemWithHeartbeat[0]

Ilheu, 2023-08-21:
**The main issue is that I used to receive a PublisherId inside "Messages" key, that used to indentify the endpoint of each node, I stop receiving it....**
[image]

**I thought it was from the version update to 2.9.1 but in the old ones I still don't receive the endpoint**

[{"MessageId":"cfb3ff97-d5f1-4fbb-a829-675e335569fc","MessageType":"ua-data","PublisherId":"railesIoThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":34422,"MetaDataVersion":{"MajorVersion":1222352431,"MinorVersion":3467188674},"Timestamp":"2023-08-17T12:58:32.2659456Z","MessageType":"ua-deltaframe","Payload":{"tempZone3":{"Value":37.7999992,"SourceTimestamp":"2023-08-16T02:28:42.0572053Z","ServerTimestamp":"2023-08-16T02:28:42.0572053Z"}}}]},{"MessageId":"5306015b-a5fe-492a-9530-4c051228e0dd","MessageType":"ua-data","PublisherId":"railesIoThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":34423,"MetaDataVersion":{"MajorVersion":1222352431,"MinorVersion":3467188674},"Timestamp":"2023-08-17T12:58:32.2663678Z","MessageType":"ua-deltaframe","Payload":{"tempBicoHR1":{"Value":74.8000031,"SourceTimestamp":"2023-08-16T02:28:41.0562053Z","ServerTimestamp":"2023-08-16T02:28:41.0562053Z"}}}]},{"MessageId":"6d4fb472-3a1f-45ce-ab1b-bd9bfe23b4ab","MessageType":"ua-data","PublisherId":"railesIoThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":34424,"MetaDataVersion":{"MajorVersion":1222352431,"MinorVersion":3467188674},"Timestamp":"2023-08-17T12:58:32.2667118Z","MessageType":"ua-deltaframe","Payload":{"tempZone4":{"Value":41.1000023,"SourceTimestamp":"2023-08-16T02:28:42.0572053Z","ServerTimestamp":"2023-08-16T02:28:42.0572053Z"}}}]},{"MessageId":"3ab77e9a-f915-4778-b3a2-f8d44b0a8170","MessageType":"ua-data","PublisherId":"railesIoThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":34425,"MetaDataVersion":{"MajorVersion":1222352431,"MinorVersion":3467188674},"Timestamp":"2023-08-17T12:58:32.2668737Z","MessageType":"ua-deltaframe","Payload":{"tempZone1":{"Value":42.4000015,"SourceTimestamp":"2023-08-16T02:28:41.0562053Z","ServerTimestamp":"2023-08-16T02:28:41.0562053Z"}}}]},{"MessageId":"6cf634a5-7638-413f-96f9-15b7956fdca4","MessageType":"ua-data","PublisherId":"railesIoThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":34426,"MetaDataVersion":{"MajorVersion":1222352431,"MinorVersion":3467188674},"Timestamp":"2023-08-17T12:58:32.2670361Z","MessageType":"ua-deltaframe","Payload":{"tempZone2":{"Value":39.7000008,"SourceTimestamp":"2023-08-16T02:28:42.0572053Z","ServerTimestamp":"2023-08-16T02:28:42.0572053Z"}}}]},{"MessageId":"947fe169-78d6-4908-83d4-612c748d506e","MessageType":"ua-data","PublisherId":"railesIoThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":34427,"MetaDataVersion":{"MajorVersion":1222352431,"MinorVersion":3467188674},"Timestamp":"2023-08-17T12:58:32.2672634Z","MessageType":"ua-deltaframe","Payload":{"currentPressure":{"Value":5,"SourceTimestamp":"2023-08-16T02:28:41.0562053Z","ServerTimestamp":"2023-08-16T02:28:41.0562053Z"}}}]},{"MessageId":"05a3557d-6eb6-4967-b627-235df6cbf99a","MessageType":"ua-data","PublisherId":"railesIoThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":34428,"MetaDataVersion":{"MajorVersion":1222352431,"MinorVersion":3467188674},"Timestamp":"2023-08-17T12:58:32.2676325Z","MessageType":"ua-deltaframe","Payload":{"tempBicoHR2":{"Value":76.7000046,"SourceTimestamp":"2023-08-16T02:28:41.0562053Z","ServerTimestamp":"2023-08-16T02:28:41.0562053Z"}}}]},{"MessageId":"979737f7-68f0-4654-ab3b-9c7911bb1f32","MessageType":"ua-data","PublisherId":"railesIoThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":12961,"MetaDataVersion":{"MajorVersion":1222352442,"MinorVersion":3904273100},"Timestamp":"2023-08-17T12:58:37.7531561Z","MessageType":"ua-deltaframe","Payload":{"dosingTime":{"Value":0,"SourceTimestamp":"2023-08-16T10:02:09.242Z","ServerTimestamp":"2023-08-16T10:02:09.242Z"}}}]},{"MessageId":"9f95a09f-bede-4bfc-b075-c201dfc1ee5f","MessageType":"ua-data","PublisherId":"railesIoThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":1 [... cut ...] oThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":12967,"MetaDataVersion":{"MajorVersion":1222352442,"MinorVersion":3904273100},"Timestamp":"2023-08-17T12:58:37.7560776Z","MessageType":"ua-deltaframe","Payload":{"shots":{"Value":69234,"SourceTimestamp":"2023-08-16T10:02:09.241Z","ServerTimestamp":"2023-08-16T10:02:09.241Z"}}}]}]

marcschier, 2023-08-22:
PublisherId behavior has changed to be more logically correct with the OPC UA standard. It can be set once per publisher using the command line variable and now represents the identifier of the OPC Publisher module.  

I think you are thinking of the DataSetWriterId value, which is part of each Message in "Messages".  (Note that if you are running with OPC Publisher in --strict mode than DataSetWriterId is now called DataSetWriterName which is per spec).  In backcompat mode it is the string you configured as DataSetWriterId or some hash identifier that OPC Publisher generates to distinguish multiple differently configured endpoints with same Url.  

DataSetWriterGroup or DataSetWriterId/Name can be configured in the publishednodes.json configuration per endpoint (https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/readme.md#configuration-schema), you can choose whatever names or strings you like here.  If you need contextualization information in each data set message, you can use the new Extension Fields (https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/readme.md#key-frames-delta-frames-and-extension-fields).

Ilheu, 2023-08-22:
**@marcschier Thank you for the help, but I change the DataSetWriterId in my file to a String, and it keeps the same output:
Am I doing something wrong?**

{
"EndpointUrl": "opc.tcp://192.168.131.21:4840",
"DataSetWriterId": "opc.tcp://192.168.131.21:4840", 
    "UseSecurity": false, 
    "OpcNodes": [ 
    
    
    [{"MessageId":"1fb7e39c-bbeb-4c2e-95e7-5466d89d9cad","MessageType":"ua-data","PublisherId":"railesIoThub_device_railesIoTedge%5Fvipex%5F00_module_OPCPublisher","DataSetWriterGroup":"<<UnknownWriterGroup>>","Messages":[{"DataSetWriterId":"<<UnknownDataSet>>_($a6a0845258a40575703021e5244ff9c70838a23b)","SequenceNumber":34445,"MetaDataVersion":{"MajorVersion":1222352431,"MinorVersion":3467188674},"Timestamp":"2023-08-17T12:59:09.8195591Z","MessageType":"ua-deltaframe","Payload":{"tempBlocoHR1":{"Value":41.1000023,"SourceTimestamp":"2023-08-16T02:27:39.0072053Z","ServerTimestamp":"2023-08-16T02:27:39.0072053Z"}}}]}]

marcschier, 2023-08-24:
@Ilheu , I assume this closes the issue. Let me know if not and we shall re-open.

closed 2023-08-24 by marcschier