Azure/Industrial-IoT #2039: Invoke direct Method and configure OPC Publisher through MQTT topic

SaifAlmaliki, 2023-08-28:
I am publishing my telemetry successsfully from opcpublisher to MQTT broker using --ttt (telemetrytopictemplate). 

Since MQTT v5 supports RPC which allows a client to call the OPC Publisher over MQTT. I want to Invoke and configure OPC Publisher through MQTT topic.

As mentioned in the doc [here](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/directmethods.md#:~:text=In%20addition%20to%20calling%20the%20API%20through%20HTTP%20REST%20calls%20(Preview)%20you%20can%20call%20the%20configuration%20API%20through%20MQTT%20v5%20(Preview)%20in%20OPC%20Publisher%202.9.): ` In addition to calling the API through HTTP REST calls (Preview) you can call the configuration API through MQTT v5 (Preview) in OPC Publisher 2.9.`

Following [commandline.md](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/commandline.md#:~:text=%2D%2Dttt%2C%20%2D%2Dtelemetrytopictemplate%2C%20%2D%2DTelemetryTopicTemplate%5B%3DVALUE%5D) , I configured my opcpublisher by adding --mtt(method topic tempelate) flag and append the method name after the topic template: xx/xxxx/methods/GetConfiguredEndpoints_V2.

However I wasn't able to invoke/Configure the opcpublisher

I need a demo or simple example showing end-to-end how to invoke and configure opcpublisher module using mqtt topic

thanks

marcschier, 2023-08-28:
This is correct. Can you provide me with the --mtt value you used so I can repro this?  Did you make sure you provide a response topic in the MQTT v5 message sent to the topic?

Samples are on the list of most requested feature. I hope I can get around to it after 2.9.1 is out.

SaifAlmaliki, 2023-08-28:
Thanks marcshier for your response.
In my case --mtt =“fi/publisher/methods/GetConfiguredEndpoints_V2”
But not sure how to provide the response topic in the MQTT message, I assumed the response will be on the same above topic. is in it?

marcschier, 2023-08-29:
The mtt option allows setting the root path for all methods.  Using a templating model.  When you use it though the actual topic to call for a method is the configured path, plus the method name.  So if you use --mtt x/y/z to call the method a, you send the payload to /x/y/z/a. If using mqtt 5 the payload needs to also specify the response topic. That can be any tpoc you have already subscribed to on the broker. Although I recommend to use a unique one per call. Your client library will allow you to specify that topic during publish, typically on the message object you publish. In mqttnet you do this as a property of the applocationmessage object you pass to Publish method.

marcschier, 2023-08-31:
I have tried the same and found a hole in our testing.  I also wrote a quick sample using mqttnet while at it.

marcschier, 2023-08-31:
Samples showing several calls via mqtt are now in the samples folder (in main). They only work with 2.9.1 which will be released soon.

closed 2023-08-31 by marcschier

SaifAlmaliki, 2023-08-31:
Thanks marcshier