Azure/Industrial-IoT #2041: Read init value for new nodes after calling SetConfiguredEndpoints

KenRenAtHilti, 2023-09-01:
**Is your feature request related to a problem? Please describe.**
After calling SetConfiugredEndpoints in 2.9.1, the new nodes would be in the State GoodNoData instead of the current value of the nodes if the value of the nodes is not changing during the updates of nodes.

**Describe the solution you'd like**
After new nodes are updated, the current value of the nodes should be read once for init value and published.(We are using heartbeat setting, so every timeout, if no data are sent, the latestest output would be sent still)

marcschier, 2023-09-01:
The ask is a flag per node/endpoint to read a value into the "lastValue" field of the monitored item, similar to Displayname so that it can be sent in keyframes or heartbeats.  This is something we can put on the 2.9.2 backlog if I have captured the ask correctly.  Marking as feature request.

KenRenAtHilti, 2023-09-01:
Thx. 
Would be better if a global command-line option is also provided. To read the monitored item for the first time when the publisher module is restarted or new nodes are added through some direct method like SetConfiguredEndpoints.

KenRenAtHilti, 2023-09-07:
The real issue for us is mentioned in the first part of the response and we need this fix ASAP. Current temp solution to solve this GoodNodata is to restart our broker and thus the data value is renewed to the current value.  

Regarding the behaviour of heartbeat, i would prefer that the output should reflect the current status of the monitored item for the purpose of transparency. This is the same kind of the request as for the bugfix: always knowing the current status of the desired node, either it is disconnected or "GOOD" when no value change happened.

marcschier, 2023-09-07:
Great, if you would not mind, could you test "mcr.microsoft.com/iotedge/opc-publisher:2.9.2-preview2" build to see if this fixes this issue for you or if you have more needs?

closed 2023-09-08 by marcschier

KenRenAtHilti, 2023-09-11:
Thx. It is tested and the fix works as expected.