Azure/Industrial-IoT #2051: PublishedNodes as Desired Properties

KenRenAtHilti, 2023-09-12:
**Is your feature request related to a problem? Please describe.**
Currently it is possible to update the nodes using DirectMethods. But still the nodes/endpoints are persisted as a mounted file. Would it be nice if the configuration would be persisted and monitored as module desired properties, which is more consistent.

**Describe the solution you'd like**
Using Desired properties of the module. The nodes are set through desired properties on the HUB for the device and the change of the properties would also be monitored. If changes happen, the monitored item on the module is also updated.

b-enigma-con, 2024-01-31:
Please note that the size of desired properties is fairly limited; it would not be possible to add large amounts of nodes.

marcschier, 2024-07-04:
We have decided to not use device twin for storage, we now have 3 different configuration apis that allow programmatic access including via the Azure portal to the publisher.  I have also added #2284 which provides a broader approach, if you have any feedback for this feature, please add, greatly appreciated.

closed 2024-07-04 by marcschier