Azure/Industrial-IoT #2053: Since Industrial IoT 2.8.6 expired on July 15, 2023, could someone please provide information about the current or latest version of Industrial IoT?

gauravsep1, 2023-09-13:
Since Industrial IoT 2.8.6 expired on July 15, 2023, could someone please provide information about the current or latest version of Industrial IoT?

marcschier, 2023-09-14:
2.9 is the latest version, specifically 2.9.1.  OPC Publisher 2.9 is generally a drop in replacement for 2.8, but we are providing extended support for OPC Publisher 2.8 until end of the year.  Industrial IoT Platform 2.8 is no longer supported, but most of the capabilities are available in the web-api project: https://github.com/Azure/Industrial-IoT/blob/main/docs/web-api/readme.md.

closed 2023-09-14 by marcschier

gauravsep1, 2023-09-14:
Thanks!!! @marcschier

gauravsep1, 2023-09-21:
@marcschier Based on your response, **2.9.1** is the latest version of the **Azure Industrial IoT suite**. Please correct me if I'm wrong.

marcschier, 2023-09-22:
Now it is 2.9.2, and it contains OPC Publisher edge module and the OPC Publisher companion web application.  The AKS/micro services cloud platform (last version was 2.8.6) is EOL.  The companion web application enables most of the scenarios of the micro services based version, but is not a drop in replacement, meaning work has to be done to move from 2.8.6 to 2.9.2.  Whereas the OPC Publisher edge module 2.9.2+ is intended to be a drop in replacement for 2.8.7+ OPC Publisher standalone version (run without any cloud platform and configured through pn.json).