Azure/Industrial-IoT #2059: Can't connect to OPC PLC server - BadServiceUnsupported

Freshchris01, 2023-09-19:
**Describe the bug**
I am trying to connect to a simulated OPC PLC server, following the [Postman tutorial](https://github.com/Azure/Industrial-IoT/blob/main/docs/web-api/tut-use-postman.md).
After the service discovery, the TCP connection inside the PLC server closes.

**To Reproduce**
Steps to reproduce the behavior:
1. Install the full deployment with `-type all -version 2.9`   
2. Create a VM for the PLC Server, open port 50000 for the TCP connection
3. After request as described [here](https://github.com/Azure/Industrial-IoT/blob/main/docs/web-api/tut-use-postman.md#register-the-demo-opc-ua-server-manually-using-its-discovery-url), the OCP PLC server throws errors:
```
 ██████╗ ██████╗  ██████╗    ██████╗ ██╗      ██████╗
██╔═══██╗██╔══██╗██╔════╝    ██╔══██╗██║     ██╔════╝
██║   ██║██████╔╝██║         ██████╔╝██║     ██║
██║   ██║██╔═══╝ ██║         ██╔═══╝ ██║     ██║
╚██████╔╝██║     ╚██████╗    ██║     ███████╗╚██████╗
 ╚═════╝ ╚═╝      ╚═════╝    ╚═╝     ╚══════╝ ╚═════╝

[09:53:49 INF] Current directory: /app
[09:53:49 INF] Log file: /app/cc8170bcfb50-plc.log
[09:53:49 INF] Log level: debug
[09:53:49 INF] OpcPlc v2.9.9 (SDK 1.4.372.56) starting up ...
[09:53:49 DBG] Informational version: v2.9.9.0+16d28ce723 (SDK 1.4.372.56+b90f76235135840774f898c59c20ff02ac994e2a from 08/25/2023 18:28:10)
[09:53:49 DBG] Build date: 09/18/2023 15:25:02
[09:53:49 INF] Alternate base addresses (for server binding and certificate DNSNames and IPAddresses extensions): ["opc.tcp://cc8170bcfb50:50000"]
[09:53:49 INF] Application Certificate store type is: Directory
[09:53:49 INF] Application Certificate store path is: pki/own
[09:53:49 INF] Rejection of SHA1 signed certificates is disabled
[09:53:49 INF] Minimum certificate key size set to 1024
[09:53:49 INF] Trusted Issuer store type is: Directory
[09:53:49 INF] Trusted Issuer Certificate store path is: pki/issuer
[09:53:49 INF] Trusted Peer Certificate store type is: Directory
[09:53:49 INF] Trusted Peer Certificate store path is: pki/trusted
[09:53:49 INF] Rejected certificate store type is: Directory
[09:53:49 INF] Rejected Certificate store path is: pki/rejected
[09:53:49 WRN] Automatically accepting all client certificates, this is a security risk!
[09:53:49 INF] Added security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode SignAndEncrypt
[09:53:49 INF] Added security policy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep with mode SignAndEncrypt
[09:53:49 INF] Added security policy http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss with mode SignAndEncrypt
[09:53:49 INF] Added security policy http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256 with mode Sign
[09:53:49 INF] Added security policy http://opcfoundation.org/UA/SecurityPolicy#Aes128_Sha256_RsaOaep with mode Sign
[09:53:49 INF] Added security policy http://opcfoundation.org/UA/SecurityPolicy#Aes256_Sha256_RsaPss with mode Sign
[09:53:49 INF] LDS(-ME) registration interval set to 0 ms (0 means no registration)
[09:53:49 INF] The OPC UA trace mask is set to: 0x3A5
[09:53:49 INF] No existing application certificate found. Creating a self-signed application certificate valid since yesterday for 12 months, with a 2048 bit key and 256 bit hash
[09:53:49 INF] Checking application instance certificate.
[09:53:49 INF] Creating application instance certificate.
[09:53:49 INF] Imported the PFX private key for [EAC988B443270D0EE81B79880F55DAF136BBA5DB].
[09:53:49 INF] Certificate created for urn:OpcPlc:cc8170bcfb50. [CN=OpcPlc] [EAC988B443270D0EE81B79880F55DAF136BBA5DB]
[09:53:49 INF] Application certificate with thumbprint EAC988B443270D0EE81B79880F55DAF136BBA5DB created
[09:53:49 INF] Application certificate is for ApplicationUri urn:OpcPlc:cc8170bcfb50, ApplicationName OpcPlc and Subject is CN=OpcPlc
[09:53:49 INF] Application store contains 1 certs
[09:53:49 INF] 01: Subject CN=OpcPlc (thumbprint: EAC988B443270D0EE81B79880F55DAF136BBA5DB)
[09:53:49 INF] Trusted issuer store contains 0 certs
[09:53:49 INF] Trusted issuer store has 0 CRLs
[09:53:49 INF] Trusted peer store contains 0 certs
[09:53:49 INF] Trusted peer store has 0 CRLs
[09:53:49 INF] Rejected certificate store contains 0 certs
[09:53:49 INF] Application configured with MaxSessionCount 100 and MaxSubscriptionCount 100
[09:53:49 INF] Starting server on endpoint opc.tcp://cc8170bcfb50:50000 ...
[09:53:49 INF] Simulation settings are:
[09:53:49 INF] One simulation phase consists of 50 cycles
[09:53:49 INF] One cycle takes 100 ms
[09:53:49 INF] Reference test simulation: Enabled
[09:53:49 INF] Simple events: Disabled
[09:53:49 INF] Alarms: Disabled
[09:53:49 INF] Deterministic alarms: Disabled
[09:53:49 INF] Anonymous authentication: Enabled
[09:53:49 INF] Reject chain validation with CA certs with unknown revocation status: Enabled
[09:53:49 INF] Username/Password authentication: Enabled
[09:53:49 INF] Certificate authentication: Enabled
[09:53:50 INF] Server - Start application OpcPlc.
[09:53:50 INF] Server - CreateResourceManager.
[09:53:50 INF] Server - CreateRequestManager.
[09:53:50 INF] Server - CreateMasterNodeManager.
[09:53:50 INF] MasterNodeManager.Startup - NodeManagers=5
[09:53:50 INF] Creating 1 GUID node(s) of type: UInt
[09:53:50 INF] Node values will change every 1000 ms
[09:53:50 DBG] NodeId type is System.Guid
[09:53:50 INF] Creating 1 Fast nodes of type: UInt
[09:53:50 INF] Node values will change every 1000 ms
[09:53:50 INF] Node values sampling rate is 0 ms
[09:53:50 INF] Creating 1 BadFast nodes of type: UInt
[09:53:50 INF] Node values will change every 1000 ms
[09:53:50 INF] Node values sampling rate is 0 ms
[09:53:50 DBG] NodeId type is System.Byte[]
[09:53:50 INF] Creating 1 Slow nodes of type: UInt
[09:53:50 INF] Node values will change every 10000 ms
[09:53:50 INF] Node values sampling rate is 0 ms
[09:53:50 INF] Creating 1 BadSlow nodes of type: UInt
[09:53:50 INF] Node values will change every 10000 ms
[09:53:50 INF] Node values sampling rate is 0 ms
[09:53:51 INF] Server - CreateEventManager.
[09:53:51 INF] Server - CreateAggregateManager.
[09:53:51 INF] Server - CreateSessionManager.
[09:53:51 INF] Server - Session Monitor Thread Started.
[09:53:51 INF] Server - CreateSubscriptionManager.
[09:53:51 INF] Subscription - Publish Thread 00000009 Started.
[09:53:51 INF] Subscription - ConditionRefresh Thread 0000000A Started.
[09:53:51 INF] Server - Enter Running state.
[09:53:51 INF] Server - Started.
[09:53:51 INF] OPC UA Server started
[09:53:51 INF] PLC simulation started, press Ctrl+C to exit ...
[09:53:55 INF] ChannelId 1: in Connecting state.
[09:53:55 INF] TCPSERVERCHANNEL SOCKET ATTACHED: 01FA0ABC, ChannelId=1
[09:53:55 INF] ChannelId 1: in Opening state.
[09:53:55 INF] Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
[09:53:55 INF] Sender Certificate: (none)
[09:53:55 INF] ChannelId 1: Token #0 created. CreatedAt=09:53:55.068. Lifetime=3600000.
[09:53:55 INF] SECURE CHANNEL CREATED [.NET Standard ServerChannel UA-TCP 1.4.372.56] [ID=1] Connected To: opc.tcp://cc8170bcfb50:50000/
[09:53:55 INF] ChannelId 1: Token #1 activated. CreatedAt=09:53:55.068. Lifetime=3600000.
[09:53:55 INF] ChannelId 1: in Open state.
[09:53:55 WRN] SERVER - Service Fault Occurred. Reason=BadServiceUnsupported
[09:53:55 INF] TCPSERVERCHANNEL ProcessCloseSecureChannelRequest success, ChannelId=1, TokenId=1, Socket=01FA0ABC
[09:53:55 INF] ChannelId 2: in Connecting state.
[09:53:55 INF] TCPSERVERCHANNEL SOCKET ATTACHED: 03161E6F, ChannelId=2
[09:53:55 INF] ChannelId 2: in Opening state.
[09:53:55 INF] Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
[09:53:55 INF] Sender Certificate: (none)
[09:53:55 INF] ChannelId 2: Token #0 created. CreatedAt=09:53:55.183. Lifetime=3600000.
[09:53:55 INF] SECURE CHANNEL CREATED [.NET Standard ServerChannel UA-TCP 1.4.372.56] [ID=2] Connected To: opc.tcp://cc8170bcfb50:50000/
[09:53:55 INF] ChannelId 2: Token #1 activated. CreatedAt=09:53:55.183. Lifetime=3600000.
[09:53:55 INF] ChannelId 2: in Open state.
[09:53:55 WRN] SERVER - Service Fault Occurred. Reason=BadServiceUnsupported
[09:53:55 INF] ChannelId 1: in Closed state.
[09:53:55 INF] TCPSERVERCHANNEL ProcessCloseSecureChannelRequest success, ChannelId=2, TokenId=1, Socket=03161E6F
[09:53:55 INF] ChannelId 2: in Closed state.
[09:53:55 INF] ChannelId 2: closed
[09:53:55 INF] ChannelId 1: closed
[09:53:55 ERR] TCPSERVERCHANNEL ForceChannelFault Socket=FFFFFFFF, ChannelId=2, TokenId=1, Reason=BadConnectionClosed 'Remote side closed connection.'
[09:53:55 ERR] TCPSERVERCHANNEL ForceChannelFault Socket=FFFFFFFF, ChannelId=1, TokenId=1, Reason=BadConnectionClosed 'Remote side closed connection.'
[09:53:55 INF] ChannelId 2: in Faulted state.
[09:53:55 INF] ChannelId 1: in Faulted state.

```
4. Because of these errors, I am not able to activate the endpoints.

Am I missing something?

marcschier, 2023-09-19:
I looked at the tutorials and noticed that they have not been updated to 2.9, there is some work to do. Still the CLI tutorial should technically work, but the output will look different: https://github.com/Azure/Industrial-IoT/blob/main/docs/web-api/tut-use-cli.md.  

There was also a new version of the PLC published, but I am fairly certain the error you are seeing is the attempt to call FindServerOnNetwork (to see if the server is a fully fledged LDS) during discovery, which it does not support (because it is none), so it should be ok and not the issue here.  There are no sessions opened once discovery finishes, so once it is done, the connection is closed.  So, the question is if the discovery operation worked and the OPC Publisher created the respective entries in IoT Hub.  You can test this by using `apps list` in the CLI or call the GetApplications operation on the controller.  If so, then it should be possible to subscribe or call methods, etc. on it.

Freshchris01, 2023-09-19:
Okay, thanks a lot. I can see the api changed. A mix of CLI and Postman brought me further.

marcschier, 2023-09-20:
The biggest change is the removal of everything "job" and "worker" related from the API (Apps should not have needed them in the first place), and the "Activate" and "Deactivate" API.  For the latter, while the HTTP resource path is the same, both methods have entirely different meaning now.  They are not necessary either, you can just skip over that part of the tutorial.  

For the remaining capabilities, we have end to end tests that sit on top of the HTTP API directly, and they are passing on 2.8 and 2.9.  If, however, you see other changes that we might not be covering through our end-to-end tests, I would be curious to learn about those as we can explore then how to align them to past behavior.

marcschier, 2023-09-28:
We decided to remove the Postman tutorial.  The new HTTP API samples to call OPC Publisher directly, or use the IoT Hub samples to call API through IoT Hub are a partial replacement.  Furthermore, the CLI sample and openapi.json might be a good starting point for how to use the web-api.

closed 2023-09-28 by marcschier