Azure/Industrial-IoT #2065: For a new configured in pn.json Dataset publisher must try to reuse an existing session for this EndpointUrl with the identical security settings (if exists).

dimtix, 2023-09-21:
Let's assume that in a pn.json several datasets are configured for the server with the same security settings. Then the publisher should create only one session for all these datasets.

The reason why it is important. Currently publisher creates a new session every time for each configured dataset. But the maximum number of sessions is usually limited per server. In addition, the maximum number of sessions per client is often limited as well. For a custom event you usually need to define a new dataset. So, the number of sessions can grow almost unlimited and so at some point the limit is reached, so that either only the publisher can't subscribe to any nodes by this server anymore or even the whole server doesn't accept connections from other clients too anymore.

In our use case the problem is even more critical. All our servers in a shopfloor are aggregated by a gateway running on an edge device. The publisher runs on the edge device too and subscribes to nodes of aggregated servers via the aggregated gateway address space tree. We configure multiple datasets (dataitems and event/alerts) per each aggregated server and all datasets logically go to the same EndpointUrl – the Url of the gateway. So, the maximum number of open sessions on the gateway for the publisher is reached very fast.

marcschier, 2023-09-22:
@dimtix , would you be so kind and provide a (hypothetical) configuration so I can validate?  We have done quite some work from 2.8 to ensure that sessions are shared across multiple endpoint configurations if the configuration is the same, so maybe it is a configuration issue or it is a bug or we are missing a feature :-)

marcschier, 2023-09-22:
While writer group is tied to a session (the only way to partition across multiple sessions if needed), the same thing should not happen when using the same writer group (e.g., default) but different data set writers.  But I was able to reproduce what you are seeing. I am investigating and we will fix in 2.9.3 unless you are blocked.

marcschier, 2023-09-22:
How it is supposed to behave is as follows:  DataSetWriterGroups are tied to Session, DataSetWriterIds are tied to Subscriptions.  If you specify multiple data set writer groups (with otherwise same endpoint configuration) then you get multiple sessions opened, if you use the same data set writer group (e.g., default), but configure differently named writers, then those are part of the same session.  I created a couple configurations to test this out, but if you have one that behaves differently, would be great to get and use to repro.

What I can see might be absolutely useful to have the ability to re-use a session across multiple writer groups.  But to understand what we can do there to support your use case, it would be great to learn more about your setup and output mapping needs.

dimtix, 2023-09-22:
You have right - my first test (afterr we had an idea that we have a problem mit sessions) was with different DataSetWriterGroups:
<details>
  <summary>pn.test.multipledatasets.json</summary>
  
  
  ```json
[
    {
        "EndpointUrl": "opc.tcp://localhost:4840",
        "UseSecurity": false,
        "DataSetWriterId": "1",
        "DataSetWriterGroup": "1",
        "OpcNodes": [
            {
                "Id": "nsu=http://opcfoundation.org/UA/;i=2256",
                "OpcSamplingInterval": 60000
            },
            {
                "Id": "nsu=http://opcfoundation.org/UA/;i=2255",
                "HeartbeatInterval": 10
            }
        ]
    },
    {
        "EndpointUrl": "opc.tcp://localhost:4840",
        "UseSecurity": false,
        "DataSetWriterId": "2",
        "DataSetWriterGroup": "2",
        "OpcNodes": [
            {
                "Id": "nsu=http://opcfoundation.org/UA/;i=2256",
                "OpcSamplingInterval": 60000
            },
            {
                "Id": "nsu=http://opcfoundation.org/UA/;i=2255",
                "HeartbeatInterval": 10
            }
        ]
    },
    {
        "EndpointUrl": "opc.tcp://localhost:4840",
        "UseSecurity": false,
        "DataSetWriterId": "3",
        "DataSetWriterGroup": "3",
        "OpcNodes": [
            {
                "Id": "nsu=http://opcfoundation.org/UA/;i=2256",
                "OpcSamplingInterval": 60000
            },
            {
                "Id": "nsu=http://opcfoundation.org/UA/;i=2255",
                "HeartbeatInterval": 10
            }
        ]
    },
    {
        "EndpointUrl": "opc.tcp://localhost:4840",
        "UseSecurity": false,
        "DataSetWriterId": "4",
        "DataSetWriterGroup": "4",
        "OpcNodes": [
            {
                "Id": "nsu=http://opcfoundation.org/UA/;i=2256",
                "OpcSamplingInterval": 60000
            },
            {
                "Id": "nsu=http://opcfoundation.org/UA/;i=2255",
                "HeartbeatInterval": 10
            }
        ]
[... 270 lines cut ...]
        "UseSecurity": false,
        "DataSetWriterId": "server3/alarm1",
        "DataSetWriterGroup": "server3",
        "OpcNodes": [
            {
                "Id": "nsu=http://opcfoundation.org/UA/;i=2256",
                "HeartbeatInterval": 10
            }
        ]
    },
    {
        "EndpointUrl": "opc.tcp://localhost:4840",
        "UseSecurity": false,
        "DataSetWriterId": "server3/event1",
        "DataSetWriterGroup": "server3",
        "OpcNodes": [
            {
                "Id": "nsu=http://opcfoundation.org/UA/;i=2257",
                "OpcSamplingInterval": 60000
            }
        ]
    }
]
  ```
</details>

marcschier, 2023-09-22:
@dimtix, if I understand your use case correctly, you would like flexible routing using DataSetWriterGroup, where network messages are sent to different consumers.  Yet use a single session for all writer groups.  

The simplest option is to remove tying a WriterGroup to a Session (potentially a command line option to prevent backwards compatibility breakage).  Instead providing another way of splitting sessions, e.g., by adding a "Tag" to split an endpoint configuration into multiple sessions. Let's see how I can get this into 2.9.3.

dimtix, 2023-09-25:
Exactly - you are right - the best solution may be would be to have a CLI switch to enable / remove tying a WriterGroup to a Session.

marcschier, 2023-09-25:
#2069 implements said CLI switch (--dsg) and is now in main.

closed 2023-09-25 by marcschier