Azure/Industrial-IoT #2066: The configuration of the OpcPublisher 1.9.2 fails using the REST Api

Haeavar, 2023-09-22:
**Describe the bug**
The configuration of the OpcPublisher 2.9.2 fails using the REST Api with the configuration sample documented here - https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/directmethods.md#addorupdateendpoints_v1

(There are no current datasetwriter configured)

REST PATCH /v2/configuration
BODY:
`[
   {
      "EndpointUrl": "opc.tcp://opcplc:50000",
      "DataSetWriterGroup": "Asset1",
      "DataSetWriterId": "DataFlow1",
      "DataSetPublishingInterval": 5000,
      "OpcNodes": [
         {
            "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt0"
         }
      ]
   },
   {
      "EndpointUrl": "opc.tcp://opcplc:50001",
      "DataSetWriterGroup": "Asset2",
      "DataSetWriterId": "DataFlow2",
      "OpcNodes": []
   }
]`

response: statuscode 500 without any body. Even the exception in the log does not contains any helpful information
`
[10:36:41.0305] fail: Microsoft.AspNetCore.Server.Kestrel[13]
      => SpanId:fdfb6c99fdbbc48f, TraceId:d2658c25171e5fd652de78b6ec4dcc32, ParentId:0000000000000000 => ConnectionId:0HMTRAR922013 => RequestPath:/v2/configuration RequestId:0HMTRAR922013:00000002
      Connection id "0HMTRAR922013", Request id "0HMTRAR922013:00000002": An unhandled exception was thrown by the application.
      System.InvalidOperationException: The type 'System.ReadOnlySpan`1[System.Byte]' of property 'Span' on type 'System.ReadOnlyMemory`1[System.Byte]' is invalid for serialization or deserialization because it is a pointer type, is a ref struct, or contains generic parameters that have not been replaced by specific types.
         at System.Text.Json.ThrowHelper.ThrowInvalidOperationException_CannotSerializeInvalidType(Type typeToConvert, Type declaringType, MemberInfo memberInfo)
         at System.Text.Json.Serialization.Metadata.ReflectionJsonTypeInfo`1.CreateProperty(Type typeToConvert, MemberInfo memberInfo, JsonSerializerOptions options, Boolean shouldCheckForRequiredKeyword)
         at System.Text.Json.Serialization.Metadata.ReflectionJsonTypeInfo`1.CacheMember(Type typeToConvert, MemberInfo memberInfo, Boolean& propertyOrderSpecified, Dictionary`2& ignoredMembers, Boolean shouldCheckForRequiredKeyword)
         at System.Text.Json.Serialization.Metadata.ReflectionJsonTypeInfo`1.LateAddProperties()
         at System.Text.Json.Serialization.Metadata.JsonTypeInfo.InitializePropertyCache()
         at System.Text.Json.Serialization.Metadata.JsonTypeInfo.Configure()
         at System.Text.Json.Serialization.Metadata.JsonTypeInfo.<EnsureConfigured>g__ConfigureLocked|143_0()
      --- End of stack trace from previous location ---
         at System.Text.Json.Serialization.Metadata.JsonTypeInfo.<EnsureConfigured>g__ConfigureLocked|143_0()
         at System.Text.Json.Serialization.Metadata.JsonPropertyInfo.get_JsonTypeInfo()
         at System.Text.Json.WriteStackFrame.GetNestedJsonTypeInfo()
         at System.Text.Json.WriteStack.Push()
         at System.Text.Json.Serialization.JsonConverter`1.TryWrite(Utf8JsonWriter writer, T& value, JsonSerializerOptions options, WriteStack& state)
         at System.Text.Json.Serialization.Metadata.JsonPropertyInfo`1.GetMemberAndWriteJson(Object obj, WriteStack& state, Utf8JsonWriter writer)
         at System.Text.Json.Serialization.Converters.ObjectDefaultConverter`1.OnTryWrite(Utf8JsonWriter writer, T value, JsonSerializerOptions options, WriteStack& state)
         at System.Text.Json.Serialization.JsonConverter`1.WriteCore(Utf8JsonWriter writer, T& value, JsonSerializerOptions options, WriteStack& state)
         at System.Text.Json.Serialization.JsonConverter`1.WriteCoreAsObject(Utf8JsonWriter writer, Object value, JsonSerializerOptions options, WriteStack& state)
         at System.Text.Json.JsonSerializer.WriteCore[TValue](Utf8JsonWriter writer, TValue& value, JsonTypeInfo jsonTypeInfo, WriteStack& state)
         at System.Text.Json.JsonSerializer.WriteStreamAsync[TValue](Stream utf8Json, TValue value, JsonTypeInfo jsonTypeInfo, CancellationToken cancellationToken)
         at System.Text.Json.JsonSerializer.WriteStreamAsync[TValue](Stream utf8Json, TValue value, JsonTypeInfo jsonTypeInfo, CancellationToken cancellationToken)
         at System.Text.Json.JsonSerializer.WriteStreamAsync[TValue](Stream utf8Json, TValue value, JsonTypeInfo jsonTypeInfo, CancellationToken cancellationToken)
         at Microsoft.AspNetCore.Mvc.Formatters.SystemTextJsonOutputFormatter.WriteResponseBodyAsync(OutputFormatterWriteContext context, Encoding selectedEncoding)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeResultAsync>g__Logged|22_0(ResourceInvoker invoker, IActionResult result)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeNextResultFilterAsync>g__Awaited|30_0[TFilter,TFilterAsync](ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.Rethrow(ResultExecutedContextSealed context)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.ResultNext[TFilter,TFilterAsync](State& next, Scope& scope, Object& state, Boolean& isCompleted)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.InvokeAlwaysRunResultFilters()
      --- End of stack trace from previous location ---
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeFilterPipelineAsync>g__Awaited|20_0(ResourceInvoker invoker, Task lastTask, State next, Scope scope, Object state, Boolean isCompleted)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Logged|17_1(ResourceInvoker invoker)
         at Microsoft.AspNetCore.Mvc.Infrastructure.ResourceInvoker.<InvokeAsync>g__Logged|17_1(ResourceInvoker invoker)
         at Microsoft.AspNetCore.Routing.EndpointMiddleware.<Invoke>g__AwaitRequestTask|6_0(Endpoint endpoint, Task requestTask, ILogger logger)
         at Swashbuckle.AspNetCore.SwaggerUI.SwaggerUIMiddleware.Invoke(HttpContext httpContext)
         at Swashbuckle.AspNetCore.Swagger.SwaggerMiddleware.Invoke(HttpContext httpContext, ISwaggerProvider swaggerProvider)
         at Microsoft.AspNetCore.Authorization.AuthorizationMiddleware.Invoke(HttpContext context)
         at Microsoft.AspNetCore.Authentication.AuthenticationMiddleware.Invoke(HttpContext context)
         at Microsoft.AspNetCore.Server.Kestrel.Core.Internal.Http.HttpProtocol.ProcessRequests[TContext](IHttpApplication`1 application)`

**Expected behavior**
As stated in the documentation the response should contains an error message or the OcPublisher log should contains an exception that the request payload contains deletion request for an endpoint (DataSetWriter) that isn't present in publisher configuration.

marcschier, 2023-09-27:
The code tries to serialize the exception. This plus adding samples and making sal working will be fixed in 2.9 3.

marcschier, 2023-09-28:
Please see the new samples in the samples folder showing how to use http api. Please note that these samples only work with the opc publisher in main, which will release as 2.9.3.

closed 2023-09-28 by marcschier