Azure/Industrial-IoT #2072: Azure services like Azure Cosmos DB, Service Bus, Event Hubs, SignalR Service and Time Series Insights are not deployed after deploying the Azure IIOT 2.9.2

gauravsep1, 2023-09-28:
**Describe the bug**
Azure services like Azure Cosmos DB, Service Bus, Event Hubs, SignalR Service and Time Series Insights are not deployed after deploying the Azure IIOT 2.9.2

**To Reproduce**
Steps to reproduce the behavior:
1. Download the code in local machine using git clone https://github.com/Azure/Industrial-IoT
cd Industrial-IoT
2. Run the poweshell ./aad-register.ps1 -Name testapp -ReplyUrl https://sampleapp.azurewebsites.net/ -TenantId ************************* -Output aad.json
3. After that run the powershell ./deploy.ps1 -version 2.9.2 -type all -aadConfig aad.json-applicationName sampleapp
4. Error : Azure Cosmos DB, Service Bus, Event Hubs, SignalR Service and Time Series Insights are not deployed after deplying the Azure IIOT 2.9.2

**Expected behavior**
Azure Cosmos DB, Service Bus, Event Hubs, SignalR Service and Time Series Insights should be deployed when run the powershell command 
./deploy.ps1 -version 2.9.2 -type all -aadConfig aad.json-applicationName sampleapp
 deplying the  Azure IIOT 2.9.2.

Freshchris01, 2023-09-28:
From my understanding, this is expected due to the changes from 2.8.X to 2.9, see [here](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/migrationpath.md#breaking-changes).

marcschier, 2023-09-28:
Correct.

closed 2023-09-28 by marcschier