Azure/Industrial-IoT #2073: UI is not accessible after deployment of Azure IIOT 2.9.2

gauravsep1, 2023-09-28:
**Describe the bug**
UI is not accessible after deployment of Azure IIOT 2.9.2. After deployment of azure IIoT 2.9.2, script will return the ui URL and using that url we can access the UI. 

**To Reproduce**
Steps to reproduce the behavior:
1. Download the code in local machine using git clone https://github.com/Azure/Industrial-IoT
cd Industrial-IoT
2. Run the poweshell ./aad-register.ps1 -Name testapp -ReplyUrl https://sampleapp.azurewebsites.net/ -TenantId ************************* -Output aad.json
3. After that run the powershell ./deploy.ps1 -version 2.9.2 -type all -aadConfig aad.json-applicationName sampleapp
4. Error : UI is not accessible after deployment of Azure IIOT 2.9.2.

**Expected behavior**
UI should be accessible after deployment of Azure IIOT 2.9.2.

**Screenshots**
[image]

Freshchris01, 2023-09-28:
The engineering tool isn't available since 2.9.X, see [this comment](https://github.com/Azure/Industrial-IoT/issues/2022#issuecomment-1656568178).

marcschier, 2023-09-28:
This is correct. The engineering tool is in the engineeringtool branch.  If there is interest in making it work, it would be a great addition to the "samples" folder.

closed 2023-09-28 by marcschier