Azure/Industrial-IoT #2083: I would like to view the architecture diagram of the Azure IIoT 2.9.2 platform. Could someone please guide me on where I can find it?

gauravsep1, 2023-10-06:
**Is your feature request related to a problem? Please describe.**
I would like to view the architecture diagram of the Azure IIoT 2.9.2 platform. Could someone please guide me on where I can find it?

**Describe the solution you'd like**
I would like to view the architecture diagram of the Azure IIoT 2.9.2 platform. Could someone please guide me on where I can find it?

marcschier, 2023-10-06:
Thanks for pointing out that this is missing. You can find the architecture diagram under https://github.com/Azure/Industrial-IoT/tree/main/docs.

gauravsep1, 2023-10-06:
> Thanks for pointing out that this is missing. You can find the architecture diagram under https://github.com/Azure/Industrial-IoT/tree/main/docs.

@marcschier In the provided architecture, using the link https://github.com/Azure/Industrial-IoT/tree/main/docs, I cannot find the Azure Storage, Azure Key Vault services, and Application Insights. However, when I deployed the Azure IIOT 2.9.2 platform in services mode, these services were also deployed. Please provide me with an updated architecture.

closed 2023-10-10 by marcschier