Azure/Industrial-IoT #2087: I want to provide the custom name for the IoT Hub, storage and Key Vault  when deploying the Azure IIoT 2.9.2 platform. Can someone guide me on how I can do that?

gauravsep1, 2023-10-12:
**Describe the bug**
I want to provide the custom name for the IoT Hub, storage and Key Vault  when deploying the Azure IIoT 2.9.2 platform. Can someone guide me on how I can do that?

**To Reproduce**
Steps to reproduce the behavior:
1. Download the code in local machine using git clone https://github.com/Azure/Industrial-IoT
cd Industrial-IoT
2. Run the poweshell ./aad-register.ps1 -Name testapp -ReplyUrl https://sampleapp.azurewebsites.net/ -TenantId ************************* -Output aad.json
3. After that run the powershell ./deploy.ps1 -version 2.9.2 -type all -aadConfig aad.json-applicationName sampleapp
4. After deploying Azure IIoT 2.9.2 in the Azure portal, the IoT hub is named 'iothub-6qwp6t.' However, I would like to choose a custom name for the IoT hub, such as 'sampleiothub.' I'd appreciate guidance on how to achieve this.

**Expected behavior**
I would like to choose a custom name for the IoT hub, such as 'sampleiothub.' I'd appreciate guidance on how to achieve this.

marcschier, 2023-10-12:
The deploy.ps1 script is just a convenient wrapper around the ARM templates in the templates folder. 

However, you can deploy azuredeploy.json those yourself using az cli or portal.  Right now the resource names are only parameters of an inner template, they would need to be added to the outer template and plumbed through to the inner.  Contributions welcome.

marcschier, 2023-11-18:
I checked and azuredeploy.json has parameters to name the requested resources.  Closing as complete.

closed 2023-11-18 by marcschier