Azure/Industrial-IoT #2088: How to authenticate with CLI container image

Freshchris01, 2023-10-18:
I try to use the `mcr.microsoft.com/iot/industrial-cli:latest` container image to interact with my industrial IoT 2.9 deployment.
I found out that I can add my commands when running the image to execute them, e.g. `docker run mcr.microsoft.com/iot/industrial-cli:latest apps list`.
To get the authentication right, I need to pass 2 things, the `.env` file and Azure credentials/auth information.
I tried passing the env file to the container by mounting the file, but how can I pass the Azure login information?
Or am I using the container in a wrong way?

Mounting the `.env`file:
`docker run --mount type=bind,source=<path_to_repo>/Industrial-IoT/.env,target=/.env mcr.microsoft.com/iot/industrial-cli:latest  apps list
`

marcschier, 2023-10-18:
Aren't you prompted for your credentials when you run the command?  Can you try using the 'console' run mode and enter 'apps list'? What error do you see?

marcschier, 2023-10-19:
Thanks for the additional information!  The key vault contains the aad info to set up the confidential client to access the rest API.  Azure identity is used to access key vault. This can be an MSI, or the credential in the credential cache of az CLI or ps. If run inside the container you have 2 options.  You can export the content of the key vault into the .env file.  There is a sample script here: . https://github.com/Azure/Industrial-IoT/blob/main/tools/e2etesting/GetSecrets.ps1. Then you can mount the .env file like you did.  Drawback, now you have secrets in .env file.  Or, you can provide the Azure identity credential to access key vault as environment variable (https://learn.microsoft.com/en-us/dotnet/api/azure.identity.environmentcredential?view=azure-dotnet).  This is a bit better, but I recommend to run the container in production with access to the MSI endpoint (like the REST API web app container which runs in app service with access to the MSI) Caveat: I have not tested it.

closed 2023-11-18 by marcschier