Azure/Industrial-IoT #2090: Message mode PubSub with Json message encoding has value in Binary encoding for complex (multilevel structure) data type

nelsonmorais, 2023-10-19:
**Describe the bug**
We have OPC Publisher v2.9.2 running with the following command line parameters `--strict --mm=PubSub --me=Json`.

We're subscribing to an (internally developed) OPC-UA Server on a node that is a Variable with a DataType of a custom Structure which is made of multiple levels (custom Structure that has fields that are also a custom Structure), and where one of the fields on the inner and deepest level Structures is an optional Field.

**Bug**: We're publishing the messages to an Azure IoT Hub and **observed that the serialization of the Node of this multilevel Structure DataType is not serialized as Json, instead the value is showing the Binary encoding**. Other custom Structures with a single level (so none of the Structure Fields point to another custom Structure) are serialized properly.

Here's a sample of the serialization observed on the IoT Hub (see the value of `ns=1;s=PostBondingInspection` that shows the Binary encoding instead of Json, like the other custom Structures on the same message payload:
```Json
{
  "body": {
    "MessageId": "6178ab9f-7b4c-436f-8027-7de6d87b79f1",
    "MessageType": "ua-data",
    "PublisherId": "xxx_module_msft-opc-publisher",
    "DataSetWriterGroup": "EVO3",
    "Messages": [
      {
        "MetaDataVersion": {
          "MajorVersion": 1222365154,
          "MinorVersion": 2465399015
        },
        "MessageType": "ua-deltaframe",
        "DataSetWriterName": "DSW_EVO3",
        "Payload": {
          "ns=1;s=PostBondingInspection": {
            "Value": "YMBavekB2gEGAAAAMTEyMjU2AiQAAAAwODI0MTEyMi0wMTdiLWI0NzMtMDMxNi1jYmEwOTlkM2IwZWUAAAAAAAAAAAAAAAABAAAACQAAAFBEQV9CQVNFRAEAAAAAAAAAAAAAAAAAAAAAAAAAEgAAAEZDXzIwMjNfMDlfMDVfMDAxNAEAAAD6////AAAAAAYAAAAAAAAAAQAAAAAAAAAMAAAAWUdBR19QQkktcGJpBwAAAFVOS05PV04GAAAAKm51bGwqBgAAACpudWxsKgYAAAAqbnVsbCoGAAAAKm51bGwqBgAAACpudWxsKgYAAAAqbnVsbCpjAAAACgAAAF9MT1RfTk9ORV8AAAAAAAAAAAEAAADKembt6uEZwGrPzXBdFNw/fWr87QZ0G0AGAAAAKm51bGwq",
            "SourceTimestamp": "2023-07-09T05:10:33.5197086Z"
          },
          "ns=1;s=TransportUnitEnd": {
            "Value": {
              "Clock": "2023-10-18T17:37:21.15Z",
              "UserName": "112256",
              "TransportUnitEndSystemId": 1,
              "TuEndId": "*null*",
              "LotId": "_LOT_NONE_"
            },
            "SourceTimestamp": "2023-07-09T05:10:33.5277084Z"
          },
          "ns=1;s=DieHasBeenPicked": {
            "Value": {
              "Clock": "2023-10-18T17:37:01.26Z",
              "UserName": "112256",
              "DiePickMid": "*null*",
              "PickupVacuum": -1,
              "ComponentId": "27241122-017b-b4d6-0316-ef2e483eccd5",
              "EjectorLifetimeCounter": -1
            },
            "SourceTimestamp": "2023-07-09T05:10:33.5277084Z"
          },
          "ns=1;s=TransportUnitStart": {
            "Value": {
              "Clock": "2023-10-18T17:37:23.47Z",
              "UserName": "112256",
              "TransportUnitStartSystemId": 2,
              "TuStartId": "*null*",
              "LotId": "_LOT_NONE_"
            },
            "SourceTimestamp": "2023-07-09T05:10:33.5277084Z"
          },
          "ns=1;s=ProcessStateMachineChangedState": {
            "Value": {
              "Clock": "2023-10-18T17:37:25.25Z",
              "UserName": "112256",
              "ProcessState": 7
            },
            "SourceTimestamp": "2023-07-09T05:10:33.5277084Z"
          }
        }
      }
    ]
[... 147 lines cut ...]
        public TwoDCartesianCoordinatesInt32 PbiOffset { get; set; }
        public TwoDCartesianCoordinatesInt32 DiePlaceSub { get; set; }
        public string PbiDataset { get; set; }
        public string PbiType { get; set; }
        public string HostData1 { get; set; }
        public string HostData2 { get; set; }
        public string HostData3 { get; set; }
        public string HostData4 { get; set; }
        public string HostData5 { get; set; }
        public string PackageId { get; set; }
        public int PbiQuality { get; set; }
        public string LotId { get; set; }
        public double PbiArea { get; set; }
        public TwoDCartesianCoordinatesDouble PbiOffsetDouble { get; set; }
        public string PbiStepLayoutSpecifier { get; set; }
    }
```

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04.6 LTS (Bionic Beaver)
 - OPC Publisher Version: 2.9.2.

**Additional context**
The problem seems to be similar to the one reported here: #1334 that was closed without conclusion.
@koepalex, @mregen we can provide the information model file (*.Nodeset2.xml) on a direct message if needed.

nelsonmorais, 2023-10-19:
Updated the original description to include the OPC Publisher metadata message as it looks like it might contain relevant information for the bug analysis.

nelsonmorais, 2023-10-20:
Hi @marcschier I've seen that you've labeled this issue to be addressed on v2.9.3, is there an ETA that you can share?
Thanks!

marcschier, 2023-10-20:
> Hi @marcschier I've seen that you've labeled this issue to be addressed on v2.9.3, is there an ETA that you can share? Thanks!

Goal is to release 2.9.3 end of November, but since this bug must be addressed in the OPC UA stack dependency, we might need to push this out to 2.9.4 if we do not have an updated nuget before end of November.

nelsonmorais, 2023-10-20:
@marcschier  Thanks for the update!

Given that this issue might take some time to be fixed and this is blocking us to get access to an important set of data form one of our machines that we've just connected and for which we were not expecting this issue, is there any workaround that you could share that would allow us to solve the problem in the meantime, so that we can get access to the data?
 
If not, any "push" to have this issue addressed sooner would be greatly appreciated 😄 

From our end, and if possible, we could even consider to temporarily do and use a private build from the source on this repository once the fix is available from your end and a final release (v2.9.3, v2.9.4) is not published yet.

marcschier, 2023-10-21:
Once the fix is checked in Using the preview nuget of the stack.   This will help confirm that the issue is really addressed.

mregen, 2023-10-26:
Hi @nelsonmorais, please provide your Nodeset2 to test my fix. I couldn't reach you on yahoo mail. Please feel free to send to my mregen*AT*microsoft.com address.

mregen, 2023-10-26:
Hi @nelsonmorais, I *think* the issue is fixed in the stack PR because there are types derived from the CartesianCoordinates abstract type used for some structure members, which is the same issue. But to be sure it needs additional verification.

nelsonmorais, 2023-10-26:
@mregen Thanks for the update! 
Let me know when and where can I get access to a _preview_ build of the OPC Publisher that I can try to let you know if the problem is resolved at our end.

marcschier, 2023-10-27:
Reopening to track the integration of the preview stack nuget.

marcschier, 2023-11-13:
A latest build is available at mcr.microsoft.com/iotedge/opc-publisher:2.9.3-preview2.  Closing, please re-open if issues persist.

closed 2023-11-13 by marcschier

nelsonmorais, 2023-11-13:
Thanks @marcschier!

PR #2094 (2023-10-28): Bump to Ua stack 1.4.372.76

Release version of UA Nuget package

for diff with preview package: Full Changelog: https://github.com/OPCFoundation/UA-.NETStandard/compare/1.4.372.67-preview...1.4.372.76

fixes #2090