Azure/Industrial-IoT #2092: Extension field values show up wrong in samples mode.

marcschier, 2023-10-20:
Extension field values are incorrectly serialized in samples mode:

`
ExtensionFields": {
      "WriterGroupId": "Asset1",
      "FactoryId": "[FactoryId, { \"Value\": \"11\", \"Raw\": \"11\", \"IsObject\": False, \"IsArray\": False, \"IsBoolean\": False, \"IsString\": True, \"IsBytes\": False, \"IsDecimal\": True, \"IsDouble\": True, \"IsFloat\": True, \"IsInt64\": True, \"IsUInt64\": True, \"IsInteger\": True, \"IsGuid\": False, \"IsDateTime\": False, \"IsTimeSpan\": True }]",
      "ExternalEquipmentId": "[ExternalEquipmentId, { \"Value\": \"1234\", \"Raw\": \"1234\", \"IsObject\": False, \"IsArray\": True, \"IsBoolean\": False, \"IsString\": True, \"IsBytes\": True, \"IsDecimal\": True, \"IsDouble\": True, \"IsFloat\": True, \"IsInt64\": True, \"IsUInt64\": True, \"IsInteger\": True, \"IsGuid\": False, \"IsDateTime\": False, \"IsTimeSpan\": True }]"
    }
`

marcschier, 2023-10-20:
See also #2091

closed 2023-10-22 by marcschier

hvetter-de, 2023-11-03:
@marcschier Can you please provide a sample how the new ExtensionFields format will look like in version 2.9.3?