Azure/Industrial-IoT #2093: NodeRead_v2 returning HTTP 410 "A task was canceled"

emulic, 2023-10-27:
**Describe the bug**
direct method NodeRead_v2 of fully configured publisher module v 2.9.2.1 is returning following response instead of the node value:
 `{
    "status": 410,
    "payload": {
        "Message": "A task was canceled.",
        "Details": {
            "ClassName": "System.Threading.Tasks.TaskCanceledException",
            "Message": "A task was canceled.",
            "Data": null,
            "InnerException": null,
            "HelpURL": null,
            "StackTraceString": null,
            "RemoteStackTraceString": null,
            "RemoteStackIndex": 0,
            "ExceptionMethod": null,
            "HResult": -2146233029,
            "Source": null,
            "WatsonBuckets": null
        }
    }
}`

**To Reproduce**
Steps to reproduce the behavior:
1. Publisher module is properly configured to establish a connection to a OPC server. This has been validated by receiving values changes for some node (in my case ns=3;s=SimpleOPC.ScanResultDiscription1).
2. Send following payload (or similar) for the direct method NodeRead_v2 
`{
    "connection": {
        "endpoint": {
            "url": "opc.tcp:///192.168.222.147:50000",
            "securityMode": "None"
        }
    },
    "request": {
        "attributes": [
            {
                "nodeId": "ns=3;s=SimpleOPC.ScanResultDiscription1",
                "attribute": "Value"
            }
        ]
    }
}`
3.  Returned response is 410 with the message "A task was canceled"
4. publisher module logs do not show any errors, just warnings about disconnected OPCUA session:
`[12:43:22.3048] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from Disconnected to Connecting
[12:43:22.3048] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Connecting Client opc.tcp:///192.168.222.147:50000_21725920 [state:Connecting|refs:1] to opc.tcp:///192.168.222.147:50000...
[12:43:22.3055] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from Connecting to NotReachable
[12:43:22.3055] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #1: Failed to connect opc.tcp:///192.168.222.147:50000_21725920 [state:NotReachable|refs:1] to opc.tcp:///192.168.222.147:50000: Error establishing a connection: BadNotConnected...
[12:43:23.3052] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from NotReachable to Connecting
[12:43:23.3053] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Connecting Client opc.tcp:///192.168.222.147:50000_21725920 [state:Connecting|refs:1] to opc.tcp:///192.168.222.147:50000...
[12:43:23.3061] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from Connecting to NotReachable
[12:43:23.3062] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #1: Failed to connect opc.tcp:///192.168.222.147:50000_21725920 [state:NotReachable|refs:1] to opc.tcp:///192.168.222.147:50000: Error establishing a connection: BadNotConnected...
[12:43:24.3200] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from NotReachable to Connecting
[12:43:24.3201] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Connecting Client opc.tcp:///192.168.222.147:50000_21725920 [state:Connecting|refs:1] to opc.tcp:///192.168.222.147:50000...
[12:43:24.3211] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from Connecting to NotReachable
[12:43:24.3212] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #1: Failed to connect opc.tcp:///192.168.222.147:50000_21725920 [state:NotReachable|refs:1] to opc.tcp:///192.168.222.147:50000: Error establishing a connection: BadNotConnected...
[12:43:25.4162] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from NotReachable to Connecting
[12:43:25.4162] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Connecting Client opc.tcp:///192.168.222.147:50000_21725920 [state:Connecting|refs:1] to opc.tcp:///192.168.222.147:50000...
[12:43:25.4174] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from Connecting to NotReachable
[12:43:25.4175] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #1: Failed to connect opc.tcp:///192.168.222.147:50000_21725920 [state:NotReachable|refs:1] to opc.tcp:///192.168.222.147:50000: Error establishing a connection: BadNotConnected...
[12:43:26.4606] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from NotReachable to Connecting
[12:43:26.4606] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Connecting Client opc.tcp:///192.168.222.147:50000_21725920 [state:Connecting|refs:1] to opc.tcp:///192.168.222.147:50000...
[12:43:26.4619] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from Connecting to NotReachable
[12:43:26.4619] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #1: Failed to connect opc.tcp:///192.168.222.147:50000_21725920 [state:NotReachable|refs:1] to opc.tcp:///192.168.222.147:50000: Error establishing a connection: BadNotConnected...
[12:43:27.3040] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session disconnected during service call with message A task was canceled., retrying.
[12:43:27.3042] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Session opc.tcp:///192.168.222.147:50000_21725920 with opc.tcp:///192.168.222.147:50000 changed from NotReachable to Disconnected`

**Expected behavior**
Current value of the node "ns=3;s=SimpleOPC.ScanResultDiscription1" should be returned.

**Additional context**
I observed the same issue with NodeWrite_V1.
Maybe my request is wrong. However, I built it by following [publisher API documentation](https://github.com/Azure/Industrial-IoT/blob/release/2.9.2/docs/opc-publisher/api.md#noderead).
We used this direct method in the v2.8.7 of the twin module and it worked properly. However, the request we used was slightly different from the request we are using now  for the same direct method in the v2.9 publisher module:
``{
        "endpoint": {
            "url": "opc.tcp:///192.168.222.147:50000",
            "securityMode": "None"
        },
       "attributes": [
            {
                "nodeId": "ns=3;s=SimpleOPC.ScanResultDiscription1",
                "attribute": "Value"
            }
        ]
}``  
As you can see, there were no "connection" and "request" properties. Direct method of v2.9 publisher is throwing HTTP400 with the message "Value cannot be null. (Parameter 'request.Connection')" when sending request without these properties.

marcschier, 2023-10-27:
The call can timeout and this might have happened due to the unreachable server at the time of the call.  

That the endpoint is not reachable could be due to the three slashes in the url maybe.  Can you try using two, I e. 'opc.tcp://...'?

The old OPC twin rest API should still work through the companion web app.  But when using direct method directly the connection (endpoint) that is targeted is required in the request.

emulic, 2023-10-28:
Still facing the same issue, even with only 2 slashes.
In the meantime we setup a new EdgeDevice for testing this issue, since the device we used before is required to be used for somethign else.

Here is my configuration for both modules publisher and opcplc running on our new Test EdgeDevice for this issue:
```json
"modules": {
                    "opcplc": {
                        "restartPolicy": "always",
                        "settings": {
                            "image": "mcr.microsoft.com/iotedge/opc-plc:2.9.3",
                            "createOptions": "{\"Hostname\":\"opcplc\",\"Cmd\":[\"--ph=opcpl\",\"--autoaccept\",\"--pn=50000\",\"--sph=true\",\"--sn=5\",\"--sr=300\",\"--st=uint\",\"--fn=5\",\"--vfr=100\",\"--ft=uint\",\"--gn=5\"],\"HostConfig\":{\"Binds\":[\"/mnt/opcplc/certs:/app/pki\"],\"PortBindings\":{\"50000/tcp\":[{\"HostPort\":\"50000\"}],\"8080/tcp\":[{\"HostPort\":\"8080\"}]}},\"ExposedPorts\":{\"50000/tcp\":{},\"8080/tcp\":{}}}"
                        },
                        "status": "running",
                        "type": "docker"
                    },
                    "opc-publisher": {
                        "restartPolicy": "always",
                        "settings": {
                            "image": "mcr.microsoft.com/iotedge/opc-publisher:2.9",
                            "createOptions": "{\"Cmd\":[\"--PkiRootPath=/mount/pki\",\"--PublishedNodesFile=/appdata/pn.json\",\"--CreatePublishFileIfNotExistKey=false\",\"--AutoAcceptUntrustedCertificates=true\",\"--AddAppCertToTrustedStore=true\",\"--LogLevel=Debug\",\"--strict\",\"--fm=true\",\"--mm=Samples\",\"--me=Json\",\"--bs=2\",\"--DataSetKeyFrameCount=1\"],\"HostConfig\":{\"Binds\":[\"/mnt/opcpublisher/pki:/mount/pki\",\"/mnt/opcpublisher/appdata:/appdata\"],\"CapDrop\":[\"CHOWN\",\"SETUID\"]}}"
                        },
                        "status": "running",
                        "type": "docker"
                    }
                }
```

Here is content of the pn.json for publisher module, all 6 nodes are successfully subscribed and we are getting the value changes all the time in IoT Hub - which means, the connection to OPCPLC is open and working:
```json
[
  {
    "Version": 3,
    "LastChangeTimespan": "2023-10-28T06:35:39.9794501Z",
    "DataSetWriterId": "DataFlow_1",
    "DataSetWriterGroup": "Asset_1",
    "OpcNodes": [
      {
        "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt0"
      },
      {
        "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt1"
      },
      {
        "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt2"
      },
      {
        "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt3"
      },
      {
        "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt4"
      },
      {
        "Id": "nsu=http://microsoft.com/Opc/OpcPlc/;s=FastUInt5"
      }
    ],
    "EndpointUrl": "opc.tcp://opcplc:50000",
    "UseSecurity": true,
    "OpcAuthenticationMode": "Anonymous"
  }
]
```

NodeRead_v2 request:
```json
{
    "connection": {
        "endpoint": {
            "url": "opc.tcp://opcplc:50000",
            "securityMode": "None"
        }
[... 614 lines cut ...]
      #12: Sending 32:38 acks and deferring no acks. (0)
[06:55:10.8502] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #12: Sending 32:39 acks and deferring no acks. (0)
[06:55:10.8518] dbug: Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageEncoder[0]
      2 Notifications encoded into a network message (chunks:1)...
[06:55:10.8562] dbug: Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink[0]
      #1: Network message sent.
[06:55:10.8563] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Advancing stream #32 to #38
[06:55:10.8563] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Advancing stream #32 to #39
[06:55:11.8519] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #12: Sending 32:40 acks and deferring no acks. (0)
[06:55:12.8540] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      #12: Sending 32:41 acks and deferring no acks. (0)
[06:55:12.8545] dbug: Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageEncoder[0]
      2 Notifications encoded into a network message (chunks:1)...
[06:55:12.8598] dbug: Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink[0]
      #1: Network message sent.
[06:55:12.8599] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Advancing stream #32 to #40
[06:55:12.8599] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Advancing stream #32 to #41
```
Just for your understanding, in the past we haven't used twin rest API calls. We used directly direct methods of the twin module from other modules to invoke i.e. NodeRead_v2.

marcschier, 2023-10-28:
The OPC PLC does not have any endpoint with security mode None.  But it can be turned on using a CLI option.  Or you use Sign or SignAndEncrypt.

emulic, 2023-10-29:
This worked, thanks a lot!

closed 2023-10-29 by emulic