Azure/Industrial-IoT #2096: How to invoke a publisher direct method from another module with chunking (Furly)

emulic, 2023-10-30:
This is not a bug, it is just a kindly request to help us to invoke one of the publisher module direct methods with chunking, to overcome the direct methods payload size limitations by using the same Furly libraries you used in the publisher module and other components.
Unfortunately there seems to be no public documentation about the used Furly packages.
I was able to figure out, how to call the publisher module direct methods from cloud (analogous to how this can be done without chunking when using Microsoft.Azure.Devices.ServiceClient library).
If you are interested on that, I could contribute with the sample code for that, just let me know, what is the best way to provide my code.

However I'm struggling to figure out how to invoke a publisher direct method from another module, running either on the same EdgeDevice like publisher or on some child EdgeDevice. We used to use Microsoft.Azure.Devices.Client.ModuleClient method `_moduleClient.InvokeMethodAsync(deviceId, moduleId, methodRequest);` for doing that as chunking was not required.
Since I was not able to find any code in the Industrial-IoT solution doing the same with the Furly libraries,  maybe you can give me some hint which of the Furly libraries can be used and how to achieve that from another module?

I would also really appreciate if you could point to some class/method in the Industrial-IoT solution where I could figure out, how the implementation of a publisher direct method (i.e. GetConfiguredNodesOnEndpoint) looks like and how the Furly is used to create and return the big response payload in chunks. I expected to find somewhere code like `_moduleClient.SetMethodHandlerAsync(methodName, methodHandler, userContext, cancellationToken);`, used to register a direct method handler by using Microsoft.Azure.Devices.Client.ModuleClient library, but there is no code like that. 
I suppose this is somehow encapsulated in one of the Furly libraries, but I have no clue what and where to look for.

marcschier, 2023-10-30:
I recommend using the http rest API from another module to work around size limits.

emulic, 2023-10-31:
Hi @marcschier. Thanks for supporting us.
What is about security of the http rest API when being invoked from other Edge Devices? Since the endpoint in this case wouldn't be isolated in docker engine network only anymore, we would need to secure it by some kind of authentication and with TLS.
This was actually the main reason for us to keep using Direct Methods, since all these security aspects are already solved there. 
Do you have any recommendations about securing the publisher http rest API? 
Please note, our questions are not related to the publisher module only. In our solution we also have our own custom module providing i.e. some plant configuration data via Direct Methods to other Edge Devices in the plant. We are also looking for some solution to overcome the Direct Methods size limitations for already existing Direct Methods of our own modules. We need to consider security aspects, . The idea was to solve it in the similar way, like you solved it in Industrial IoT.

marcschier, 2023-10-31:
There is an API key that must be provided per call.  It and the server certificate can be retrieved from the module twin.  But that will not be possible for module to module so I would propose to add a device method in 2.9.3 to retrieve both and the use those when interacting with the https endpoint.  A feature that I can add in November or early December if that is ok.  Better than chunking which is unreliable and bound to change as it comes through said dependency which I would like to remove eventually.

emulic, 2023-10-31:
This sounds like a reasonable solution to me. 
Just one more question related to the server certificate: is this going to be a kind of a self signed certificate issued by the publisher module itself?
And regarding the dependency you would like to remove eventually: does it mean, the chunking mechanism used when invoking a publisher direct methods with big payloads from the server (cloud) side is also going to be removed or changed?

marcschier, 2023-11-13:
The latest preview is available at mcr.microsoft.com/iotedge/opc-publisher:2.9.3-preview2.  There are 2 new API: GetApiKey and GetServerCertificate which can only be invoked over IoT Hub direct methods.  Samples and documentation will follow.

closed 2023-11-13 by marcschier