Azure/Industrial-IoT #2099: Issue with DataType Handling in OPC Publisher Resulting in BadMonitoredItemIdInvalid Error

nikofalke, 2023-11-06:
OPC Publisher fails to publish nodes with data type mismatches which UA Expert can access. The node `ns=0;i=2258` ("CurrentTime") returns a `BadMonitoredItemIdInvalid` error in the OPC Publisher despite the UA Expert being able to read the value.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure OPC Publisher to monitor node `ns=0;i=2258`.
2. Observe that no data is published and a `BadMonitoredItemIdInvalid` error is logged.
3. Compare with UA Expert, which can access and display the node's value.

**Expected behavior**
OPC Publisher should tolerate data type mismatches similar to UA Expert or provide a clear indication of how to handle such discrepancies.

**Additional context**
UA Expert is able to handle the node despite it reporting an invalid data type (`Expected OpcUaType_NodeId, Received OpcUaType_Null`). This issue affects downstream data processing in our systems and needs to be addressed to ensure proper data collection and management.

[image]
[image]

nikofalke, 2023-11-06:
we are using the following module: mcr.microsoft.com/iotedge/opc-publisher:2.8.2

marcschier, 2023-11-07:
Could you try 2.9.2, or is above a typo?

nikofalke, 2023-11-08:
It is not a typo. I will try the newer version. Is it more resilient to bad configurations? I guess the UaExpert somehow catches this. We have the problem to communicate why our solution is not working compared to other tools.

marcschier, 2023-11-10:
Hi, the 2.8.2 version is very old, and I would want to first try 2.9 as it has a very recent version of the OPC UA stack where this issue might have been fixed.  If not, then could you provide logs of the error that is logged and the surrounding information?

marcschier, 2023-11-13:
@nikofalke, would you mind testing this with 2.9.3-preview2 tag of OPC Publisher and if it also fails send a snippet of the logs when the error occurrs? If you cannot, then even the log of the issue in the 2.8 publisher might already help.

marcschier, 2023-11-14:
One more, can you leave off the "ns=0;" part of the node ID just for kicks?

nikofalke, 2023-11-14:
Thank you for the quick replay, I will check with the customer as soon as possible and provide more detailed logs. Is there a migration guide? Some arguments seem not to be supported in 2.9.2 anymore. 
[image]

marcschier, 2023-11-14:
There is no migration guide as we aimed to be compatible for the production use cases.  This is an unfortunate exception, where we replaced serilog with Extension logging based on popular demand and their log level names are different (but standard .net log levels).  For Verbose, use "Trace", for "Info" use "Information".

marcschier, 2023-11-27:
Closing for now, please open when you find the issue reproducing on 2.9.2 (ideally 2.9.3-preview4).  Please send logs and server used.

closed 2023-11-27 by marcschier