Azure/Industrial-IoT #2101: Publisher HTTPS API returning SSL_ERROR_SYSCALL  error

emulic, 2023-11-08:
**Describe the bug**
When calling publisher HTTP API via HTTPS like i.e.
`curl https://****:9705/swagger/v2/openapi.json`
or
`curl -X POST https://*****:9705/v2/configuration/diagnostics`
following error occurs:
`curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to *****:9705`

Calling the API methods via HTTP is working as expected. 

I followed descriptions [#calling-the-api-over-http](https://github.com/Azure/Industrial-IoT/blob/release/2.9.2/docs/opc-publisher/readme.md#calling-the-api-over-http) and [#built-in-http-api-server](https://github.com/Azure/Industrial-IoT/blob/release/2.9.2/docs/opc-publisher/transports.md#built-in-http-api-server) but used different host ports then described there.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure publisher to map and expose the ports 80 and 443 as following:
`,
        "PortBindings": {
            "80/tcp": [
                {
                    "HostPort": "9704"
                }
            ],
            "443/tcp": [
                {
                    "HostPort": "9705"
                }
            ]
        }
    },
    "ExposedPorts": {
        "443/tcp": {},
        "80/tcp": {}
    }`
2.  From the EdgeDevice call some of the API methods via HTTPS i.e. (replace **** with your host name):
`curl https://****:9705/swagger/v2/openapi.json`
3. Take note about error occurred
`curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to *****:9705`

**Expected behavior**
The called API method would return a response just like calling via HTTP is doing.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.4
 - Publisher v 2.9.2.1

**Additional context**
Add any other context about the problem here.

marcschier, 2023-11-18:
The issue happens only when running in Iot edge.  The server certificate produced by the workload API is wrong somehow (looks like the private key is missing).  I have added code to check that a private key exists in it, and if not fall back to generating a self signed certificate (like when running outside iot edge).  Could you test the preview build published under tag '2.9.3-preview3' and if you still see issues, send me the log of OPC Publisher starting?

emulic, 2023-11-20:
I'll need some time to get back to this topic and find time for testing but I'll do that.
Can we expect  in some future version to get this solved in the way as it was originally intended, without using self signed certificates and using existing certificate chain rolled out by the customer PKI infrastructure instead?

marcschier, 2023-11-28:
We have a feature (#2081) that I have opened that is tracking this for the application certificate. We could expand this to also provision the ssl certificate. I cannot talk to timing though.

emulic, 2023-11-28:
Actually this is not what I asked. The original idea to reuse already enrolled certificate of the host is actually perfect. 
Whatever certificate is there, enrolled by using EST, SCEP, manually or however, can simply be used. No need to depend on some specific certificate enrollment solution or product. I mean, a certificate must be there to allow EdgeDevice to IoT Hub communication, device to device communication in a nested hierarchy or communication to OPC UA endpoints. Why to not reuse that certificate by the REST API of the publisher? This issue with the missing private key shouldn't be the reason to drift away from that great idea. Finally, edgeHub is using the same certificate in the similar way if I properly understand the documentation. If it works in edgeHub, why shouldn't it work in the publisher module?

marcschier, 2023-12-10:
Issue lies in an incompatibility with .net.

`Not supported: The 'unix' scheme is not supported.. Unable to use workload API to obtain`

marcschier, 2023-12-27:
Fix is in main and will be in 2.9.4-preview1

closed 2023-12-27 by marcschier