Azure/Industrial-IoT #2102: Dapr Config Document

xky0007, 2023-11-13:
**Is your feature request related to a problem? Please describe.**

There is no document nor example for how to use dapr with OPC-Publisher. 

For example, if I have a customized daprd binary, how do I set the daprd path and config path with publisher? 

**Describe the solution you'd like**

Add examples and docs for dapr support.

marcschier, 2023-11-14:
Hi,  docs are at https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/transports.md#dapr.    

Please start with the compose sample configuration: https://github.com/Azure/Industrial-IoT/blob/main/deploy/docker/with-dapr.yaml.

marcschier, 2023-11-27:
Cannot reproduce the error mentioned using main branch.  To use the main branch you must first run the build.cmd and then -f with-localimage.yaml.  Until 2.9.3 has been released to the 2.9 tag.

closed 2023-11-27 by marcschier