Azure/Industrial-IoT #2104: Reconnect issue in OPC UA stack with BadSessionIdINvalid:

marcschier, 2023-11-13:
Customer reported reconnect issue in OPC UA stack with BadSessionIdInvalid:

[08:02:06.1576] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616002, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:07.1456] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616003, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:07.1658] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616004, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:07.6348] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616005, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:08.1628] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616006, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:08.6346] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616007, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:09.1523] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616008, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:09.1587] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616009, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:09.1713] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616010, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.1620] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616011, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.3439] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616012, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.4200] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616013, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.4242] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616014, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.5396] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616015, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.5639] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616016, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.5917] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616017, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.6183] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616018, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.6243] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616019, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.6781] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616020, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.7094] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616021, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.7178] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616022, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.7192] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616023, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.7717] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616024, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.8064] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616025, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.8390] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616026, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.8619] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616027, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.8908] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616028, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:10.9304] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616029, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:11.1539] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616030, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:11.6616] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616031, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:11.9771] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Unexpected keep alive error occurred: BadSessionIdInvalid
[08:02:12.1521] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616032, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:12.1607] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616033, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:15.1671] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616036, Reconnecting=False, Error: BadSessionIdInvalid
[... 111 lines cut ...]
      Publish #6616090, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:21.9756] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Unexpected keep alive error occurred: BadSessionIdInvalid
[08:02:22.1520] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616091, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:22.1590] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616092, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:25.1725] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616111, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:25.1728] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616112, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:25.1729] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616110, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:25.1730] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616109, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:25.1731] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616108, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:25.1732] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616107, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:25.1733] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616106, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:25.1734] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616105, Reconnecting=False, Error: BadSessionIdInvalid
[08:02:25.1735] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #6616104, Reconnecting=False, Error: BadSessionIdInvalid

nelsonmorais, 2023-11-15:
Hi,

Not sure if this can be related, but in one of our cases we're seeing a similar message:

```
[18:28:43.6074] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #162, Reconnecting=False, Error: BadTimeout
[18:28:43.6840] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      Publish #162, Reconnecting=False, Error: BadTimeout
[18:28:43.7077] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      PUBLISH #162 - Unhandled error BadTimeout during Publish.
      Opc.Ua.ServiceResultException: BadTimeout
         at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
         at Opc.Ua.SessionClient.EndPublish(IAsyncResult result, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
         at Opc.Ua.Client.Session.OnPublishComplete(IAsyncResult result)
[18:28:43.7842] fail: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaStack[0]
      PUBLISH #162 - Unhandled error BadTimeout during Publish.
      Opc.Ua.ServiceResultException: BadTimeout
         at Opc.Ua.ClientBase.ValidateResponse(ResponseHeader header)
         at Opc.Ua.SessionClient.EndPublish(IAsyncResult result, UInt32& subscriptionId, UInt32Collection& availableSequenceNumbers, Boolean& moreNotifications, NotificationMessage& notificationMessage, StatusCodeCollection& results, DiagnosticInfoCollection& diagnosticInfos)
         at Opc.Ua.Client.Session.OnPublishComplete(IAsyncResult result)
```

This occurs without a specific cadence, but with quite some frequency, sometimes within 2 minutes, sometimes within 3-4 minutes.

Apparently we're receiving all the data, but we continuously see messages like the above on the logs in between the statistics messages. 

We have just 2 nodes subscribed on 2 servers, so a very simple configuration:

```Json
[
  {
    "DataSetWriterGroup": "1231",
    "DataSetWriterId": "DSWID:1231",
    "EndpointUrl": "opc.tcp://xxx.xxx.xxx.xxx:nnnn",
    "DataSetPublishingInterval": 10000,
    "OpcNodes": [
      {
        "Id": "ns=7;s=<identifier 1 here>",
        "OpcSamplingInterval": 400
      },
      {
        "Id": "ns=7;s=<identifier 2 here>",
        "OpcSamplingInterval": 400
      }
    ]
  },
  {
    "DataSetWriterGroup": "1232",
    "DataSetWriterId": "DSWID:1232",
    "EndpointUrl": "opc.tcp://yyy.yyy.yyy.yyy:nnnn",
    "DataSetPublishingInterval": 10000,
    "OpcNodes": [
      {
        "Id": "ns=7;s=<identifier 1 here>",
        "OpcSamplingInterval": 400
      },
      {
        "Id": "ns=7;s=<identifier 2 here>",
        "OpcSamplingInterval": 400
      }
    ]
  }
]
```

marcschier, 2023-11-27:
https://github.com/OPCFoundation/UA-.NETStandard/issues/2394#issue-2009440639

closed 2023-12-27 by marcschier