Azure/Industrial-IoT #2116: endpointState / notSeenSince missing (EndpointInfoModel)

krkappa, 2023-11-19:
**Describe the bug**
I have deployed the current version (2.9.2) with option "-type all". When testing the Web API (with Postman), I never get the information about endpointState / notSeenSince by calling 

[image]

At the time of the call, I am already connected to the OPC server and, for example, can browse the root node.

**To Reproduce**
1. Deploy current version 2.9.2
2. Configure Postman to retrieve access token + get a new access token
3. Add application
[image]
4. Get available endpoints
[image]
5. Connect endpoint
[image]
6. Browse target node, browse nodes or read a value to test that the connection to the OPC server is established
7. Get detail information of the selected endpoint
[image]
--> No "endpointState" and "notSeenSince" in the response

**Expected behavior**
If the connection to the OPC server is successfully established, the respone of 
[image]
includes the current information for "endpointState" and "notSeenSince".

marcschier, 2023-11-20:
Endpoint state API remained for backwards compatibility, but not supported anymore in 2.9.  Is there a particular scenario you need it for, or is this just for testing?

krkappa, 2023-11-20:
Thanks for the fast reply. Did I understand the point correctly that the EndpointInfo model remains unchanged, but the information (endpointState, notSeenSince) is no longer output or supported? Which query can I use to get the status of the endpoint (ready, disconnected, ...)? Or can this information be obtained from the execution of other queries (browse node)?

krkappa, 2023-11-21:
Thanks for the replies. We currently only need the feature for testing purposes. It would be helpful to adapt the API documentation to determine which functions are supported in version 2.9 and which are still available due to backwards compatibility. The issue ticket can be closed. Thanks.

closed 2024-07-05 by marcschier