Azure/Industrial-IoT #2117: Replace Furly dapr related library

xky0007, 2023-11-20:
**Is your feature request related to a problem? Please describe.**
After trying to run the publisher in local debug mode, I found the Furly related library is not open source. I cannot check the source code for dapr related code. 

I found following problems:

1. In the configure dapr options part. It only uses grpc protocol even though I only set http port in the env. It still uses grpc protocol and generates following error.

```
#24: Error 'Publish operation failed: the Dapr endpoint indicated a failure. See InnerException for details.' during sending network message. Retrying in 00:00:02.4000000...
      Dapr.DaprException: Publish operation failed: the Dapr endpoint indicated a failure. See InnerException for details.
       ---> Grpc.Core.RpcException: Status(StatusCode="Unavailable", Detail="Error connecting to subchannel.", DebugException="System.Net.Sockets.SocketException: No connection could be made because the target machine actively refused it.")
       ---> System.Net.Sockets.SocketException (10061): No connection could be made because the target machine actively refused it.
         at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError error, CancellationToken cancellationToken)
         at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
         at System.Net.Sockets.Socket.<ConnectAsync>g__WaitForConnectWithCancellation|285_0(AwaitableSocketAsyncEventArgs saea, ValueTask connectTask, CancellationToken cancellationToken)
         at Grpc.Net.Client.Balancer.Internal.SocketConnectivitySubchannelTransport.TryConnectAsync(ConnectContext context)
         --- End of inner exception stack trace ---
         at Grpc.Net.Client.Balancer.Internal.ConnectionManager.PickAsync(PickContext context, Boolean waitForReady, CancellationToken cancellationToken)
         at Grpc.Net.Client.Balancer.Internal.BalancerHttpHandler.SendAsync(HttpRequestMessage request, CancellationToken cancellationToken)
         at System.Net.Http.HttpMessageInvoker.<SendAsync>g__SendAsyncWithTelemetry|6_0(HttpMessageHandler handler, HttpRequestMessage request, CancellationToken cancellationToken)
         at Grpc.Net.Client.Internal.GrpcCall`2.RunCall(HttpRequestMessage request, Nullable`1 timeout)
         at Dapr.Client.DaprClientGrpc.MakePublishRequest(String pubsubName, String topicName, ByteString content, Dictionary`2 metadata, String dataContentType, CancellationToken cancellationToken)
         --- End of inner exception stack trace ---
         at Dapr.Client.DaprClientGrpc.MakePublishRequest(String pubsubName, String topicName, ByteString content, Dictionary`2 metadata, String dataContentType, CancellationToken cancellationToken)
         at Furly.Extensions.Dapr.Clients.DaprPubSubClient.DaprPubSubEvent.SendAsync(CancellationToken ct)
         at Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink.SendAsync(ValueTuple`2 message) 
```

2. In the class `internal sealed class Dapr : ConfigureOptionBase<DaprOptions>`, it sets grpc and http ports to https which should not support dapr endpoint - at least from the dapr official website, I only see http endpoint. After change from https to http, enable grpc port for my dapr sidecar, the publisher works now.

**Describe the solution you'd like**
We should found open source dapr related libs to implement dapr support. For example, the dapr sdk from Dapr or [DaprSideKick](https://github.com/man-group/dapr-sidekick-dotnet)

Just curious why choose [furlysoft](https://github.com/furlysoft)?

HeikoS-Trumpf, 2023-11-20:
Or open source the furly library. The nuget-packages have MIT license.

marcschier, 2023-11-20:
Yep, the microsoft/project-furly repo will be public with the release of 2.9.3 next week.

closed 2023-11-20 by marcschier