Azure/Industrial-IoT #2118: HTTP to Dapr does not work

xky0007, 2023-11-20:
**Describe the bug**

Even the connection string only set HttpPort, it still tries to communicate via GRPC protocol.

In environment variable, set `DaprConnectionString="PubSubComponent=redis-pubsub-cloud;HttpPort=7001;"`,

Get error: `Grcp.Core.RpcException`

```
#24: Error 'Publish operation failed: the Dapr endpoint indicated a failure. See InnerException for details.' during sending network message. Retrying in 00:00:02.4000000...
      Dapr.DaprException: Publish operation failed: the Dapr endpoint indicated a failure. See InnerException for details.
       ---> Grpc.Core.RpcException: Status(StatusCode="Unavailable", Detail="Error connecting to subchannel.", DebugException="System.Net.Sockets.SocketException: No connection could be made because the target machine actively refused it.")
       ---> System.Net.Sockets.SocketException (10061): No connection could be made because the target machine actively refused it.
         at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.ThrowException(SocketError error, CancellationToken cancellationToken)
         at System.Net.Sockets.Socket.AwaitableSocketAsyncEventArgs.System.Threading.Tasks.Sources.IValueTaskSource.GetResult(Int16 token)
         at System.Net.Sockets.Socket.<ConnectAsync>g__WaitForConnectWithCancellation|285_0(AwaitableSocketAsyncEventArgs saea, ValueTask connectTask, CancellationToken cancellationToken)
         at Grpc.Net.Client.Balancer.Internal.SocketConnectivitySubchannelTransport.TryConnectAsync(ConnectContext context)
         --- End of inner exception stack trace ---
         at Grpc.Net.Client.Balancer.Internal.ConnectionManager.PickAsync(PickContext context, Boolean waitForReady, CancellationToken cancellationToken)
         at Grpc.Net.Client.Balancer.Internal.BalancerHttpHandler.SendAsync(HttpRequestMessage request, CancellationToken cancellationToken)
         at System.Net.Http.HttpMessageInvoker.<SendAsync>g__SendAsyncWithTelemetry|6_0(HttpMessageHandler handler, HttpRequestMessage request, CancellationToken cancellationToken)
         at Grpc.Net.Client.Internal.GrpcCall`2.RunCall(HttpRequestMessage request, Nullable`1 timeout)
         at Dapr.Client.DaprClientGrpc.MakePublishRequest(String pubsubName, String topicName, ByteString content, Dictionary`2 metadata, String dataContentType, CancellationToken cancellationToken)
         --- End of inner exception stack trace ---
         at Dapr.Client.DaprClientGrpc.MakePublishRequest(String pubsubName, String topicName, ByteString content, Dictionary`2 metadata, String dataContentType, CancellationToken cancellationToken)
         at Furly.Extensions.Dapr.Clients.DaprPubSubClient.DaprPubSubEvent.SendAsync(CancellationToken ct)
         at Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink.SendAsync(ValueTuple`2 message) in C:\TRUMPF\Industrial-IoT\src\Azure.IIoT.OpcUa.Publisher\src\Services\NetworkMessageSink.cs:line 235
[02:31:49.4082] dbug: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]

```

**Expected behavior**
If HttpPort is set and no GrpcPort, dapr should use http protocol.

marcschier, 2023-11-20:
Looks like the dapr c# sdk does only allow building a grpc dapr client. Why would you want to use http instead of Grpc, is this a requirement?

xky0007, 2023-11-21:
> Looks like the dapr c# sdk does only allow building a grpc dapr client. Why would you want to use http instead of Grpc, is this a requirement?

Which dapr c# sdk is using now? Because dapr sidecar can run under http protocol only, grpc protocol only and http + grpc protocols mode, if only http port is set, it should use http protocol insteand of grpc protocol. 

What do you think?

marcschier, 2023-11-21:
I think so too, but the official dapr SDK (https://github.com/dapr/dotnet-sdk) only ever allows creating the latter (http+grpc) using default urls if not overridden.  

If you are ok running both grpc and http I would close this as soon as the other issue is fixed. If you have a need for not using grpc, let me know and I will see what it takes to do it.

closed 2023-11-21 by marcschier