Azure/Industrial-IoT #2141: Docs - Prometheus metrics named as IotHub, but I assume also track MQTT publish at this point

pesetskyps, 2023-12-18:
**Is your feature request related to a problem? Please describe.**
On page [obesevability](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/observability.md#available-metrics) there are descriptions like `Messages sent to IoTHub per second	` as others that have IotHub inside, but output of publisher is MQTT now also, so 
1. Is my assumption is correct that the metrics track MQTT publish messages?
2. Should we rename the description to `Messages sent to **outputs** per second`

marcschier, 2023-12-19:
You are correct.  I think changing the name of the output text is reasonable.

marcschier, 2023-12-27:
Changed names used when logging the diagnostics to the console in 2.9.4 but will keep the metric names as is.  New output:

```
  DIAGNOSTICS INFORMATION for          : MES
  # OPC Publisher Version (Runtime)    : 2.9.4.1-preview+6a56240de5 [DEBUG] (.NET 8.0.0/win-x64)
  # Time                               : 2023-12-27T15:11:36.6854888Z
  # Ingestion duration                 :    00:00:00:56 (dd:hh:mm:ss)
  # Opc endpoint connected?            :           True
  # Connection retries                 :              0
  # Monitored Opc nodes succeeded count:              6
  # Monitored Opc nodes failed count   :              1
  # Subscriptions count                :              1
  # Ingress value changes              :            298 (All time ~5.27/s; 57 in last 60s ~5.27/s)
  # Ingress events                     :              0
  # Received Data Change Notifications :             57 (All time ~1.01/s; 57 in last 60s ~1.01/s)
  # Received Event List notifications  :              0
  # Received Keep Alive notifications  :              0
  # Generated Cyclic read notifications:              0
  # Generated Heartbeat notifications  :              0
  # Notification batch buffer size     :              0
  # Encoder input/output buffer size   :              0 | 0
  # Encoder Notifications processed    :             57
  # Encoder Notifications dropped      :              0
  # Encoder Network Messages produced  :             58
  # Encoder avg Notifications/Message  :              1
  # Encoder worst Message split ratio  :              0
  # Encoder avg Message body size      :          1,131
  # Encoder avg Chunk (4 KB) usage     :            0.3
  # Estimated Chunks (4 KB) per day    :              0
  # Egress messages ready to send      :              0
  # Egress messages dropped            :              0
  # Egress messages successfully sent  :             58 (0.99/s)

```

closed 2023-12-27 by marcschier