Azure/Industrial-IoT #2145: aarch64/ARM64 error

TruthSeekerX, 2023-12-20:
**Describe the bug**
The latest update broke the OpcPublisher on rPi400. Tried both ubuntu 22.04 server LTS and Raspbian OS (bookworm) the same error: "_exec /usr/bin/dotnet: exec format error_"

**To Reproduce**
sudo iotedge list

**Expected behavior**
It worked before.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04 Server LTS 64bit, Raspbian OS(bookworm) 64bit

**Hardware""
RaspberryPi 400 (ARM Cortex A72)

**Additional context**
As a side project. I was working on opcPublisher only 1day/week. It worked 10 days ago when setup. But from yesterday 19.Dec.2023, OpcPublisher stoped working.

marcschier, 2023-12-21:
Build system is selecting the wrong Mariner image as base, change to use 8.0.0-cbl-mariner2.0-arm64v8 explicitly.

TruthSeekerX, 2023-12-21:
Thanks for quick reply. 👍

marcschier, 2024-01-02:
Should be fixed in mcr.microsoft.com/iotedge/opc-publisher:2.9.4-preview1.

closed 2024-01-02 by marcschier