Azure/Industrial-IoT #2146: Unable to Connect to OPC Publisher 2.9.2  Rest End Points (Connection refused)

vishutyagi000, 2023-12-26:
**Describe the bug**
Unable to Connect to OPC Publisher Rest End Points 
Used below commands 
curl http://localhost:8080/swagger/v2/openapi.json
curl http://{opcpublisherModuleName}:8080/swagger/v2/openapi.json in my case curl http://OPCPublisher:8080/swagger/v2/openapi.json

**To Reproduce**

1. Add OPC publisher module 2.9.3  using image  mcr.microsoft.com/iotedge/opc-publisher:2.9.3 in edge deployment.
2. Go to the VM(ubuntu 20.0)  Used as edge device using ssh
3. Use curl http://localhost:8080/swagger/v2/openapi.json
4. Got error Connection refused.

**Expected behavior**
The End point should get called  as 8080 is exposed .
Refer -https://github.com/Azure/Industrial-IoT/blob/246ff6e4b4105baa562c2c01870895432d21ea73/docs/opc-publisher/readme.md#calling-the-api-over-http

**Screenshots**

[image]
[image]

If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - Ubuntu (20.0) as edge device
 - Browser [e.g. chrome, safari]
 -OPC Publisher version 2.9.3

**Additional context**
Add any other context about the problem here.

marcschier, 2023-12-27:
The 443/80 ports must be routed to 8080/8081. It looks like this is not the case in your setup.  If you want to use 8080: `{\"PortBindings\":{\"80/tcp\":[{\"HostPort\":\"8080\"}],\"443/tcp\":[{\"HostPort\":\"8081\"}]}`. see [here](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/deployment.json).

If this did not work, please provide the full "CreateOptions" you used in your deployment manifest.

vishutyagi000, 2023-12-27:
@marcschier  Thanks its working ,Yeah  PortBindings were wrong  ,instead of \"80/tcp\":[{\"HostPort\":\"8080\"}  i was using 

\"8080/tcp\":[{\"HostPort\":\"8080\"} and In OPCPublisher log i can see  8080 was being overridden.

[image]

closed 2023-12-27 by vishutyagi000