Azure/Industrial-IoT #2149: Not all metrics available

alxy, 2024-01-04:
I'm trying to scrape the prometheus metrics exposes by the opcpublisher as [explained here](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/observability.md#available-metrics). I'm using the `metricscollector` module and looking at the data in log analytics. I scoped the collection to only scrape `iiot_edge_publisher_*` metrics, which are the ones that are interesting to us. However, I noticed that not all metrics that are documented are available.

For example, the following metrics are missing entirely:

- iiot_edge_publisher_good_nodes 
- iiot_edge_publisher_bad_nodes

We have set a rather low scraping frequency of `300 seconds` (5 minutes, default ist 30 seconds) to reduce the log analytics costs a bit. Not sure if that is an issue? Are all metrics exposed at each scraping action by the `metricscollector`? Is there a minimal sampling frequency we have to use to get all metrics?

pesetskyps, 2024-02-01:
@marcschier I see you put as the bug, did you check that it actually a bug so no telemetry tracked correctly for MQTT outputs?

marcschier, 2024-02-01:
No, have not tried this out yet.  I do believe this is not a bug, as the information that is missing is actually diagnostics instruments that are driving the --di output to the console log.  But I will confirm this in 2.9.5 time

pesetskyps, 2024-02-02:
@marcschier I'd really appreaciate that you can point us to the place where we can check what metrics we can use to track good, bad nodes and other metrics that are tracked when we output to MQTT (if it matters)?

filipzanpwccom, 2024-02-02:
Same for me as well.

Also what metrics should we rely on to check issues with MQTT broker (or any other) output?

marcschier, 2024-02-02:
> We have the same issue. The metric `iiot_edge_publisher_sent_iot_messages` that we used to monitor opc publisher v2.8 has been removed/renamed? in v2.9 In my opinion, renaming prometheous metrics should have been documented as breaking changes in release note.
> 
> Besides, when collecting metrics with metrics-collector IoT Edge module, there are some warnings: It might explain why "good_nodes" and "bad_nodes" metrics are missing, likely due to whitespace
> 
> ```
> [2024-02-02 10:43:04.303 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_chunk_size_average_4kb Chunks{publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",siteId="<<UnknownSite>>",writerGroupId="<<UnknownWriterGroup>>"} 0.4933375839359504 1706870584229]
> [2024-02-02 10:43:04.303 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_chunk_size_average_4kb Chunks{publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",siteId="<<UnknownSite>>",writerGroupId="test"} 0.5018673731275833 1706870584229]
> [2024-02-02 10:43:04.303 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_missing_keep_alives_Keep Alives_total{dataSetWriterId="<<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c)",endpointUrl="opc.tcp://opc-plc-sim:50000",publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",securityMode="SignAndEncrypt",siteId="<<UnknownSite>>",writerGroupId="<<UnknownWriterGroup>>"} 0 1706870584229]
> [2024-02-02 10:43:04.303 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_missing_keep_alives_Keep Alives_total{dataSetWriterId="<<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c)",endpointUrl="opc.tcp://opc-plc-sim:50000",publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",securityMode="SignAndEncrypt",siteId="<<UnknownSite>>",writerGroupId="test"} 0 1706870584229]
> [2024-02-02 10:43:04.303 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_good_nodes_Monitored items{dataSetWriterId="<<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c)",endpointUrl="opc.tcp://opc-plc-sim:50000",publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",securityMode="SignAndEncrypt",siteId="<<UnknownSite>>",writerGroupId="<<UnknownWriterGroup>>"} 1 1706870584229]
> [2024-02-02 10:43:04.303 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_good_nodes_Monitored items{dataSetWriterId="<<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c)",endpointUrl="opc.tcp://opc-plc-sim:50000",publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",securityMode="SignAndEncrypt",siteId="<<UnknownSite>>",writerGroupId="test"} 1 1706870584229]
> [2024-02-02 10:43:04.303 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_bad_nodes_Monitored items{dataSetWriterId="<<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c)",endpointUrl="opc.tcp://opc-plc-sim:50000",publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",securityMode="SignAndEncrypt",siteId="<<UnknownSite>>",writerGroupId="<<UnknownWriterGroup>>"} 0 1706870584229]
> [2024-02-02 10:43:04.304 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_bad_nodes_Monitored items{dataSetWriterId="<<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c)",endpointUrl="opc.tcp://opc-plc-sim:50000",publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",securityMode="SignAndEncrypt",siteId="<<UnknownSite>>",writerGroupId="test"} 0 1706870584229]
> [2024-02-02 10:43:04.304 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_monitored_items_Monitored items{dataSetWriterId="<<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c)",endpointUrl="opc.tcp://opc-plc-sim:50000",publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",securityMode="SignAndEncrypt",siteId="<<UnknownSite>>",writerGroupId="<<UnknownWriterGroup>>"} 1 1706870584229]
> [2024-02-02 10:43:04.304 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_monitored_items_Monitored items{dataSetWriterId="<<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c)",endpointUrl="opc.tcp://opc-plc-sim:50000",publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",securityMode="SignAndEncrypt",siteId="<<UnknownSite>>",writerGroupId="test"} 1 1706870584229]
> [2024-02-02 10:43:04.304 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_connection_retries_Connection attempts{dataSetWriterId="<<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c)",endpointUrl="opc.tcp://opc-plc-sim:50000",publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",securityMode="SignAndEncrypt",siteId="<<UnknownSite>>",writerGroupId="<<UnknownWriterGroup>>"} 1 1706870584229]
> [2024-02-02 10:43:04.304 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_connection_retries_Connection attempts{dataSetWriterId="<<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c) [... cut ...] ",publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher",securityMode="SignAndEncrypt",siteId="<<UnknownSite>>",writerGroupId="test"} 1 1706870584229]
> [2024-02-02 10:43:04.305 WRN] Ignoring metric line because it does not match the expected format: [iiot_edge_publisher_session_count_Number of active sessions.{publisherId="iot-gptmpxp01-eun-dev-core-01_device_device01_module_opc-publisher"} 2 1706870584229]
> ```

Agreed 100%, this is a bug that will be fixed in 2.9.4.

marcschier, 2024-02-04:
Otel SDK forms a metric name by adding the units from the .net metrics api, so I removed all units which should now result again in the 2.8 and doced instrument names on the scrape endpoint.  Plus no more spaces.

closed 2024-02-04 by marcschier

jacqueskang, 2024-02-06:
Hi @marcschier, can we start testing this fix in a preview release?
I tested 2.9.4-preview4 and it seems that the whitespaces are still there

marcschier, 2024-02-07:
Yes, correct, there is no preview release yet with this change, this goes into an imminent 2.9.4 release and any issues would then be fixed in 2.9.5 in early March.  However, if you can test this from main line by building an image on your machine (e.g., run deploy/docker/build.bat and use the local iotedge/opc-publisher:latest image that results from it) and confirm, then 2.9.4 will be golden.

jacqueskang, 2024-02-07:
@marcschier thanks for information. Unfortunately I cannot install build tools (e.g., Docker, .NET SDK, etc.) in my computer due to company IT restrictions. But I can test once official 2.9.4 docker image will be released and confirm the results

marcschier, 2024-02-14:
2.9.4 is out, please open a new issue of you find things not as expected, then I see it right away.