Azure/Industrial-IoT #2151: OPCPublisher - no data published

mtraina, 2024-01-07:
Dear team

I am testing the OPC UA simulation, following [this guide](https://azure.github.io/Industrial-IoT/opc-publisher/#how-opc-publisher-works), however, even if the connectiivy seems in place, once the simulation starts, I do not see any message exchanged between the OPC simulator and the OPCPublisher.

My setup is:
- [Prosys UA simulator](https://www.prosysopc.com/products/opc-ua-simulation-server/) running locally on Macos
- Iot Edge Device in a Ubuntu 22 (ARM64) VM, still running on my laptop, setup made following [this guide](https://learn.microsoft.com/en-us/azure/iot-edge/how-to-provision-single-device-linux-symmetric?view=iotedge-1.4&tabs=azure-portal%2Cubuntu)
- OPCPublisher v2.8, as v2.9 could not start once deployed

No errors in the logs.

I then tried different combinations, like:
- changed the VM to a Debian 11 ARM64 -> same result
- changed the simulator to the [OPC-PLC server](https://github.com/Azure-Samples/iot-edge-opc-plc) -> same result

From this last test, here the logs (changed the ip to a placeholder):
`[2024-01-06 20:56:45.349 INF OpcUa] TransportChannel RECONNECT: Reconnecting to opc.tcp://$server:50000/.
[2024-01-06 20:56:45.349 INF OpcUa] ChannelId 0: in Connecting state.
[2024-01-06 20:56:45.372 INF OpcUa] ChannelId 0: in Opening state.
[2024-01-06 20:56:45.373 INF OpcUa] ChannelId 0: Token #0 created. CreatedAt=20:56:45.373. Lifetime=3600000.
[2024-01-06 20:56:45.417 INF OpcUa] ChannelId 0: in Closed state.
[2024-01-06 20:56:45.417 INF OpcUa] ChannelId 0: CLIENTCHANNEL SOCKET CLOSED: 01549307
[2024-01-06 20:56:45.418 INF OpcUa] ChannelId 0: in Closed state.
[2024-01-06 20:56:45.420 INF OpcUa] ChannelId 0: CLIENTCHANNEL SOCKET CLOSED: FFFFFFFF
[2024-01-06 20:56:45.420 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Failed to reconnect session 'opc.tcp://$server:50000_83CF7F93' due to Error received from remote host: Could not verify security on OpenSecureChannel request.,  disposing and trying to create new
[2024-01-06 20:56:45.420 INF OpcUa] Subscription DELETE, Id=801808655, LastNotificationTime=20:56:15, GoodPublishRequestCount=0, PublishingInterval=1000, KeepAliveCount=10, PublishingEnabled=True, MonitoredItemCount=3
[2024-01-06 20:56:45.421 INF OpcUa] ChannelId 0: in Connecting state.
[2024-01-06 20:56:45.424 WRN OpcUa] ChannelId 0: Force reconnect reason=BadConnectionClosed 'Remote side closed connection.'
[2024-01-06 20:56:45.424 INF OpcUa] ChannelId 0: in Closed state.
[2024-01-06 20:56:45.424 INF OpcUa] ChannelId 0: CLIENTCHANNEL SOCKET CLOSED: 01E9E8B3
[2024-01-06 20:56:45.424 WRN OpcUa] ChannelId 0: Force reconnect reason=BadConnectionClosed 'Remote side closed connection'
[2024-01-06 20:56:45.425 INF OpcUa] ChannelId 0: in Connecting state.
[2024-01-06 20:56:45.430 INF OpcUa] ChannelId 0: in Opening state.
[2024-01-06 20:56:45.430 INF OpcUa] ChannelId 0: Token #0 created. CreatedAt=20:56:45.430. Lifetime=3600000.
[2024-01-06 20:56:45.442 INF OpcUa] Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
[2024-01-06 20:56:45.442 INF OpcUa] Sender Certificate: (none)
[2024-01-06 20:56:45.442 INF OpcUa] ChannelId 3: Token #1 activated. CreatedAt=20:56:45.430. Lifetime=3600000.
[2024-01-06 20:56:45.442 INF OpcUa] ChannelId 3: in Open state.
[2024-01-06 20:56:45.442 INF OpcUa] ChannelId 3: Token Expiry 01/06/2024 21:56:45, renewal scheduled in 2699990 ms.
[2024-01-06 20:56:45.460 INF OpcUa] ChannelId 3: in Closing state.
[2024-01-06 20:56:45.462 WRN OpcUa] ChannelId 3: Force reconnect reason=BadConnectionClosed 'Remote side closed connection'
[2024-01-06 20:56:45.462 INF OpcUa] ChannelId 3: in Closed state.
[2024-01-06 20:56:45.462 INF OpcUa] ChannelId 3: CLIENTCHANNEL SOCKET CLOSED: 0369120D
[2024-01-06 20:56:45.463 INF Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Creating session 'opc.tcp://$server:50000_83CF7F93' for endpoint 'opc.tcp://$server:50000'...
[2024-01-06 20:56:45.464 INF OpcUa] ChannelId 0: in Connecting state.
[2024-01-06 20:56:45.466 INF OpcUa] ChannelId 0: in Opening state.
[2024-01-06 20:56:45.466 INF OpcUa] ChannelId 0: Token #0 created. CreatedAt=20:56:45.466. Lifetime=3600000.
[2024-01-06 20:56:45.467 INF OpcUa] Security Policy: http://opcfoundation.org/UA/SecurityPolicy#None
[2024-01-06 20:56:45.468 INF OpcUa] Sender Certificate: (none)
[2024-01-06 20:56:45.468 INF OpcUa] ChannelId 4: Token #1 activated. CreatedAt=20:56:45.466. Lifetime=3600000.
[2024-01-06 20:56:45.468 INF OpcUa] ChannelId 4: in Open state.
[2024-01-06 20:56:45.468 INF OpcUa] ChannelId 4: Token Expiry 01/06/2024 21:56:45, renewal scheduled in 2699998 ms.
[2024-01-06 20:56:45.470 INF OpcUa] ChannelId 4: in Closing state.
[2024-01-06 20:56:45.473 WRN OpcUa] ChannelId 4: Force reconnect reason=BadConnectionClosed 'Remote side closed connection'
[2024-01-06 20:56:45.473 INF OpcUa] ChannelId 4: in Closed state.
[2024-01-06 20:56:45.473 INF OpcUa] ChannelId 4: CLIENTCHANNEL SOCKET CLOSED: 0396417E
[2024-01-06 20:56:45.483 WRN OpcUa] Certificate Validation failed. Reason=BadCertificateUntrusted. [CN=OpcPlc] [B8E6E8DD552504EA7DEE5BF28F2011A06C1450BE]
[2024-01-06 20:56:45.483 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Accepting untrusted peer certificate B8E6E8DD552504EA7DEE5BF28F2011A06C1450BE, 'CN=OpcPlc' due to AutoAccept(UntrustedCertificates) set!
[2024-01-06 20:56:45.483 WRN OpcUa] Validation errors suppressed:  [CN=OpcPlc] [B8E6E8DD552504EA7DEE5BF28F2011A06C1450BE]
[2024-01-06 20:56:45.483 INF OpcUa] ChannelId 0: in Connecting state.
[2024-01-06 20:56:45.485 INF OpcUa] ChannelId 0: in Opening state.
[2024-01-06 20:56:45.486 INF OpcUa] ChannelId 0: Token #0 created. CreatedAt=20:56:45.486. Lifetime=3600000.
[2024-01-06 20:56:45.658 WRN OpcUa] Certificate Validation failed. Reason=BadCertificateUntrusted. [CN=OpcPlc] [B8E6E8DD552504EA7DEE5BF28F2011A06C1450BE]
[2024-01-06 20:56:45.658 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.DefaultSessionManager] Accepting untrusted peer certificate B8E6E8DD552504EA7DEE5BF28F2011A06C1450BE, 'CN=OpcPlc' due to AutoAccept(UntrustedCertificates) set!
[2024-01-06 20:56:45.658 WRN OpcUa] Validation errors suppressed:  [CN=OpcPlc] [B8E6E8DD552504EA7DEE5BF28F2011A06C1450BE]
[2024-01-06 20:56:45.669 INF OpcUa] Security Policy: http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256
[2024-01-06 20:56:45.669 INF OpcUa] Sender Certificate: [CN=OpcPlc] [B8E6E8DD552504EA7DEE5BF28F2011A06C1450BE]
[2024-01-06 20:56:45.669 INF OpcUa] ChannelId 5: Token #1 activated. CreatedAt=20:56:45.486. Lifetime=3600000.
[2024-01-06 20:56:45.669 INF OpcUa] ChannelId 5: in Open state.
[2024-01-06 20:56:45.669 INF OpcUa] ChannelId 5: Token Expiry 01/06/2024 21:56:45, renewal scheduled in 2699862 ms.
[... 15 lines cut ...]
[2024-01-06 20:56:46.900 WRN Microsoft.Azure.IIoT.OpcUa.Protocol.Services.SubscriptionServices+SubscriptionWrapper] DataChange for subscription '1000'/'opc.tcp://$server:50000_83CF7F93' has unexpected sequenceNumber 1 vs expected 118, publishTime 01/06/2024 20:56:46
[2024-01-06 20:56:56.748 INF Microsoft.Azure.IIoT.OpcUa.Edge.Publisher.Engine.DataFlowProcessingEngine]
  DIAGNOSTICS INFORMATION for          : opc.tcp://$server:50000_83CF7F93
  # Ingestion duration                 :    00:00:12:00 (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :            498 (All time ~0.69/s; 14 in last 60s ~0.23/s)
  # Ingress ValueChanges (from OPC)    :          1,494 (All time ~2.07/s; 42 in last 60s ~0.7/s)
  # Ingress BatchBlock buffer size     :              0
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :          1,494
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :            497
  # Encoder avg Notifications/Message  :              3
  # Encoder avg IoT Message body size  :            629 (0%)
  # Encoder avg IoT Chunk (4 KB) usage :            0.2
  # Estimated IoT Chunks (4 KB) per day:         59,639
  # Outgress Batch Block buffer size   :              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :            497 (0.69/s)
  # Connection retries                 :              2
  # Opc endpoint connected?            :           True
  # Monitored Opc nodes succeeded count:              3
  # Monitored Opc nodes failed count   :              0`

Any help would be greatly appreciated. Thank you.

marcschier, 2024-01-08:
The logs show that data is flowing between plc and publisher: ` # Ingress ValueChanges (from OPC)    :          1,494 (All time ~2.07/s; 42 in last 60s ~0.7/s)`.  Is the issue that you do not see data in IoT Hub?  Here also `# Outgress IoT message count         :            497 (0.69/s)` shows data flowing to edgehub.  Routes and configuration look ok.

mtraina, 2024-01-08:
Hi Marc

exactly, I do not see any data in IoTHub: I have plugged an eventhub + ADX to the IoTHub and it is basically writing empty rows:

[image]

marcschier, 2024-01-09:
Can you use the az cli to [monutor the the iot Hub messages?](https://github.com/AzureIoTGBB/How-To-Monitor-Azure-IoT-Hub-with-Azure-CLI-2.0).  I would like to exclude the hub to add part here first given that data is published...

mtraina, 2024-01-09:
Here the output, no results:
`
az iot hub monitor-events --hub-name $iothub
Starting event monitor, use ctrl-c to stop...
`

marcschier, 2024-01-10:
Have you looked at the logs of EdgeHub and EdgeAgent inside the IoT Edge installation?  Do you seen any errors?

mtraina, 2024-01-10:
I have follow the troubleshooting page but everything seemed ok. I then removed the event-hub, added a route to storage and then I was able to see the data saved as json. I then rebuilt the route to event-hub + ADX, new ingestion and now it is working there too. Frankly I am not sure what was wrong, anyhow good it works.

Thanks Marc for the support.

One thing that still I cannot do is to see the messages via az cli (or Iot explorer): is this related to the $default consumer group being already busy (I am using the free iothub)?

closed 2024-01-15 by marcschier