Azure/Industrial-IoT #2152: Unable to Authenticate to Matrikon  OPC Server while calling Browse rest API endpoint

vishutyagi000, 2024-01-08:
Unable to Authenticate to Matrikon  OPC Server Simulation Server while calling Browse rest API endpoint.
Calling http://opcpublisher:80/v2/browse/first or http://localhost:80/v2/browse/first.
Successfully hitting the rest API endpoint however got error of invalid user/password.

**To Reproduce**

1. On an Edge Device(VM) ,Use OPCPublisher version 2.9.3
2. Call  http://opcpublisher:80/v2/browse/first or  http://localhost:80/v2/browse/first
3. Body Content to pass this post method 

{
  "Connection": {
    "Endpoint": {
      "Url": "opc.tcp://OPCSERVERIP:21381/MatrikonOpcUaWrapper",
      "securityMode": "Best",
      "securityPolicy": "http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15"
    },
    "user": {
      "type": "UserName",
      "value": {
        "UserName": "testuser",
        "Password": "password"
      }
    }
  },
  "Request": {
    "view": {
      "viewId": "ns=2;s=0:Random.Int1"
    },
    "nodeId": "ns=2;s=0:Random.Int1",
    "direction": "Forward",
    "header": {
      "namespaceFormat": "Uri"
    }
  }
} 

Getting error Username password invalid.

As per documentation  https://github.com/Azure/Industrial-IoT/blob/release/2.9.3/docs/opc-publisher/definitions.md#credentialmodel

CredentailModel is like 
[image]
But in Source code it is like
[image]

Here "value" property in user object is not defined in documentation
"user": {
      "type": "UserName",
      "value": {
        "UserName": "testuser",
        "Password": "password"
      }
    }

Not sure how to pass credentials  of opc server as this is not described in documentation.

[image]

**Expected behavior**
It should browse the OPC Simulation server.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: [Ubuntu]
 - Version [20.04]
 - OPCpublisher version -2.9.3

**Additional context**
Please share the example request to call http://localhost:80/v2/browse/first or http://localhost:80/v2/browse  or any historical data related API.

API

marcschier, 2024-01-08:
Yes, it is not documented well.  Use "user" and "password" instead of "UserName" and "Password" as property strings.

vishutyagi000, 2024-01-09:
@marcschier  Thank you its working ,is there any way where we can generate these json input for rest endpoints ,will be very helpful.
However thanks for you quick response, should i close this issue?

commit d527fddf (2024-01-10): Concrete types for credentials (#2154)

* Update System.Web.Identity to latest
* Fix #2152
closed 2024-01-10 by marcschier