Azure/Industrial-IoT #2156: IIoT platform + Simulation deployed on Azure: no endpoints?

greg-mcnamara, 2024-01-13:
**Describe the bug**
I have deployed the latest IIoT platform services and Simulation components on Azure. I can connect to the API via Postman, and I can see the gateways via GET registry/v2/gateways, but when I query endpoints there are none listed. Should I see endpoints already discovered in the simulation deployment, or is there another step I need to do?

I wondered if the problem was related to discovery configuration. I noticed that GET registry/v2/discovery returns a single discoverer with "discovery" set to "Off". I have tried to enable discovery via the API as per the instructions but when I check again it's still set to "Off".

**To Reproduce**
Steps to reproduce the behavior:
1. Install Industrial-IoT platform with services and simulation.
2. Connect to the web API and query endpoints, none found.
3. Query the simulated Linux gateway and receive the following response:

 ```
{
    "gateway": {
        "id": "linuxgateway0-54zwr5s",
        "connected": true
    },
    "modules": {
        "supervisor": {
            "id": "device_linuxgateway0-54zwr5s_module_publisher",
            "connected": true,
            "version": "2.9.3.1",
            "apiKey": "x9qVHCVP6Qgo61YNEqGAGwsCmV4="
        },
        "publisher": {
            "id": "device_linuxgateway0-54zwr5s_module_publisher",
            "connected": true,
            "version": "2.9.3.1",
            "apiKey": "x9qVHCVP6Qgo61YNEqGAGwsCmV4="
        },
        "discoverer": {
            "id": "device_linuxgateway0-54zwr5s_module_publisher",
            "discovery": "Off",
            "connected": true,
            "version": "2.9.3.1",
            "apiKey": "x9qVHCVP6Qgo61YNEqGAGwsCmV4="
        }
    }
 }
```

5. Try to update the discoverer with PUT/PATCH registry/v2/discovery/[discovererId] methods. Updates accepted ok by the API.
6. Check discoverer, "discovery" still "Off".

**Expected behavior**
Discovery on the simulated gateway would be on by default, and endpoints on the PLC simulation would be discoverable via the API.

marcschier, 2024-07-01:
The issue here is that the functionality to update the "discovery" state has been removed in 2.9 with the consolidation of all functionalities of the OPC Discovery module into the OPC Publisher. The code was left to aid in porting, but it is misleading as you would only see that it is unsupported by looking at the web api server logs.

We did not carry this capability forward, because a) it was not desired to have constantly scanning capability, and b) originally the discovery state was maintained as a desired property in the IoT Device Twin.  However, you can use the programmatic "Discover_V2" API to kick off a one-time discovery run, which takes the discovery mode and optional configuration as arguments.

We will remove the misleading documentation and API from 2.9.10.  Sorry for the confusion.

marcschier, 2024-07-01:
#2276

closed 2024-07-01 by marcschier