Azure/Industrial-IoT #2159: Add OPC UA method calls and Configuration API to DAPR transport

Haeavar, 2024-01-23:
**Is your feature request related to a problem? Please describe.**
We are planning to use the OpcPublisher DAPR integration. But we need at least the feature to call OPC UA methods using DAPR.
In addition, it would be great to call the configuration API via DAPR.

**Describe the solution you'd like**
Call OPC UA methods via DAPR 
Use the configuration API via dapr.

**Describe alternatives you've considered**
The alternative would be using the MQTT protocol in between.

Please let us now if you have plans to add this feature and when it would be available.

Haeavar, 2024-01-29:
We are now using the DAPR service invocation (on the OpcPublisher REST Api) from the same DAPR sidecar as we use for the DAPR PubSub messaging. This works very well. But I'm wondering, why this is not documented and if this will be supported in future releases.

marcschier, 2024-01-29:
Ah, I was about to reply that this should already be possible.  Since the service invoke in DAPR is just an adapter over http endpoints with injected extra goodness.  

To help better document, what would you have wanted to see there the first time you checked it out?  

And for configuration, to confirm, you would find it useful if the configuration comes through [configuration services](https://docs.dapr.io/developing-applications/building-blocks/configuration/configuration-api-overview/), correct?  What configuration is it that is most important for you?

Haeavar, 2024-01-29:
Sorry for the confusion, I meant the OpcPublisher own REST Api (https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/api.md). Not the DAPR configuration. We do not want to use the Api directly but only using the DAPR service invocations.

A few words about this possibility in the /docs/opc-publisher/transports.md#dapr documentation would be enough. Currently, it marks the 'Configuration API' as 'no'.

marcschier, 2024-02-08:
Updated the docs to reference the service invoke info on dapr.

closed 2024-02-08 by marcschier