Azure/Industrial-IoT #2165: Cyclic read above 300 nodes not working.

b-enigma-con, 2024-01-30:
**Describe the bug**
OPC Publisher 2.9.4-preview3. We have a publishednodes file containing 1000 nodes, cyclic read every 5 seconds. I noticed that the expected output was not even close to the configured settings (every 5 seconds I expect 1000 values). I noticed that the output matches the configuration up to until 300 configured nodes. Above 300 nodes the output seems not to increase anymore, also the log diagnostics seems to change above this number. The publishednodes.json looks like this:

```
[
  {
    "EndpointUrl": "opc.tcp://xxx.xxx.xxx.xxx:49320",
    "UseSecurity": true,
    "OpcNodes": [
      {
        "Id": "i=2294",
        "UseCyclicRead": true,
        "OpcSamplingInterval": 5000
      },
      {
        "Id": "i=2737",
        "UseCyclicRead": true,
        "OpcSamplingInterval": 5000
      },
      {
        "Id": "i=2736",
        "UseCyclicRead": true,
        "OpcSamplingInterval": 5000
      },
      {
        "Id": "i=2735",
        "UseCyclicRead": true,
        "OpcSamplingInterval": 5000
      },
      {
        "Id": "i=2272",
        "UseCyclicRead": true,
        "OpcSamplingInterval": 5000
      },
      {
        "Id": "i=2994",
        "UseCyclicRead": true,
        "OpcSamplingInterval": 5000
      },
      {
        "Id": "i=2267",
        "UseCyclicRead": true,
        "OpcSamplingInterval": 5000
      },
      {
        "Id": "ns=2;s=xxx",
        "UseCyclicRead": true,
        "OpcSamplingInterval": 5000
      },
      {
        "Id": "ns=2;s=yyy",
        "UseCyclicRead": true,
        "OpcSamplingInterval": 5000
      },
      {
        "Id": "ns=2;s=xxx",
        "UseCyclicRead": true,
        "OpcSamplingInterval": 5000
      },
      ************ SNIP ************
    ]
  }
]
```

### Diagnostics output 250 nodes
[24-01-30 08:57:07.8584] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c):0 in session opc.tcp://xxx.xxx.xxx.xxx:49320_774158C3_<<UnknownWriterGroup>>.
[24-01-30 08:57:07.8735] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c):74'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 08:57:07.9221] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession[0]
      Complex type system loaded into client opc.tcp://xxx.xxx.xxx.xxx:49320_774158C3_<<UnknownWriterGroup>>.
[24-01-30 08:57:08.0123] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      **Now monitoring 250 (Good:250/Bad:0) nodes in subscription** <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c):74.
[24-01-30 08:57:13.0159] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherDiagnosticCollector[0]
      Tracking diagnostics for <<UnknownWriterGroup>> was (re-)started.
[24-01-30 08:57:13.0160] info: Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink[0]
      Started data flow with message from subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c) on opc.tcp://xxx.xxx.xxx.xxx:49320/.

### Diagnostics output 300 nodes
[24-01-30 09:06:30.6572] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c):0 in session opc.tcp://xxx.xxx.xxx.xxx:49320_FF381093_<<UnknownWriterGroup>>.
[24-01-30 09:06:30.6740] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c):78'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 09:06:30.7245] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession[0]
      Complex type system loaded into client opc.tcp://xxx.xxx.xxx.xxx:49320_FF381093_<<UnknownWriterGroup>>.
[24-01-30 09:06:30.8206] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      **Now monitoring 300 (Good:300/Bad:0) nodes in subscription** <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c):78.
[24-01-30 09:06:35.8241] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherDiagnosticCollector[0]
      Tracking diagnostics for <<UnknownWriterGroup>> was (re-)started.
[24-01-30 09:06:35.8242] info: Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink[0]
      Started data flow with message from subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c) on opc.tcp://xxx.xxx.xxx.xxx:49320/.

### Diagnostics output 400 nodes
[24-01-30 09:02:43.6898] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c):0 in session opc.tcp://xxx.xxx.xxx.xxx:49320_0A088325_<<UnknownWriterGroup>>.
[24-01-30 09:02:43.7055] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c):76'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 09:02:43.7566] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession[0]
      Complex type system loaded into client opc.tcp://xxx.xxx.xxx.xxx:49320_0A088325_<<UnknownWriterGroup>>.
[24-01-30 09:02:48.8729] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherDiagnosticCollector[0]
      Tracking diagnostics for <<UnknownWriterGroup>> was (re-)started.
[24-01-30 09:02:48.8731] info: Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink[0]
      Started data flow with message from subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c) on opc.tcp://xxx.xxx.xxx.xxx:49320/.
**Note, I'm missing the "Now monitoring x (Good:x/Bad:0) nodes in subscription" line and I'm not receiving the right amount of read nodes.** 

### Diagnostics output 500 nodes
[24-01-30 09:07:44.7320] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c):0 in session opc.tcp://xxx.xxx.xxx.xxx:49320_A6322432_<<UnknownWriterGroup>>.
[24-01-30 09:07:44.7521] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c):79'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 09:07:44.8004] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession[0]
      Complex type system loaded into client opc.tcp://xxx.xxx.xxx.xxx:49320_A6322432_<<UnknownWriterGroup>>.
[24-01-30 09:07:49.9437] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherDiagnosticCollector[0]
      Tracking diagnostics for <<UnknownWriterGroup>> was (re-)started.
[24-01-30 09:07:49.9440] info: Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink[0]
      Started data flow with message from subscription <<UnknownDataSet>>_($b6589fc6ab0dc82cf12099d1c2d40ab994e8410c) on opc.tcp://xxx.xxx.xxx.xxx:49320/.
**Note, I'm missing the "Now monitoring x (Good:x/Bad:0) nodes in subscription" line and I'm not receiving the right amount of read nodes.** 

**To Reproduce**
Steps to reproduce the behavior:
1. Create publishednodes.json file as above with 300 nodes, restart OPC publisher and inspect diagnostics and output.
2. Create publishednodes.json file as above with 500 nodes, restart OPC publisher and inspect diagnostics and output. Output does not match 500 nodes and diagnostics does not show "Now monitoring x (Good:x/Bad:0) nodes in subscription".

**Expected behavior**
The right amount of read output and correct log diagnostics.

b-enigma-con, 2024-01-30:
Another finding: when using 1007 nodes (maxnodesperdataset has default value, 1000), I see that there's no report about the first 1000 nodes (/subscription), but the remaining 7 nodes in the second subscription are being mentioned in the diagnostics.

[24-01-30 10:29:20.2471] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      New Session opc.tcp://xxx.xxx.xxx.xxx:49320_84A938D8_<<UnknownWriterGroup>> created with endpoint opc.tcp://xxx.xxx.xxx.xxx:49320/ (opc.tcp://xxx.xxx.xxx.xxx:49320).
[24-01-30 10:29:20.2472] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Client opc.tcp://xxx.xxx.xxx.xxx:49320_84A938D8_<<UnknownWriterGroup>> [state:Ready|refs:2] CONNECTED to opc.tcp://xxx.xxx.xxx.xxx:49320/!
[24-01-30 10:29:20.2610] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($ec6116c58e227cbac93cd3136e242d705e1ac3c6):0 in session opc.tcp://xxx.xxx.xxx.xxx:49320_84A938D8_<<UnknownWriterGroup>>.
[24-01-30 10:29:20.2775] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($ec6116c58e227cbac93cd3136e242d705e1ac3c6):83'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 10:29:20.3298] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession[0]
      Complex type system loaded into client opc.tcp://xxx.xxx.xxx.xxx:49320_84A938D8_<<UnknownWriterGroup>>.
[24-01-30 10:29:20.5879] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($544391a18285ce8424d46480e77254ca93ef8088):0 in session opc.tcp://xxx.xxx.xxx.xxx:49320_84A938D8_<<UnknownWriterGroup>>.
[24-01-30 10:29:20.5891] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($544391a18285ce8424d46480e77254ca93ef8088):84'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 10:29:20.6078] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      **Now monitoring 7 (Good:7/Bad:0) nodes in subscription** <--_I expected this line for the first 1000 nodes as well._ <<UnknownDataSet>>_($544391a18285ce8424d46480e77254ca93ef8088):84.
[24-01-30 10:29:25.5757] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherDiagnosticCollector[0]
      Tracking diagnostics for <<UnknownWriterGroup>> was (re-)started.
[24-01-30 10:29:25.5760] info: Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink[0]
      Started data flow with message from subscription <<UnknownDataSet>>_($ec6116c58e227cbac93cd3136e242d705e1ac3c6) on opc.tcp://xxx.xxx.xxx.xxx:49320/.

b-enigma-con, 2024-01-30:
I downgraded from OPC Publisher OPC Publisher 2.9.4-preview3 to OPC Publisher 2.9.3 with the same behavior (I receive way less data values than expected). Then I downgraded to OPC Publisher 2.9.2 and I receive way too much datavalues (1007 configured nodes, cyclic read every 5 seconds): I receive (random) 208234 values in a minute, while I would expect approx 12084 values per minute (1007 * 12). Eventually, I think this causes the Memory issues it all started with (see linked issue).

  DIAGNOSTICS INFORMATION for          : <<UnknownWriterGroup>> (OPC Publisher 2.9.2.1)
  # Time                               : 2024-01-30T11:13:38.4252431Z
  # Ingestion duration                 :    **00:00:05:52** (dd:hh:mm:ss)
  # Ingress DataChanges (from OPC)     :        615,603 (All time ~1746.88/s; 186818 in last 60s ~**3113.63/s**)
  # Ingress ValueChanges (from OPC)    :        615,603 (All time ~1746.88/s; 186818 in last 60s ~3113.63/s)
  # of which are Heartbeats            :              0
  # of which are Cyclic reads          :        **615,603**
  # Ingress EventData (from OPC)       :              0
  # Ingress Events (from OPC)          :              0
  # Ingress BatchBlock buffer size     :            272
  # Encoding Block input/output size   :              0 | 0
  # Encoder Notifications processed    :        615,332
  # Encoder Notifications dropped      :              0
  # Encoder IoT Messages processed     :          1,418
  # Encoder avg Notifications/Message  :            434
  # Encoder worst Message split ratio  :              0
  # Encoder avg IoT Message body size  :        206,184
  # Encoder avg IoT Chunk (4 KB) usage :           50.3
  # Estimated IoT Chunks (4 KB) per day:              0
  # Outgress input buffer count        :              0
  # Outgress input buffer dropped      :              0
  # Outgress IoT message count         :          1,406 (3.92/s)
  # Connection retries                 :              1
  # Opc endpoint connected?            :           True
  # Monitored Opc nodes succeeded count:              7
  # Monitored Opc nodes failed count   :              0

Bottom line: I think something is seriously wrong with cyclic reads of lots amounts of nodes. With 1000 nodes cyclic read, every 5 seconds, I experience the following issues:
- V2.9.2 Way too much cyclic read data values, eventually seems to be leading to memory issues
- V2.9.3 Way too less cyclic read data values, diagnostics not complete/clear.
- V2.9.4-preview3 Idem to V2.9.3.
-

marcschier, 2024-01-30:
What we have seen with some (more than we expected) servers is that they have a hard time when loading the metadata, it takes extremely long.  Can you try 2 things: 1) use `--dm`  to disable emitting pub/sub metadata and 2) use `--npd=100` to split the subscription into 11?  

One other thing we found is that while we honor the operation limits of the server, we do not translate this to observing the channel limits, so if the server supports 1000 read value requests, if these or their responses are larger than the max message size on the channel, then the read call fails.  This could be the root cause here as it seems the magic boundary there is 300.  What can be tried then is to set different sampling rates for batches of 300, and then it should work (as the batches of read requests will then be 300).  If that is so, it might be best to limit the sampler size to the `--npd` value or allow limiting the actually applied operations limits via some configuration.

I will try to test with different server side configurations to see where we break and what best to do then.

b-enigma-con, 2024-01-30:
I configured the OPC Publisher with "--npd=100" and "--dm". I see that we get more output, but not yet for all 1007 nodes. If I look at the logs, it seems that 9 subscriptions are created, covering 807 nodes in total, instead of 11 subscriptions covering 1007 nodes:

[24-01-30 13:38:51.6520] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      New Session opc.tcp://xxx.xxx.xxx.xxx:49320_0377757F_<<UnknownWriterGroup>> created with endpoint opc.tcp:// xxx.xxx.xxx.xxx:49320/ (opc.tcp:// xxx.xxx.xxx.xxx:49320).
[24-01-30 13:38:51.6521] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
      Client opc.tcp:// xxx.xxx.xxx.xxx:49320_0377757F_<<UnknownWriterGroup>> [state:Ready|refs:11] CONNECTED to opc.tcp:// xxx.xxx.xxx.xxx:49320/!
[24-01-30 13:38:51.6685] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($ec6116c58e227cbac93cd3136e242d705e1ac3c6):0 in session opc.tcp:// xxx.xxx.xxx.xxx:49320_0377757F_<<UnknownWriterGroup>>.
[24-01-30 13:38:51.6885] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($ec6116c58e227cbac93cd3136e242d705e1ac3c6):95'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 13:38:51.7384] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession[0]
      Complex type system loaded into client opc.tcp://xxx.xxx.xxx.xxx:49320_0377757F_<<UnknownWriterGroup>>.
[24-01-30 13:38:51.7970] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      **Now monitoring 100 (Good:100/Bad:0) nodes in subscription** <<UnknownDataSet>>_($ec6116c58e227cbac93cd3136e242d705e1ac3c6):95.
[24-01-30 13:38:51.8115] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($5e0dced00fc26d65d2d9fe9cbb119fa902dd3393):0 in session opc.tcp:// xxx.xxx.xxx.xxx:49320_0377757F_<<UnknownWriterGroup>>.
[24-01-30 13:38:51.8127] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($5e0dced00fc26d65d2d9fe9cbb119fa902dd3393):96'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 13:38:51.8394] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      **Now monitoring 100 (Good:100/Bad:0) nodes in subscription** <<UnknownDataSet>>_($5e0dced00fc26d65d2d9fe9cbb119fa902dd3393):96.
[24-01-30 13:38:51.8405] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($b534c754abe48186900b102b4ee00b0d88317dc2):0 in session opc.tcp:// xxx.xxx.xxx.xxx:49320_0377757F_<<UnknownWriterGroup>>.
[24-01-30 13:38:51.8414] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($b534c754abe48186900b102b4ee00b0d88317dc2):97'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 13:38:51.8892] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      **Now monitoring 100 (Good:100/Bad:0) nodes in subscription** <<UnknownDataSet>>_($b534c754abe48186900b102b4ee00b0d88317dc2):97.
[24-01-30 13:38:51.8904] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($ee48ecec6568040b8e1430209a521ce4637a5fc7):0 in session opc.tcp:// xxx.xxx.xxx.xxx:49320_0377757F_<<UnknownWriterGroup>>.
[24-01-30 13:38:51.8912] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($ee48ecec6568040b8e1430209a521ce4637a5fc7):98'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 13:38:51.9169] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($85c36854981d4170d4e6a6e2e3850d6eb4dac8c8):0 in session opc.tcp:// xxx.xxx.xxx.xxx:49320_0377757F_<<UnknownWriterGroup>>.
[24-01-30 13:38:51.9178] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Successfully created subscription <<UnknownDataSet>>_($85c36854981d4170d4e6a6e2e3850d6eb4dac8c8):99'.
      Actual (revised) state/desired state:
      # PublishingEnabled False/False
      # PublishingInterval 1000/1000
      # KeepAliveCount 10/10
      # LifetimeCount 100/100
[24-01-30 13:38:51.9440] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      **Now monitoring 100 (Good:100/Bad:0) nodes in subscription** <<UnknownDataSet>>_($85c36854981d4170d4e6a6e2e3850d6eb4dac8c8):99.
[24-01-30 13:38:51.9452] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
      Creating new disabled subscription <<UnknownDataSet>>_($de702e824c0081a4fc4108c2bcbd34f5e85cc2db):0 in session opc.tcp:// xxx.xxx.xxx.xxx:49320_0377757F_<<UnknownWriterGroup>>.
[... cut ...]
      **Now monitoring 7 (Good:7/Bad:0) nodes in subscription** <<UnknownDataSet>>_($544391a18285ce8424d46480e77254ca93ef8088):105.
[24-01-30 13:38:56.8046] info: Azure.IIoT.OpcUa.Publisher.Services.PublisherDiagnosticCollector[0]
      Tracking diagnostics for <<UnknownWriterGroup>> was (re-)started.
[24-01-30 13:38:56.8049] info: Azure.IIoT.OpcUa.Publisher.Services.NetworkMessageSink[0]
      Started data flow with message from subscription <<UnknownDataSet>>_($ec6116c58e227cbac93cd3136e242d705e1ac3c6) on opc.tcp:// xxx.xxx.xxx.xxx:49320/.

marcschier, 2024-02-02:
This should work now in 2.9.4-preview4, I am collecting more work in this area in #2169 .

closed 2024-02-02 by marcschier