Azure/Industrial-IoT #2166: Certificate issue during migration of OPCPublisher 2.8 to OPCPublisher 2.9.3

vishutyagi000, 2024-01-31:
# Migrating from OPCPublisher 2.8 to OPCPublisher 2.9.3

1. On an IoT Edge Device(Virtual Machine) 
2. Deploy 2.9.3 version instead of opcpublisher 2.8

When migrating from OPCPublisher 2.8 to OPCPublisher 2.9.3, it's crucial to ensure that the existing certificate is used instead of generating a new one. This is to avoid the need to accept the certificate again on the OPC server.

Despite successful deployment and operation of the new module, a new certificate is created instead of using the existing one (created by the older version, OPCPublisher 2.8). 
In Below screens shot we can see a self generated certificate was already there on Jan 10 but instead of using already created certificate the new version is creating new one as we can see a new certificate is created on Jan 31.
It happening in both cases  migrating 2.8 to 2.9.3 or vice versa .

[image]

[image]

## Container Create Options

The Container Create Options used for both 2.8 and 2.9.3 versions are as follows:

```json
"Cmd": [
    "publisher",
    "--pf=/appdata/publishednodes.json",
    "--lf=/appdata/publisher.log",
    "--aa",
    "--me=Json",
    "--mm=PubSub",
    "--fm=false",
    "--fd=false",
    "--bs=250",
    "--si=10",
    "--di=60",
    "--tm=True",
    "--ih=Mqtt",
    "--hb=3600",
    "--ot=300000",
    "--tp=/appdata/pki/trusted",
    "--rp=/appdata/pki/rejected",
    "--ap=/appdata/pki/own",
    "--ip=/appdata/pki/issuer"
],
 "Binds": [
            "/etc/aziot/opcuaconfig:/appdata"
        ],
```

## Environment
Ubuntu 20.04

The creation of a new self-generated certificate during migration leads to the issue of having to accept the certificate repeatedly if we decide to switch OPC publisher module versions.

Please suggest .

vishutyagi000, 2024-04-02:
@marcschier Any update on this, I can see milestone 2.9.6 is past due .

marcschier, 2024-06-17:
The issue is due to the domain name being generated differently than in 2.8.  In 2.8 the gateway or device id is used as host name if running in IoT Edge, otherwise the Dns.GetHostName is used.  In 2.9 this was broken, and broken even more in 2.9.6. Therefore the subject name of a 2.8 certificate is different from a 2.9 cert.  

This will be fixed and back to 2.8 behavior in 2.9.9.  The problem is that now there are 3 different behaviors when generating the self-sign cert, sigh :-(.  

In all versions including 2.9 you can work around this by setting the 2.8 certificate's subjectname explicitly using the command line option `--sn`.  You can find the subject name by opening the previous certificate, but as DC it will have the IoT Edge gateway's name.  

2.9.9 will have an additional command line option `--cfa` to configure the publisher from the first certificate with highest NotAfter, using its applicationUri, domainName and subject name in the "own/certs" folder.  This setting will make setup a bit easier, also when using your own generated certificates.

closed 2024-06-18 by marcschier