Azure/Industrial-IoT #2167: Strict mode in 2.9.x seems to prevent changing messaging mode

NoTuxNoBux, 2024-02-01:
**Describe the bug**
In 2.8.x, I could pass `--strict --messagingmode=Samples` to use strict mode and maintain the `Samples` messaging mode. In 2.9.3, Enabling strict mode seems to force a single format (I think it's the `PubSub` one) and trying to change the messaging mode appears to have no effect.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the publisher with `--strict --messagingmode=Samples`
2. Observe Samples not being used as data format.

**Expected behavior**
`Samples` is used as data format, or alternatively a message in the [command line options](https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/commandline.md) reference that both modes are incompatible if this is the case. Currently it just mentions the defaults change with strict mode, but not that it's incompatible.

**Desktop (please complete the following information):**
 - OS: Linux x86-64

**Additional context**
Leaving out `--strict` on 2.9.3 fixes the issue, but probably also disables strict mode.

Not sure if relevant, but I'm building a custom SignalR transport modelled after the MQTT one, so in my transport is where I'm observing the data formats.

marcschier, 2024-02-01:
Hi, you are correct. 

`--strict` is new in 2.9, it is one of the command line options silently discarded in 2.8.  Strict means strict compliance with OPC UA Part 6/14.  Samples mode was our invention before Part14 even existed.  

That said, a log message and better documentation would be very useful, agreed.

marcschier, 2024-02-07:
Instead of logging / throwing, now allowing the overriding (just like other settings can be overriden, e.g., namespace mode) and updated the documentation and cli options as such.

closed 2024-02-07 by marcschier