Azure/Industrial-IoT #2181: ApiKey required with 2.9.4-preview4

Haeavar, 2024-02-05:
**Describe the bug**
With the latest preview version 2.9.4-preview4 we are no longer able to use the OpcPublisher REST Api. It requires an ApiKey. Since we do not use the IoTEdge we do not have this info available in the device twin.
Do you have any other options to get the ApiKey? Or is it possible to disable the authorization check?

**To Reproduce**
Steps to reproduce the behavior:
1. Start OpcPublisher with --DefaultTransport=Dapr

With 2.9.4-preview3 we were still be able to access the REST Api without an ApiKey.

marcschier, 2024-02-06:
I am adding a way to set the Api Key via command line or using an "ApiKey" environment variable.  However, I will also allow loading secrets from a file (.env format).  The file name can be set through a "ADDITIONAL_CONFIGURATION" environment variable, the default will be /run/secrets/.env.  The content then can include "ApiKey=<key>" and this will then securely be included in the container.  This should nicely integrate with docker secrets.

closed 2024-02-07 by marcschier

Haeavar, 2024-02-09:
Hi @marcschier . Do you have already a new preview build containing this feature?