Azure/Industrial-IoT #2185: Loading Complex type system takes a long time

Haeavar, 2024-02-09:
**Describe the bug**
We are using an aggregation OPC UA server to aggregate multiple OPC UA server. With increasing number of aggregated server it takes some time until the Complex type system is loaded (currently around 15sec).
For the startup of the OpcPublisher this is okay. But for REST request on the Api like read nodes its annoying.

Here is the OpcPublisher log showing the times for the /v2/read request:
```

2024-02-09T09:17:41.783+01:00 Client opc.tcp://localhost:21878_EC0C5EDC [state:Ready|refs:1] CONNECTED to opc.tcp://localhost:21878/!
2024-02-09T09:17:55.247+01:00[24-02-09 08:17:55.2471] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSession[0]
2024-02-09T09:17:55.247+01:00 => SpanId:5839bbff6556dffd, TraceId:5b7e7cfc2e868402f0b4932e61c6f776, ParentId:0000000000000000 => ConnectionId:0HN198DRNQ2JU => RequestPath:/v2/read RequestId:0HN198DRNQ2JU:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueReadAsync (Azure.IIoT.OpcUa.Publisher.Module)
2024-02-09T09:17:55.247+01:00 Complex type system loaded into client opc.tcp://localhost:21878_EC0C5EDC.

```

Would it be possible to add an option to the request or connection model to skip the complex system type loading? Currently we are reading only nodes without complex types and this would help a lot.

**Additional context**
We are using OpcPublisher mcr.microsoft.com/iotedge/opc-publisher:2.9.4-preview3

marcschier, 2024-02-09:
Can you try `---disablecomplextypepreloading` option?  

We can look at making it also part of the connection model in the next release.

Haeavar, 2024-02-12:
Hi @marcschier option `--disablecomplextypepreloading` works for fast reading of simple values but now, the subscribed values with complex values are published **no longer as json**.
For the configured subscriptions, we still want to have deserialized complex values but for reading nodes, we want to switch it off.

Haeavar, 2024-02-19:
@marcschier as I understand it correctly, the publisher will create always a new session for each /v2/read request. Is that correct? If it would be possible, to reuse a session for multiple  /v2/read requests (without loading complex types for 2nd and later requests), than this would solve my issue.

marcschier, 2024-02-19:
You can set a linger timeout which will keep the client alive for the time specified (--cl) even if no service calls are made.

marcschier, 2024-02-24:
`--disablecomplextypesystem` will disable any complex type system loading for subscriptions/writers/writergroups.

New connection options in `ConnectionModel` will allow to do so per connection when using the service APIs.  This will be so that there are 2 sessions created if a connection with and without this option is used.

closed 2024-02-24 by marcschier