Azure/Industrial-IoT #2186: Discovery for reverse connect

Haeavar, 2024-02-09:
**Is your feature request related to a problem? Please describe.**
We are using the reverse connect feature due to the firewalls protecting our OPC UA servers. Due to the firewalls, a discovery (provided by the REST Api) is not possible. But it would be great, if the REST Api would have a possibility to discovery the reverse connect *ReverseHello* message of the clients and reporting its endpoint url. 

**Describe the solution you'd like**
An additional /discovery route to discover all clients which sends *ReverseHello* messages and report its hostname and port (preferable the endpointUrl used within the endpointUrl.

marcschier, 2024-02-09:
This might be difficult due to missing capabilities in the ua stack.  I will look into this, for now moving to 2.9.6.  Any suggestions or contributions welcome.

marcschier, 2024-02-09:
@mregen