Azure/Industrial-IoT #2190: TopicTemplate and "Sub-Topics"

quality-leftovers, 2024-02-13:
**Is your feature request related to a problem? Please describe.**
We would like to route some DataSetWriterGroup to a different nested topic path.

**Describe the solution you'd like**
We'd like to be able to define or override attributes that can be used in TopicTemplate in the configuration for each WriterGroup.

**Describe alternatives you've considered**
Using the {DataSetWriterGroupand} in the topic template and setting it to something like "mytopicA/mySubTopicB" and "myTopicC/mySubTopicD". This works but we are wondering whether this is officially supported or might "break" in future publisher versions

**Additional context**
-

marcschier, 2024-02-14:
There should probably be a better way to do this.  Especially because part 14 permits mqtt topic path and query chars in names.

Haeavar, 2024-02-20:
Hi @marcschier. I created a PR #2193  which address this issue. This PR will enable us to define different topics für each datasetwriter. Please let me know whether are you ok with the changes or if there some things missing. Thanks.

marcschier, 2024-02-24:
In 2.9.5 (main branch) it is possible to set the topic per group, writer (and even at node level., but this is experimental). That way you should be able to accomplish what you want here.  Please be aware that support on this feature is only via GitHub.

closed 2024-02-24 by marcschier