Azure/Industrial-IoT #2192: "WriterGroupTransport" in PublishedNodesEntryModel will be overriden with DefaultTransport

Haeavar, 2024-02-19:
**Describe the bug**
We are configuring the OpcPublisher with following param:
    - DaprConnectionString=PubSubComponent=pubsub-opcpublisher;GrpcPort=6011;
    - MqttClientConnectionString=HostName=mosquitto;Port=1883;Protocol='v5'
    - DefaultTransport=Mqtt

Than, we configure the OpcPublisher using the REST Api PATCH `/v2/configuration` with two different dataSetWriters
1) with `"WriterGroupTransport": "Dapr",` 
2) with `"WriterGroupTransport": "Mqtt",`

with a StausCode 200.

Then we read the configuration with GET `/v2/configuration?includeNodes=false` with following result:
1) "WriterGroupTransport": "mqtt",
2) "WriterGroupTransport": "mqtt",

If we define a DefaultTransport= Dapr, we will get all DataSetWriters with Dapr transport.

**Expected behavior**
I would expect, that the transport is used which is defined in the dataSetWriters PublishedNodesEntryModel 

**Additional context**
image: mcr.microsoft.com/iotedge/opc-publisher:2.9.4

marcschier, 2024-02-19:
The behavior up till now is that groups are only differentiated using the writer group ID, and that takes the rest of the group properties not into account, which are then all overriden/coalesced by the last entry with the same group id.  I will try to address this on 2.9.5.  I think it is possible to do without breaking backcompat.

Haeavar, 2024-02-21:
The story is, that we want to use one OpcPublisher instance publishing telemetry to different targets. One in the cloud and one on premise. With a separate topic for each datasetWriter this would solve our requirement as well (#2190).
Maybe the documentation could be improved which properties can be overridden and which not.

marcschier, 2024-06-09:
Writers can already have separate topics via QueueName field.  I guess that would solve routing.  If you want to use different transports, those are at the group level, just use the writer name also as group name and the separate transport should also work. Let me know if you encounter issues.

marcschier, 2024-06-18:
Documentation updated.

closed 2024-06-18 by marcschier