Azure/Industrial-IoT #2197: OPC Publisher 2.9.4's Prometheus metrics partially missing probably due to unknown total size limit

jacqueskang, 2024-03-04:
**Describe the bug**
It appears that OPC Publisher v2.9.4's prometheus endpoint (http://<opc_publisher>/metrics) does not always expose all metrics documented in [this article](https://azure.github.io/Industrial-IoT/opc-publisher/observability.html#available-metrics).

It seems that the metrics are somehow truncated when the total data size exceeds certain threshold.

**To Reproduce**

Please find below the returned content of metrics endpoint. As example the `iiot_edge_publisher_messages_total` is not returned.

```text
# TYPE dns_lookup_duration_seconds histogram
# UNIT dns_lookup_duration_seconds seconds
# HELP dns_lookup_duration_seconds Measures the time taken to perform a DNS lookup.
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="0.005"} 159 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="0.01"} 159 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="0.025"} 160 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="0.05"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="0.075"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="0.1"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="0.25"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="0.5"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="0.75"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="1"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="2.5"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="5"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="7.5"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="10"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbompx0001.mbo.cz.corp",le="+Inf"} 161 1709313757021
dns_lookup_duration_seconds_sum{dns_question_name="czmbompx0001.mbo.cz.corp"} 0.17188980000000012 1709313757021
dns_lookup_duration_seconds_count{dns_question_name="czmbompx0001.mbo.cz.corp"} 161 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="0.005"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="0.01"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="0.025"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="0.05"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="0.075"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="0.1"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="0.25"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="0.5"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="0.75"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="1"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="2.5"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="5"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="7.5"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="10"} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="",le="+Inf"} 1 1709313757021
dns_lookup_duration_seconds_sum{dns_question_name=""} 1.34E-05 1709313757021
dns_lookup_duration_seconds_count{dns_question_name=""} 1 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="0.005"} 3 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="0.01"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="0.025"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="0.05"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="0.075"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="0.1"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="0.25"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="0.5"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="0.75"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="1"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="2.5"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="5"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="7.5"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="10"} 4 1709313757021
dns_lookup_duration_seconds_bucket{dns_question_name="czmbomii2103.ls.ege.ds",le="+Inf"} 4 1709313757021
dns_lookup_duration_seconds_sum{dns_question_name="czmbomii2103.ls.ege.ds"} 0.018285299999999997 1709313757021
dns_lookup_duration_seconds_count{dns_question_name="czmbomii2103.ls.ege.ds"} 4 1709313757021

# TYPE process_runtime_dotnet_gc_collections_count_total counter
# HELP process_runtime_dotnet_gc_collections_count_total Number of garbage collections that have occurred since process start.
process_runtime_dotnet_gc_collections_count_total{generation="gen2"} 39 1709313757021
process_runtime_dotnet_gc_collections_count_total{generation="gen1"} 2175 1709313757021
process_runtime_dotnet_gc_collections_count_total{generation="gen0"} 12367 1709313757021
[... 309 lines cut ...]
iiot_edge_publisher_messages_duration_bucket{publisherId="iot-gptmpxp01-euw-pro-core-01_device_czmbompx0001_module_opc-publisher",siteId="<<UnknownSite>>",writerGroupId="x",le="7500"} 57372 1709313757021
iiot_edge_publisher_messages_duration_bucket{publisherId="iot-gptmpxp01-euw-pro-core-01_device_czmbompx0001_module_opc-publisher",siteId="<<UnknownSite>>",writerGroupId="x",le="10000"} 57372 1709313757021
iiot_edge_publisher_messages_duration_bucket{publisherId="iot-gptmpxp01-euw-pro-core-01_device_czmbompx0001_module_opc-publisher",siteId="<<UnknownSite>>",writerGroupId="x",le="+Inf"} 57373 1709313757021
iiot_edge_publisher_messages_duration_sum{publisherId="iot-gptmpxp01-euw-pro-core-01_device_czmbompx0001_module_opc-publisher",siteId="<<UnknownSite>>",writerGroupId="x"} 391736 1709313757021
iiot_edge_publisher_messages_duration_count{publisherId="iot-gptmpxp01-euw-pro-core-01_device_czmbompx0001_module_opc-publisher",siteId="<<UnknownSite>>",writerGroupId="x"} 57373 1709313757021

# EOF
```

In order to reduce the content size, we have configured the "publisherId" and "siteId" with shorter string, and the metrics are back afterwards.

**Expected behavior**

Prometheus endpoint always return all documented metrics

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Browser N/A
 - Version 20.04

**Additional context**
OPC Publisher version: 2.9.4

jacqueskang, 2024-03-04:
After have specified "publisherId" and "siteId" with a short string, all metrics are back:

```text
# TYPE process_runtime_dotnet_gc_collections_count_total counter
# HELP process_runtime_dotnet_gc_collections_count_total Number of garbage collections that have occurred since process start.
process_runtime_dotnet_gc_collections_count_total{generation="gen2"} 16 1709536344519
process_runtime_dotnet_gc_collections_count_total{generation="gen1"} 193 1709536344519
process_runtime_dotnet_gc_collections_count_total{generation="gen0"} 4299 1709536344519

# TYPE process_runtime_dotnet_gc_objects_size_bytes gauge
# UNIT process_runtime_dotnet_gc_objects_size_bytes bytes
# HELP process_runtime_dotnet_gc_objects_size_bytes Count of bytes currently in use by objects in the GC heap that haven't been collected yet. Fragmentation and other GC committed memory pools are excluded.
process_runtime_dotnet_gc_objects_size_bytes 50195032 1709536344519

# TYPE process_runtime_dotnet_gc_allocations_size_bytes_total counter
# UNIT process_runtime_dotnet_gc_allocations_size_bytes_total bytes
# HELP process_runtime_dotnet_gc_allocations_size_bytes_total Count of bytes allocated on the managed GC heap since the process start. .NET objects are allocated from this heap. Object allocations from unmanaged languages such as C/C++ do not use this heap.
process_runtime_dotnet_gc_allocations_size_bytes_total 64268684480 1709536344519

# TYPE process_runtime_dotnet_gc_committed_memory_size_bytes gauge
# UNIT process_runtime_dotnet_gc_committed_memory_size_bytes bytes
# HELP process_runtime_dotnet_gc_committed_memory_size_bytes The amount of committed virtual memory for the managed GC heap, as observed during the latest garbage collection. Committed virtual memory may be larger than the heap size because it includes both memory for storing existing objects (the heap size) and some extra memory that is ready to handle newly allocated objects in the future. The value will be unavailable until at least one garbage collection has occurred.
process_runtime_dotnet_gc_committed_memory_size_bytes 67608576 1709536344519

# TYPE process_runtime_dotnet_gc_heap_size_bytes gauge
# UNIT process_runtime_dotnet_gc_heap_size_bytes bytes
# HELP process_runtime_dotnet_gc_heap_size_bytes The heap size (including fragmentation), as observed during the latest garbage collection. The value will be unavailable until at least one garbage collection has occurred.
process_runtime_dotnet_gc_heap_size_bytes{generation="gen0"} 2303008 1709536344519
process_runtime_dotnet_gc_heap_size_bytes{generation="gen1"} 1577664 1709536344519
process_runtime_dotnet_gc_heap_size_bytes{generation="gen2"} 17829680 1709536344519
process_runtime_dotnet_gc_heap_size_bytes{generation="loh"} 30729712 1709536344519
process_runtime_dotnet_gc_heap_size_bytes{generation="poh"} 428376 1709536344519

# TYPE process_runtime_dotnet_gc_heap_fragmentation_size_bytes gauge
# UNIT process_runtime_dotnet_gc_heap_fragmentation_size_bytes bytes
# HELP process_runtime_dotnet_gc_heap_fragmentation_size_bytes The heap fragmentation, as observed during the latest garbage collection. The value will be unavailable until at least one garbage collection has occurred.
process_runtime_dotnet_gc_heap_fragmentation_size_bytes{generation="gen0"} 1991680 1709536344519
process_runtime_dotnet_gc_heap_fragmentation_size_bytes{generation="gen1"} 25872 1709536344519
process_runtime_dotnet_gc_heap_fragmentation_size_bytes{generation="gen2"} 360984 1709536344519
process_runtime_dotnet_gc_heap_fragmentation_size_bytes{generation="loh"} 7523144 1709536344519
process_runtime_dotnet_gc_heap_fragmentation_size_bytes{generation="poh"} 0 1709536344519

# TYPE process_runtime_dotnet_gc_duration_nanoseconds_total counter
# UNIT process_runtime_dotnet_gc_duration_nanoseconds_total nanoseconds
# HELP process_runtime_dotnet_gc_duration_nanoseconds_total The total amount of time paused in GC since the process start.
process_runtime_dotnet_gc_duration_nanoseconds_total 14780224000 1709536344519

# TYPE process_runtime_dotnet_jit_il_compiled_size_bytes_total counter
# UNIT process_runtime_dotnet_jit_il_compiled_size_bytes_total bytes
# HELP process_runtime_dotnet_jit_il_compiled_size_bytes_total Count of bytes of intermediate language that have been compiled since the process start.
process_runtime_dotnet_jit_il_compiled_size_bytes_total 2520844 1709536344519

# TYPE process_runtime_dotnet_jit_methods_compiled_count_total counter
# HELP process_runtime_dotnet_jit_methods_compiled_count_total The number of times the JIT compiler compiled a method since the process start. The JIT compiler may be invoked multiple times for the same method to compile with different generic parameters, or because tiered compilation requested different optimization settings.
process_runtime_dotnet_jit_methods_compiled_count_total 33952 1709536344519

# TYPE process_runtime_dotnet_jit_compilation_time_nanoseconds_total counter
# UNIT process_runtime_dotnet_jit_compilation_time_nanoseconds_total nanoseconds
# HELP process_runtime_dotnet_jit_compilation_time_nanoseconds_total The amount of time the JIT compiler has spent compiling methods since the process start.
process_runtime_dotnet_jit_compilation_time_nanoseconds_total 13506887400 1709536344519

# TYPE process_runtime_dotnet_monitor_lock_contention_count_total counter
[... cut ...]
iiot_edge_publisher_messages_duration_bucket{publisherId="mpx",siteId="czmbompx0001",writerGroupId="x",le="10000"} 21769 1709536344519
iiot_edge_publisher_messages_duration_bucket{publisherId="mpx",siteId="czmbompx0001",writerGroupId="x",le="+Inf"} 21769 1709536344519
iiot_edge_publisher_messages_duration_sum{publisherId="mpx",siteId="czmbompx0001",writerGroupId="x"} 123716 1709536344519
iiot_edge_publisher_messages_duration_count{publisherId="mpx",siteId="czmbompx0001",writerGroupId="x"} 21769 1709536344519

# EOF
```

marcschier, 2024-03-07:
Likely due to the default of MaxMetricStreams in OTEL library being 1000.  You could try setting
 
`
if (exporterOptions.Endpoint.Host != Otlp.OtlpEndpointDisabled)
{
    builder.SetMaxMetricStreams(3000);
   ...
` 

in the Configuration.cs file of the module. If that works, please let me know.  Otherwise, I have added to 2.9.5 release.

jacqueskang, 2024-03-15:
I think we should fix the issue by removing low-value and non-documented metrics. 

Why 3000 should be enough? Why not 5000 or 10000?

I attached below another sample we captured in production. Half of the metrics response is filled with `http_server_request_duration_seconds_bucket`. This metric could be useful for a web site but not for opc publisher.

We care more abount "iiot_" metrics which are unfortunately missing

```
# TYPE process_runtime_dotnet_gc_collections_count_total counter
# HELP process_runtime_dotnet_gc_collections_count_total Number of garbage collections that have occurred since process start.
process_runtime_dotnet_gc_collections_count_total{generation="gen2"} 33 1710519082698
process_runtime_dotnet_gc_collections_count_total{generation="gen1"} 131 1710519082698
process_runtime_dotnet_gc_collections_count_total{generation="gen0"} 3366 1710519082698

# TYPE process_runtime_dotnet_gc_objects_size_bytes gauge
# UNIT process_runtime_dotnet_gc_objects_size_bytes bytes
# HELP process_runtime_dotnet_gc_objects_size_bytes Count of bytes currently in use by objects in the GC heap that haven't been collected yet. Fragmentation and other GC committed memory pools are excluded.
process_runtime_dotnet_gc_objects_size_bytes 175815176 1710519082698

# TYPE process_runtime_dotnet_gc_allocations_size_bytes_total counter
# UNIT process_runtime_dotnet_gc_allocations_size_bytes_total bytes
# HELP process_runtime_dotnet_gc_allocations_size_bytes_total Count of bytes allocated on the managed GC heap since the process start. .NET objects are allocated from this heap. Object allocations from unmanaged languages such as C/C++ do not use this heap.
process_runtime_dotnet_gc_allocations_size_bytes_total 192879568168 1710519082698

# TYPE process_runtime_dotnet_gc_committed_memory_size_bytes gauge
# UNIT process_runtime_dotnet_gc_committed_memory_size_bytes bytes
# HELP process_runtime_dotnet_gc_committed_memory_size_bytes The amount of committed virtual memory for the managed GC heap, as observed during the latest garbage collection. Committed virtual memory may be larger than the heap size because it includes both memory for storing existing objects (the heap size) and some extra memory that is ready to handle newly allocated objects in the future. The value will be unavailable until at least one garbage collection has occurred.
process_runtime_dotnet_gc_committed_memory_size_bytes 240283648 1710519082698

# TYPE process_runtime_dotnet_gc_heap_size_bytes gauge
# UNIT process_runtime_dotnet_gc_heap_size_bytes bytes
# HELP process_runtime_dotnet_gc_heap_size_bytes The heap size (including fragmentation), as observed during the latest garbage collection. The value will be unavailable until at least one garbage collection has occurred.
process_runtime_dotnet_gc_heap_size_bytes{generation="gen0"} 319520 1710519082698
process_runtime_dotnet_gc_heap_size_bytes{generation="gen1"} 6565936 1710519082698
process_runtime_dotnet_gc_heap_size_bytes{generation="gen2"} 90860264 1710519082698
process_runtime_dotnet_gc_heap_size_bytes{generation="loh"} 63800504 1710519082698
process_runtime_dotnet_gc_heap_size_bytes{generation="poh"} 683816 1710519082698

# TYPE process_runtime_dotnet_gc_heap_fragmentation_size_bytes gauge
# UNIT process_runtime_dotnet_gc_heap_fragmentation_size_bytes bytes
# HELP process_runtime_dotnet_gc_heap_fragmentation_size_bytes The heap fragmentation, as observed during the latest garbage collection. The value will be unavailable until at least one garbage collection has occurred.
process_runtime_dotnet_gc_heap_fragmentation_size_bytes{generation="gen0"} 6816 1710519082698
process_runtime_dotnet_gc_heap_fragmentation_size_bytes{generation="gen1"} 127008 1710519082698
process_runtime_dotnet_gc_heap_fragmentation_size_bytes{generation="gen2"} 10170888 1710519082698
process_runtime_dotnet_gc_heap_fragmentation_size_bytes{generation="loh"} 24415312 1710519082698
process_runtime_dotnet_gc_heap_fragmentation_size_bytes{generation="poh"} 0 1710519082698

# TYPE process_runtime_dotnet_gc_duration_nanoseconds_total counter
# UNIT process_runtime_dotnet_gc_duration_nanoseconds_total nanoseconds
# HELP process_runtime_dotnet_gc_duration_nanoseconds_total The total amount of time paused in GC since the process start.
process_runtime_dotnet_gc_duration_nanoseconds_total 25368350000 1710519082698

# TYPE process_runtime_dotnet_jit_il_compiled_size_bytes_total counter
# UNIT process_runtime_dotnet_jit_il_compiled_size_bytes_total bytes
# HELP process_runtime_dotnet_jit_il_compiled_size_bytes_total Count of bytes of intermediate language that have been compiled since the process start.
process_runtime_dotnet_jit_il_compiled_size_bytes_total 2993582 1710519082698

# TYPE process_runtime_dotnet_jit_methods_compiled_count_total counter
# HELP process_runtime_dotnet_jit_methods_compiled_count_total The number of times the JIT compiler compiled a method since the process start. The JIT compiler may be invoked multiple times for the same method to compile with different generic parameters, or because tiered compilation requested different optimization settings.
process_runtime_dotnet_jit_methods_compiled_count_total 40299 1710519082698

# TYPE process_runtime_dotnet_jit_compilation_time_nanoseconds_total counter
# UNIT process_runtime_dotnet_jit_compilation_time_nanoseconds_total nanoseconds
[... cut ...]
iiot_edge_publisher_messages_duration_bucket{publisherId="mpx",siteId="plgowmpx0001",writerGroupId="test",le="10000"} 8 1710519082698
iiot_edge_publisher_messages_duration_bucket{publisherId="mpx",siteId="plgowmpx0001",writerGroupId="test",le="+Inf"} 8 1710519082698
iiot_edge_publisher_messages_duration_sum{publisherId="mpx",siteId="plgowmpx0001",writerGroupId="test"} 56 1710519082698
iiot_edge_publisher_messages_duration_count{publisherId="mpx",siteId="plgowmpx0001",writerGroupId="test"} 8 1710519082698

# EOF
```

alxy, 2024-04-30:
@jacqueskang 
> We are paying Azure for transfering metrics over IoT Hub and storaging them in Log analytics workspace. We should have an option to disable the metrics which we don't need

If you are using `metricscollector` to collect and send the metrics to Azure, I can highly recommend to setup the environment variable `AllowedMetrics=iiot_edge_publisher_*` or similar on this module. As we all know, log analytics is really not the most cheap storage in the world...

jacqueskang, 2024-06-11:
I would really appreciate if this regression can be fixed instead of being postponed again and again to later version.

We have deployed in production some Azure Monitor alert rules and dashboard relying on Prometheus metrics which are now missing since v2.9.3. This bug makes our operation team difficult to monitor the system.

marcschier, 2024-06-17:
In 2.9.9 we are adding a--eri command line option to enable all automatic .net runtime instrumentation. By default it will be off.  This will reduce the metrics significantly and align with previous behavior.  We will also disable "total" suffix, which can be enabled by --ets command line option.  Furthermore response caching will be disabled, cannot be enabled/configured.

closed 2024-06-17 by marcschier