Azure/Industrial-IoT #2198: What Nugget package do i need to use to access the publisher models?

RollsChris, 2024-03-04:
Hi,

I have tried using **Microsoft.Azure.IIoT.OpcUa.Api.Publisher** but none of the models i'm looking for are there..

For example: PublishDiagnosticInfoModel..

I think i must be doing something wrong of i have the wrong nugget package?

any ideas?

Thanks

marcschier, 2024-03-06:
We stopped publishing nuget packages years ago.  You can use generators like autorest together with the openapi.json (swagger) in the docs folders to generate API clients in any language.  The resulting models can then also be used to to serde requests/responses when using other transports than REST.

closed 2024-03-06 by marcschier

RollsChris, 2024-03-06:
Thanks for the response... when trying to generate types in csharp, using the opc-publisher/openapi.json file... im getting this error: 

> fatal   | Error: Unknown type for constant value for String 'int32'--cannot create constant value.
fatal   | Process() cancelled due to failure 
error   |   Error: Plugin modelerfour reported failure.
error   | Autorest completed with an error. If you think the error message is unclear, or is a bug, please declare an issues at https://github.com/Azure/autorest/issues with the error message you are seeing.

the command im running is:

`autorest --input-file=opcpublisher.json --csharp --ouput-folder=generated --verbose`

does this mean that its broken ?