Azure/Industrial-IoT #2199: When trying to generate types in csharp, using the opc-publisher/openapi.json file... im getting this error:

RollsChris, 2024-03-06:
When trying to generate types in csharp, using the opc-publisher/openapi.json file... im getting this error: 

> fatal   | Error: Unknown type for constant value for String 'int32'--cannot create constant value.
fatal   | Process() cancelled due to failure 
error   |   Error: Plugin modelerfour reported failure.
error   | Autorest completed with an error. If you think the error message is unclear, or is a bug, please declare an issues at https://github.com/Azure/autorest/issues with the error message you are seeing.

the command im running ( i renamed file to opcpublisher.json)  is:

`autorest --input-file=opcpublisher.json --csharp --ouput-folder=generated --verbose`

does this mean that its broken ?

marcschier, 2024-03-07:
Yes, I will look into this.

marcschier, 2024-03-07:
If you would like to help, you could take the json file and identify the offending line(s), then manually update until autorest likes it.  Then I can fix the swaggergen part and re-generate the json in main branch pretty quickly.

RollsChris, 2024-03-08:
i would but it does not tell me what line it is? or am i missing something. 

How would you problem solve it, find the offending line.. .check the actual API implementation and see if the types match?

marcschier, 2024-03-08:
Might be related to https://github.com/Azure/autorest.csharp/issues/4127 and the format for the enums breaking it.  This used to work just fine a year ago when we still generated code but I must confess we have never run Autorest since then   maybe this gives a clue and replacing the format int32 there is worth a try.

marcschier, 2024-06-28:
Fixed in main for 2.9.10 target

closed 2024-06-28 by marcschier

RollsChris, 2025-07-03:
Hi, @marcschier. I have tried this this morning and it works, but didn't contain quite what i was looking for.. 

I'm looking for the model for "ua-data" and "ua-meta" data messages coming from the opc publisher?. im thinking the rest api might not be what i wanted