Azure/Industrial-IoT #2201: 'RemoveRequest' response '404 Endpoint not found' is misleading

Haeavar, 2024-03-12:
**Describe the bug**
If the `PATCH /v2/configuration` endpoint ist called with empty `OpcNodes`, the OpcPublisher tries to remove the DataSetWriter. If the DataSetWriter was not found (see details in `PublishedNodesEntryModelEx.HasSameDataSet` it fails with an error `Response 404 Endpoint not found: opc.tcp://172.29.102.1:4842`.
If you have multiple DataSetWriter with the same endpoint its difficult to understand which one could not be found.

**To Reproduce**
Steps to reproduce the behavior:
1. Call PATCH /v2/configuration with empty OpcNodes and some more DataSetWriter with the same endpoint URL.
2. See error
Furly.Exceptions.MethodCallStatusException
  HResult=0x80131500
  Message=Response 404 Endpoint not found: opc.tcp://172.29.102.1:4842
  Source=Azure.IIoT.OpcUa.Publisher
  StackTrace:
   at Azure.IIoT.OpcUa.Publisher.Services.PublisherConfigurationService.<AddOrUpdateEndpointsAsync>d__9.MoveNext() 

**Expected behavior**
At least the 'DataSetWriterId' should be reported. Better would be to report all information which distinguish a DataSetWrier

**Additional context**
image: mcr.microsoft.com/iotedge/opc-publisher:2.9.4

marcschier, 2024-06-27:
First off, would the new [writer API}(https://github.com/Azure/Industrial-IoT/blob/main/docs/opc-publisher/api.md#writer) be a good alternative? It returns better error codes, but still returns != 200 if the writer does not exist.  

The issue really is that there is no support for "empty" writers (no nodes) today, but this is something I will add in 2.9.10.  This will make the configuration more "restful" and a) allow creating an empty writer, then filling it, and b) removing the last node and not removing the entry.

marcschier, 2024-06-27:
#2269

marcschier, 2024-06-28:
Some more changes coming to use ExceptionSummarization to provide a "ProblemDetails" response, rather than a simple string.  The response would be a json object with similar information as ProblemDetails in ASP.net core and should provide be better handling than today.

closed 2024-06-29 by marcschier