Azure/Industrial-IoT #2203: Error during command sending results in publisher getting in erroneous state

NoTuxNoBux, 2024-03-19:
**Describe the bug**
Sending a command to an OPC UA server the publisher 2.9.4 is connected to, and the connection unexpectedly dropping (the machine stopped responding) resulted in the publisher getting into a permanent exception state where it spams the same exception every time.

This has the same result as and might be related to #2189, but is caused in a different way.

**To Reproduce**
I can't always reproduce this in the exact same fashion, but this is what happened:
1. Ensure OPC UA server is up and running.
2. Get the publisher connected to it.
3. Send a command over the REST API to the OPC UA server.
4. Disconnect the OPC UA server whilst the command is executing.

**Expected behavior**
The publisher recovers and treats the connection to the server as disconnected, trying to reconnect (and not throw exceptions).

**Additional context**
Logs of what happened when I sent the command (beginning), resulting in the exceptions (end):

```
Mar 19 11:40:19 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]: [24-03-19 10:40:19.6866] info: Microsoft.AspNetCore.Hosting.Diagnostics[1]
Mar 19 11:40:19 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       => SpanId:5e980f0d671fc53b, TraceId:0f7feb9e39c9502e37f9d58b58cccc65, ParentId:657af900a3f5d520 => ConnectionId:0HN2800MLJBLV => RequestPath:/v2/write RequestId:0HN2800MLJBLV:00000001
Mar 19 11:40:19 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       Request starting HTTP/1.1 POST https://REMOVED-IP-OF-HOST:8080/v2/write - application/json;+charset=utf-8 314
Mar 19 11:40:19 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]: [24-03-19 10:40:19.8827] info: Microsoft.AspNetCore.Routing.EndpointMiddleware[0]
Mar 19 11:40:19 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       => SpanId:5e980f0d671fc53b, TraceId:0f7feb9e39c9502e37f9d58b58cccc65, ParentId:657af900a3f5d520 => ConnectionId:0HN2800MLJBLV => RequestPath:/v2/write RequestId:0HN2800MLJBLV:00000001
Mar 19 11:40:19 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       Executing endpoint 'Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)'
Mar 19 11:40:19 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]: [24-03-19 10:40:19.9227] info: Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker[102]
Mar 19 11:40:19 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       => SpanId:5e980f0d671fc53b, TraceId:0f7feb9e39c9502e37f9d58b58cccc65, ParentId:657af900a3f5d520 => ConnectionId:0HN2800MLJBLV => RequestPath:/v2/write RequestId:0HN2800MLJBLV:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
Mar 19 11:40:19 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       Route matched with {action = "ValueWrite", controller = "General"}. Executing controller action with signature System.Threading.Tasks.Task`1[Azure.IIoT.OpcUa.Publisher.Models.ValueWriteResponseModel] ValueWriteAsync(Azure.IIoT.OpcUa.Publisher.Models.RequestEnvelope`1[Azure.IIoT.OpcUa.Publisher.Models.ValueWriteRequestModel], System.Threading.CancellationToken) on controller Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController (Azure.IIoT.OpcUa.Publisher.Module).
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]: [24-03-19 10:40:20.0927] info: Microsoft.AspNetCore.Mvc.Infrastructure.ControllerActionInvoker[101]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       => SpanId:5e980f0d671fc53b, TraceId:0f7feb9e39c9502e37f9d58b58cccc65, ParentId:657af900a3f5d520 => ConnectionId:0HN2800MLJBLV => RequestPath:/v2/write RequestId:0HN2800MLJBLV:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       Executing action method Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module) - Validation state: Valid
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]: [24-03-19 10:40:20.0982] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClientManager[0]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       => SpanId:5e980f0d671fc53b, TraceId:0f7feb9e39c9502e37f9d58b58cccc65, ParentId:657af900a3f5d520 => ConnectionId:0HN2800MLJBLV => RequestPath:/v2/write RequestId:0HN2800MLJBLV:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       New client opc.tcp://robot.beacon.internal:4880/Staubli_69B8C685 [state:Disconnected|refs:0] created.
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]: [24-03-19 10:40:20.0991] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       => SpanId:5e980f0d671fc53b, TraceId:0f7feb9e39c9502e37f9d58b58cccc65, ParentId:657af900a3f5d520 => ConnectionId:0HN2800MLJBLV => RequestPath:/v2/write RequestId:0HN2800MLJBLV:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       Session opc.tcp://robot.beacon.internal:4880/Staubli_69B8C685 with opc.tcp://robot.beacon.internal:4880/Staubli changed from Disconnected to Connecting
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]: [24-03-19 10:40:20.0997] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       => SpanId:5e980f0d671fc53b, TraceId:0f7feb9e39c9502e37f9d58b58cccc65, ParentId:657af900a3f5d520 => ConnectionId:0HN2800MLJBLV => RequestPath:/v2/write RequestId:0HN2800MLJBLV:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       Connecting Client opc.tcp://robot.beacon.internal:4880/Staubli_69B8C685 [state:Connecting|refs:1] to opc.tcp://robot.beacon.internal:4880/Staubli...
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]: [24-03-19 10:40:20.9691] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       => SpanId:5e980f0d671fc53b, TraceId:0f7feb9e39c9502e37f9d58b58cccc65, ParentId:657af900a3f5d520 => ConnectionId:0HN2800MLJBLV => RequestPath:/v2/write RequestId:0HN2800MLJBLV:00000001 => Azure.IIoT.OpcUa.Publisher.Module.Controllers.GeneralController.ValueWriteAsync (Azure.IIoT.OpcUa.Publisher.Module)
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       Selecting endpoint opc.tcp://robot.beacon.internal:4880/Staubli with SecurityMode SignAndEncrypt and any SecurityPolicyUri from:
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:             #001: opc.tcp://robot.beacon.internal:4880/Staubli|None [http://opcfoundation.org/UA/SecurityPolicy#None]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:             #007: opc.tcp://robot.beacon.internal:4880/Staubli|Sign [http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:             #012: opc.tcp://robot.beacon.internal:4880/Staubli|Sign [http://opcfoundation.org/UA/SecurityPolicy#Basic256]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:             #052: opc.tcp://robot.beacon.internal:4880/Staubli|Sign [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:             #008: opc.tcp://robot.beacon.internal:4880/Staubli|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:             #013: opc.tcp://robot.beacon.internal:4880/Staubli|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:             #053: opc.tcp://robot.beacon.internal:4880/Staubli|SignAndEncrypt [http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256]
Mar 19 11:40:20 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]: [24-03-19 10:40:20.9710] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaClient[0]
[... cut ...]
Mar 19 11:40:34 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:       Opc.Ua.ServiceResultException: BadConnectionClosed
Mar 19 11:40:34 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:          at Opc.Ua.Bindings.UaSCUaBinaryClientChannel.BeginSendRequest(IServiceRequest request, Int32 timeout, AsyncCallback callback, Object state)
Mar 19 11:40:34 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]:          at Opc.Ua.Client.Session.BeginPublish(Int32 timeout)
Mar 19 11:40:35 Azure-IoT-Edge-Ubuntu-Host 6081cab9195c[770]: [24-03-19 10:40:35.0872] info: Azure.IIoT.OpcUa.Publisher.Stack.Services.OpcUaSubscription[0]
```

marcschier, 2024-06-17:
Unfortunately the OPC UA stack logs the exception, which actually are just indications of the service result status of the last publish request (since subscriptions are active).  I cannot disable the error logs from the stack, and that unfortunately includes the exception stack trace.  

The problem with MQTT transport is that the MQTT RPC call hangs forever on the reconnect (#2213), but the HTTP call should actually respond with error (at least after a while).  Is this the case @NoTuxNoBux ?

marcschier, 2024-06-18:
On hung calls, see resolution on the linked issue.  No way I can work around the publish exception logs though.

closed 2024-06-18 by marcschier