Azure/Industrial-IoT #2204: OPCPublisher accessing raw.githubusercontent.com hardcoded, but should not

corporateaffiliate, 2024-03-25:
**Describe the bug**
The OPCPublisher module seems to have hardcoded the raw.githubusercontent.com FQDN. This means deploying the module as 3rd party module on an IoT device running Azure IoT Edge 1.4 runtime will result in an invocation of raw.githubusercontent.com. This in turn implies that firewalls need be whitelisted for this FQDN. This may not be an issue in a homenetwork, but it is a security issue corporate IT infrastructure. Selling IoT devices to other companies that run Azure IoT Edge and require raw.githubusercontent.com to be whitelisted is a thorn in a security practitioner's side. This is not only an effective security issue, but also a marketing-security issue. The only Azure IoT Edge module of our solution that does require that FQDN is the OPCPublisher.

Remark: This may not by everyone considered a "bug". Some may say this is a feature (improvement), some may say this is a security vulnerability (as it is security relevant in context of deployment and its implications on IT infrastructure). I fully understand, please move it then into the approriate category.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Azure IoT Edge & Docker
2. Install/deploy OPCPublisher 2.9
3. Check the remote FQDN that have been accessed

**Expected behavior**
There is no access to raw.githubusercontent.com. There could be, for example. mcr.microsoft.com, this would be acceptable.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows 
 - Browser Edge
 - Version122.0.2365.92 (Official build) (64-bit)

marcschier, 2024-03-26:
Do you know what happens if it is not whitelisted? Can you provide the error that you see then? I am not aware of anything needing this access and the only area where I think it could be the case is e.g. a license file in a consumed nuget but even there I am not sure why and when that would be triggered.

corporateaffiliate, 2024-03-26:
All what we see is that once the OPCPublisher is being deployed, there will be some traffic to that FQDN. What exactly its doing, we dont know. In any case - even if no impact - it will raise eyebrows from security guys in corporate IT seing that its "probed". The impact we dont know exactly, we can try to test that for you and come back to you.

marcschier, 2024-04-15:
What command line options did you use? Also, what tool did you use to find the call out?  I have tried wireshark and fiddler and could not repro.

closed 2024-06-16 by marcschier